openwrt/openwrt #16698: Odhcp6c currently lacks maintaining

Headcrabed, 2024-10-13:
### Describe the bug

odcp6c package (https://github.com/openwrt/odhcp6c) hasn't been maintained for more than 1 year. During this time, the maintainer didn't reply to any PRs and issues. Seems maybe we need to change the maintainer of this project?

### OpenWrt version

r23809-234f1a2efa

### OpenWrt release

23.05.3

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R3

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Neustradamus, 2024-10-27:
@Headcrabed: I confirm it, and there are several PRs:
- https://github.com/openwrt/odhcp6c/pulls

Benqer0, 2025-05-27:
Is there any update here?

Neustradamus, 2025-10-19:
Dear @Headcrabed, @Benqer0,

Good news, there are some recent changes in odhcp6c repository!

Recall: One year ago, I have created a ticket linked to this one:
- https://github.com/openwrt/odhcp6c/issues/92

Thanks to @systemcrash, @Noltari.

cc: @Alphix.

Alphix, 2025-10-19:
> Does DNSMasq do what Odhcp6c does?

dnsmasq is a DNS forwarder, and provides a DHCPv4/DHCPv6/TFTP server (the DHCPv6 server isn't used out-of-the box in OpenWrt).

odhcpd is (inter alia) a DHCPv4/DHCPv6 server (the DHCPv6 server is used out-of-the box in OpenWrt).

odhcp6c is OpenWrt's DHCPv6 client.

udhcpc (from busybox) is the DHCPv4 client used in OpenWrt.

Noltari, 2025-11-05:
Closing this since I'm now maintaining odhcp6c as good as I can with the help of @Alphix and @systemcrash.
It would be great to get some help from other OpenWrt core developers though.

Neustradamus, 2025-11-05:
@Noltari: Nice, thanks! :)

Yes with @systemcrash and @Alphix help!

It is same for odhcpd?
- https://github.com/openwrt/odhcpd