openwrt/openwrt #16708: GNUmakefile:108: abort-due-to-no-makefile Error 1

peterwillcn, 2024-10-15:
### Describe the bug

libtool: compile:  /home/openwrt/master/openwrt/staging_dir/host/bin/gcc -DPACKAGE_NAME=\"libressl\" -DPACKAGE_TARNAME=\"libressl\" -DPACKAGE_VERSION=\"3.7.3\" "-DPACKAGE_STRING=\"libressl 3.7.3\"" -DPACKAGE_BUGREPORT=\"\" -DPACKAGE_URL=\"\" -DPACKAGE=\"libressl\" -DVERSION=\"3.7.3\" -DHAVE_STDIO_H=1 -DHAVE_STDLIB_H=1 -DHAVE_STRING_H=1 -DHAVE_INTTYPES_H=1 -DHAVE_STDINT_H=1 -DHAVE_STRINGS_H=1 -DHAVE_SYS_STAT_H=1 -DHAVE_SYS_TYPES_H=1 -DHAVE_UNISTD_H=1 -DSTDC_HEADERS=1 -DHAVE_DLFCN_H=1 -DLT_OBJDIR=\".libs/\" -DSIZEOF_TIME_T=8 -DHAVE_SYMLINK=1 -DHAVE_ENDIAN_H=1 -DHAVE_ERR_H=1 -DHAVE_NETINET_IP_H=1 -DHAVE_SYS_TYPES_H=1 -DHAVE_NETINET_IN_H=1 -DHAVE_ARPA_NAMESER_H=1 -DHAVE_NETDB_H=1 -DHAVE_RESOLV_H=1 -DHAVE_ASPRINTF=1 -DHAVE_MEMMEM=1 -DHAVE_REALLOCARRAY=1 -DHAVE_STRNDUP=1 -DHAVE_STRNLEN=1 -DHAVE_STRSEP=1 -DHAVE_TIMEGM=1 -DHAVE_SYSLOG=1 -DHAVE_ACCEPT4=1 -DHAVE_PIPE2=1 -DHAVE_POLL=1 -DHAVE_SOCKETPAIR=1 -DHAVE_EXPLICIT_BZERO=1 -DHAVE_GETAUXVAL=1 -DHAVE_GETAUXVAL=1 -DHAVE_DL_ITERATE_PHDR=1 -DHAVE_CLOCK_GETTIME=1 -DHAVE_VA_COPY=1 -DHAVE___VA_COPY=1 -DHAS_GNU_WARNING_LONG=1 -I. -I../crypto/hidden -I../include -I/home/openwrt/master/openwrt/build_dir/host/libressl-3.7.3/include -I../include/compat -DLIBRESSL_INTERNAL -D__BEGIN_HIDDEN_DECLS= -D__END_HIDDEN_DECLS= -DLIBRESSL_CRYPTO_INTERNAL -I../crypto/asn1 -I../crypto/bio -I../crypto/bn -I../crypto/bytestring -I../crypto/curve25519 -I../crypto/dh -I../crypto/dsa -I../crypto/ec -I../crypto/ecdh -I../crypto/ecdsa -I../crypto/evp -I../crypto/hmac -I../crypto/modes -I../crypto/ocsp -I../crypto/pkcs12 -I../crypto/rsa -I../crypto/x509 -I../crypto -DLIBRESSL_INTERNAL -DOPENSSL_NO_HW_PADLOCK -DOPENSSLDIR=\"/home/openwrt/master/openwrt/staging_dir/host/etc/ssl\" -DOPENSSL_NO_ASM -I../crypto/bn/arch/amd64/ -I/home/openwrt/master/openwrt/staging_dir/host/include -D_DEFAULT_SOURCE -D_BSD_SOURCE -D_POSIX_SOURCE -D_GNU_SOURCE -O2 -I/home/openwrt/master/openwrt/staging_dir/host/include -DPIC -fPIC -Wall -std=gnu99 -fno-strict-aliasing -fno-strict-overflow -fstack-protector-strong -DHAVE_GNU_STACK -Wno-pointer-sign -c engine/eng_pkey.c  -fPIC -DPIC -o engine/libcrypto_la-eng_pkey.o
make[4]: Entering directory '/home/openwrt/master/openwrt/build_dir/host/libtool-2.4.7'
There seems to be no Makefile in this directory.
You must run ./configure before running 'make'.
make[4]: *** [GNUmakefile:108: abort-due-to-no-makefile] Error 1
make[4]: Leaving directory '/home/openwrt/master/openwrt/build_dir/host/libtool-2.4.7'
make[3]: [Makefile:68: /home/openwrt/master/openwrt/build_dir/host/libtool-2.4.7/.prepared6d4a1f5d04dba1df3f3041fdaeb49c8a_18f1e190c5d53547fed41a3eaa76e9e9] Error 2 (ignored)
(cd /home/openwrt/master/openwrt/staging_dir/host/share/aclocal/ && rm -f libtool.m4 ltdl.m4 lt~obsolete.m4 ltoptions.m4 ltsugar.m4 ltversion.m4)

### OpenWrt version

r27736-eb05baff7f

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

rockchip/armv8

### Device

FriendlyElec NanoPC-T6

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

egc112, 2024-11-25:
I have the same error reported

SuperKali, 2025-01-26:
I have the same error