openwrt/openwrt #16725: mpc85xx: p1010: Watchguard T10 unbootable with current SNAPSHOT

Shine-, 2024-10-17:
### Describe the bug

The Watchguard T10 is unbootable with the current SNAPSHOT image.

After upgrading from 23.05.x to SNAPSHOT, or when booting an initramfs image, the boot log displays a kernel decompression error.
<details><summary>boot log</summary>

```
## Booting kernel from FIT Image at 01000000 ...
   Using 'config-1' configuration
   Trying 'kernel-1' kernel subimage
     Description:  POWERPC OpenWrt Linux-6.6.56
     Type:         Kernel Image
     Compression:  gzip compressed
     Data Start:   0x010000ec
     Data Size:    8660044 Bytes = 8.3 MiB
     Architecture: PowerPC
     OS:           Linux
     Load Address: 0x00000000
     Entry Point:  0x00000000
     Hash algo:    crc32
     Hash value:   1e81b8ac
     Hash algo:    sha1
     Hash value:   c5fbbe994771abf20d519779c6b488ac9f68386e
   Verifying Hash Integrity ... crc32+ sha1+ OK
## Flattened Device Tree from FIT Image at 01000000
   Using 'config-1' configuration
   Trying 'fdt-1' FDT blob subimage
     Description:  POWERPC OpenWrt watchguard_firebox-t10 device tree blob
     Type:         Flat Device Tree
     Compression:  uncompressed
     Data Start:   0x01842680
     Data Size:    11782 Bytes = 11.5 KiB
     Architecture: PowerPC
     Hash algo:    crc32
     Hash value:   a61c5c43
     Hash algo:    sha1
     Hash value:   820c704b24581709e6a9df775b8d50dda6f411eb
   Verifying Hash Integrity ... crc32+ sha1+ OK
   Booting using the fdt blob at 0x1842680
   Uncompressing Kernel Image ... Error: inflate() returned -5
GUNZIP: uncompress, out-of-mem or overwrite error - must RESET board to recover
```
</details>

### OpenWrt version

r27796-caf5caae44

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mpc85xx/p1010

### Device

Watchguard Firebox T10

### Image kind

Official downloaded image

### Steps to reproduce

Flash current SNAPSHOT image to the Watchguard T10 and reboot.
Alternatively, try booting current SNAPSHOT initramfs image via uBoot console.

### Actual behaviour

Boot loop with error message as outlined above.

### Expected behaviour

Device boots with current SNAPSHOT image.

### Additional info

I noticed this issue while I was testing #16310 for @neheb. You can find some (unsuccessful) initial troubleshooting attempts in the discussion for that PR.

While proceeding to analyse some more, I noticed that the kernel 6.6 image for likely the whole mpc85xx target - but at least for the P1010/P1020 subtarget - has grown to **more than double** the size of what it was with 23.05.

23.05.5 vmlinux = around 8.2 MB
current SNAPSHOT vmlinux = 16,8 MB

Comparison: For other targets (e.g. ath79), the kernel image has only grown by a few hundred kB, e.g. from 7.0 to 7.6 MB.

During my testing, I started suspecting that the Watchguard T10 uBoot has a hard limit of 16 MB for the uncompressed/decompressed kernel image, which would explain the behavior, despite the device having plenty of flash and RAM to handle a ~17MB kernel. This suspicion comes from my attempts of booting an uncompressed image, resulting in error messages similar to `ERROR: Failed to allocate 0xXXXX bytes below 0x1000000.`

I assume that a kernel size of ~17MB isn't how it should be, anyway, but I'm lost as to what might have caused this size increase between 5.15 and 6.6 specifically for this target, while other targets appear to be unaffected.

### Diffconfig

```text
n/a
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

neheb, 2024-10-17:
```
> diff target/linux/mpc85xx/config-5.15 ../openwrt/target/linux/mpc85xx/config-6.6 
4a5
> CONFIG_ARCH_DMA_DEFAULT_COHERENT=y
14a16,18
> CONFIG_ARCH_OPTIONAL_KERNEL_RWX=y
> CONFIG_ARCH_OPTIONAL_KERNEL_RWX_DEFAULT=y
> CONFIG_ARCH_SPLIT_ARG64=y
21a26
> CONFIG_BOOKE_OR_40x=y
26a32,33
> CONFIG_CC_IMPLICIT_FALLTHROUGH="-Wimplicit-fallthrough=5"
> CONFIG_CC_NO_ARRAY_BOUNDS=y
32a40
> CONFIG_COMPACT_UNEVICTABLE_DEFAULT=1
40a49
> CONFIG_CRYPTO_LIB_GF128MUL=y
41a51,52
> CONFIG_CRYPTO_LIB_SHA1=y
> CONFIG_CRYPTO_LIB_UTILS=y
49c60
< CONFIG_DATA_SHIFT=12
---
> CONFIG_DATA_SHIFT=24
50a62
> CONFIG_DEBUG_INFO=y
53d64
< CONFIG_E500=y
61a73
> CONFIG_EXCLUSIVE_SYSTEM_RAM=y
64,65d75
< CONFIG_FSL_BOOKE=y
< # CONFIG_FSL_DPAA2_SWITCH is not set
72a83,85
> # CONFIG_FSL_ULI1575 is not set
> CONFIG_FS_IOMAP=y
> CONFIG_FUNCTION_ALIGNMENT=0
74a88,90
> CONFIG_FW_LOADER_SYSFS=y
> CONFIG_GCC11_NO_ARRAY_BOUNDS=y
> CONFIG_GCC_ASM_GOTO_OUTPUT_WORKAROUND=y
84a101,102
> CONFIG_GENERIC_IDLE_POLL_SETUP=y
> CONFIG_GENERIC_IOREMAP=y
100a119
> CONFIG_HAS_IOPORT=y
119c138
< CONFIG_LOWMEM_CAM_NUM=3
---
> CONFIG_LOWMEM_CAM_NUM=9
126d144
< CONFIG_MEMFD_CREATE=y
127a146
> CONFIG_MMU_GATHER_MERGE_VMAS=y
128a148
> CONFIG_MMU_LAZY_TLB_REFCOUNT=y
150a171,172
> CONFIG_NET_EGRESS=y
> CONFIG_NET_INGRESS=y
151a174
> CONFIG_NET_XGRESS=y
155a179
> CONFIG_NVMEM_LAYOUTS=y
159d182
< CONFIG_OF_DMA_DEFAULT_COHERENT=y
162a186
> CONFIG_OF_GPIO_MM_GPIOCHIP=y
172a197,199
> CONFIG_PAGE_POOL=y
> CONFIG_PAGE_SIZE_LESS_THAN_256KB=y
> CONFIG_PAGE_SIZE_LESS_THAN_64KB=y
184a212
> CONFIG_PHYLIB_LEDS=y
200d227
< CONFIG_PPC_BOOK3E_MMU=y
202a230
> CONFIG_PPC_E500=y
205d232
< CONFIG_PPC_FSL_BOOK3E=y
206a234,236
> CONFIG_PPC_KUAP=y
> # CONFIG_PPC_KUAP_DEBUG is not set
> CONFIG_PPC_KUEP=y
207a238
> # CONFIG_PPC_P2020 is not set
208a240
> # CONFIG_PPC_PCI_BUS_NUM_DOMAIN_DEPENDENT is not set
212a245
> CONFIG_PREEMPT_NONE_BUILD=y
215a249
> CONFIG_RANDSTRUCT_NONE=y
228a263
> CONFIG_SMT_NUM_THREADS_DYNAMIC=y
229a265
> CONFIG_SOFTIRQ_ON_OWN_STACK=y
237c273,274
< CONFIG_SRCU=y
---
> CONFIG_SQUASHFS_DECOMP_MULTI_PERCPU=y
> # CONFIG_STATIC_CALL_SELFTEST is not set
```

maybe it's the debug info.

robimarko, 2024-10-18:
https://github.com/openwrt/openwrt/commit/7d768a9ba671e8fb0be6f3b226dc286133009d1f

It looks like the 16MB limit is real, but vmlinux doubling is rather weird

Shine-, 2024-10-18:
Thanks for pointing me to the zImage commit! With this, I could get the T10 to boot SNAPSHOT.
<details><summary>patch</summary>

```
diff --git a/target/linux/mpc85xx/image/p1010.mk b/target/linux/mpc85xx/image/p1010.mk
index 1f898c483f..c612ce706e 100644
--- a/target/linux/mpc85xx/image/p1010.mk
+++ b/target/linux/mpc85xx/image/p1010.mk
@@ -78,7 +78,15 @@ define Device/watchguard_firebox-t10
   DEVICE_VENDOR := Watchguard
   DEVICE_MODEL := Firebox T10
   DEVICE_PACKAGES := kmod-rtc-s35390a kmod-eeprom-at24
-  KERNEL = kernel-bin | gzip | fit gzip $(KDIR)/image-$$(DEVICE_DTS).dtb
+  KERNEL = kernel-bin | fit none $(KDIR)/image-$$(DEVICE_DTS).dtb
+  KERNEL_NAME := zImage.la3000000
+  KERNEL_ENTRY := 0x3000000
+  KERNEL_LOADADDR := 0x3000000
+  BLOCKSIZE := 128k
+  PAGESIZE := 2048
+  SUBPAGESIZE := 2048
+  KERNEL_IN_UBI := 1
+  UBINIZE_OPTS := -E 5
   IMAGES := sysupgrade.bin
   IMAGE/sysupgrade.bin := sysupgrade-tar | append-metadata
 endef
diff --git a/target/linux/mpc85xx/p1010/target.mk b/target/linux/mpc85xx/p1010/target.mk
index 6b54390863..b6236196f3 100644
--- a/target/linux/mpc85xx/p1010/target.mk
+++ b/target/linux/mpc85xx/p1010/target.mk
@@ -1,5 +1,5 @@
 BOARDNAME:=P1010
-KERNEL_IMAGES:=simpleImage.br200-wp simpleImage.tl-wdr4900-v1 simpleImage.ws-ap3715i
+KERNEL_IMAGES:=simpleImage.br200-wp simpleImage.tl-wdr4900-v1 simpleImage.ws-ap3715i zImage.la3000000
 
 define Target/Description
 	Build firmware images for P1010 based boards.
diff --git a/target/linux/mpc85xx/patches-6.6/108-powerpc-85xx-firebox-t10-support.patch.patch b/target/linux/mpc85xx/patches-6.6/108-powerpc-85xx-firebox-t10-support.patch.patch
index 2c722e4d9b..ada0544fc6 100644
--- a/target/linux/mpc85xx/patches-6.6/108-powerpc-85xx-firebox-t10-support.patch.patch
+++ b/target/linux/mpc85xx/patches-6.6/108-powerpc-85xx-firebox-t10-support.patch.patch
@@ -1,6 +1,6 @@
 --- a/arch/powerpc/platforms/85xx/Kconfig
 +++ b/arch/powerpc/platforms/85xx/Kconfig
-@@ -83,6 +83,16 @@ config WS_AP3825I
+@@ -83,6 +83,17 @@ config WS_AP3825I
  	  This board is a Concurrent Dual-Band wireless access point with a
  	  Freescale P1020 SoC.
  
@@ -9,6 +9,7 @@
 +	select DEFAULT_UIMAGE
 +	select ARCH_REQUIRE_GPIOLIB
 +	select GPIO_MPC8XXX
++	select PPC_ZIMAGE_LA3000000
 +	help
 +	  This option enables support for the Watchguard Firebox T10 board.
 +	  This board is a VPN Gateway-Router with a
```
Not sure whether all these additional Image options are really required (copied them from MSM460), but with only load/entry address, the device wouldn't boot.
</details>

Question remains, is it really correct that the mpc85xx kernel is ~17MB large, with >8MB being just zero-padding?

robimarko, 2024-10-18:
Nah, something changed in image generation which should be rather fixed

neheb, 2024-10-18:
```
-rw-r--r--. 1 mangix mangix     3433 Oct 10 20:03 config.buildinfo
-rw-r--r--. 1 mangix mangix      395 Oct 10 20:03 feeds.buildinfo
-rw-r--r--. 1 mangix mangix  8744292 Oct 10 20:04 openwrt-mpc85xx-p1020-aerohive_hiveap-330-initramfs-kernel.bin
-rw-r--r--. 1 mangix mangix  9440950 Oct 10 20:04 openwrt-mpc85xx-p1020-aerohive_hiveap-330-squashfs-sysupgrade.bin
-rw-r--r--. 1 mangix mangix     5292 Oct 10 20:04 openwrt-mpc85xx-p1020-default.manifest
-rw-r--r--. 1 mangix mangix  8744292 Oct 10 20:04 openwrt-mpc85xx-p1020-enterasys_ws-ap3710i-initramfs-kernel.bin
-rw-r--r--. 1 mangix mangix  9175859 Oct 10 20:04 openwrt-mpc85xx-p1020-enterasys_ws-ap3710i-squashfs-sysupgrade.bin
-rw-r--r--. 1 mangix mangix  8758704 Oct 10 20:04 openwrt-mpc85xx-p1020-extreme-networks_ws-ap3825i-initramfs-kernel.bin
-rw-r--r--. 1 mangix mangix  9175873 Oct 10 20:04 openwrt-mpc85xx-p1020-extreme-networks_ws-ap3825i-squashfs-sysupgrade.bin
-rw-r--r--. 1 mangix mangix  8745504 Oct 10 20:04 openwrt-mpc85xx-p1020-hpe_msm460-initramfs-kernel.bin
-rw-r--r--. 1 mangix mangix 10354688 Oct 10 20:04 openwrt-mpc85xx-p1020-hpe_msm460-squashfs-factory.bin
-rw-r--r--. 1 mangix mangix  9145115 Oct 10 20:04 openwrt-mpc85xx-p1020-hpe_msm460-squashfs-sysupgrade.bin
-rw-r--r--. 1 mangix mangix 12410680 Oct 10 20:04 openwrt-mpc85xx-p1020-ocedo_panda-initramfs-kernel.bin
-rw-r--r--. 1 mangix mangix    11463 Oct 10 20:04 openwrt-mpc85xx-p1020-ocedo_panda-squashfs-fdt.bin
-rw-r--r--. 1 mangix mangix 10507316 Oct 10 20:04 openwrt-mpc85xx-p1020-ocedo_panda-squashfs-sysupgrade.bin
drwxr-xr-x. 1 mangix mangix     6988 Oct 10 20:04 packages/
-rw-r--r--. 1 mangix mangix     5186 Oct 10 20:04 profiles.json
-rw-r--r--. 1 mangix mangix     1960 Oct 10 20:04 sha256sums
-rw-r--r--. 1 mangix mangix       23 Oct 10 20:03 version.buildinfo
```

that's for p1020. symbols present in p1010 but not p1020 are

```
CONFIG_BR200_WP=y
CONFIG_FIREBOX_T10=y
# CONFIG_FSL_CORENET_CF is not set
CONFIG_FSL_IFC=y
CONFIG_MEMORY=y
CONFIG_MTD_NAND_FSL_IFC=y
CONFIG_MTD_SPLIT_TPLINK_FW=y
CONFIG_NEED_DMA_MAP_STATE=y
CONFIG_NET_DSA_QCA8K_LEDS_SUPPORT=y
CONFIG_NET_DSA_QCA8K=y
CONFIG_NET_DSA_TAG_QCA=y
CONFIG_QCA83XX_PHY=y
CONFIG_REALTEK_PHY=y
CONFIG_RED_15W_REV1=y
CONFIG_SWIOTLB=y
CONFIG_TL_WDR4900_V1=y
CONFIG_WS_AP3715I=y
```

Shine-, 2024-10-18:
P1020 has the same issue, so comparing P1010 against P1020 is pointless. Likely the whole mpc85xx target has the issue. Your oversized P1010 images are probably built with "fit none" instead of "fit gzip" from testing last night, that's why the final size is >20MB. The normal, compressed P1010 images are the size you'd expect (around 8MB). Zero-padding compresses well, after all ;-)

neheb, 2024-10-18:
I've made no such modifications.

CHKDSK88, 2024-10-18:
Sometime ago addresses was moved because kernel overlap:
https://github.com/openwrt/openwrt/commit/6a8b831593895ff9c8a687de7f7d0d24cdef8b3f

Maybe it's time to change address from 0x1500000 to 0x2000000.

neheb, 2024-10-18:
Should migrate to libdeflate-gzip as well.

Shine-, 2024-10-19:
@CHKDSK88 
This T10 has a hard limit of 16 MB for the extracted or uncompressed kernel image. So you can change the load address as much as you want (I tried!), uBoot always craps out at 16MB.

That's where the zImage chainloader linked above comes into play. And its load address is already **0x3000000**, so plenty of room for future growth.

Therefore the patch I pasted above could "fix" the problem.

Nevertheless, **why** is the mpc85xx kernel suddenly 17MB large, with >8MB being just zero-padding? I doubt this is how it should be!

robimarko, 2024-10-19:
I dont think that doubling the size is normal, there is usually always some increase but not double.

Also, are you sure this happened with 6.6 and not 6.1?

Shine-, 2024-10-19:
@robimarko 

Touché, I've never built 6.1. Even 6.6, I built for the first time when I was asked by @neheb to test the NVMEM patch. So I wouldn't know whether 6.1 or 6.6 introduced the size increase.

However, the load address change linked by @CHKDSK88 is from early March, ie. before the switch to kernel 6.6 - which would be an indication that the growth already happened with 6.1.

robimarko, 2024-10-19:
Yeah, 6.1 is where the kernel doubled which makes this a pain to debug.

I honestly dont see what can double the kernel size

Shine-, 2024-10-19:
Well, it's definitely not any **data**, but pure zero padding to exactly the 16MB boundary (see my post above). I don't know how a vmlinux binary is built, are there some linking options for page alignment, where defaults may have changed e.g. with a gcc update?

Shine-, 2024-10-19:
Also what comes to my mind (hello Captain Obvious...), it was @blocktrron who introduced the zImage loader, and I'm sure he did it for a reason. Perhaps he responds if we ping him asking why he's gone to the lengths of creating a chainloader, instead of just trying to reduce or remove the zero-padding.

blocktrron, 2024-10-20:
Hey @Shine- ,

Kernel size doubled due to https://github.com/torvalds/linux/commit/fdacae8a84024474afff234bdd1dbe19ad597a10 upstream. Either it was the enablement of this option or the kernel extending over the page boundary.

Personally, I'm fine with disabling the option, although it does impede security.

Best
David

Shine-, 2024-10-20:
Thanks a lot for clarifying!

Due to my lack of understanding of how kernel options are inherited/overridden, I failed trying to unset the option for a test build, it kept getting set back to "y". However, the kernel commit is rather clear in what it says: Use these 8M pages, or fall back to 512k pages with a performance hit.

For devices with large memory, like the T10, there's no question about it, at least in my uninformed opinion: Leave it as is and work around it, e.g. with your zImage loader.

What about smaller memory devices, like the TL-WDR4900v1? Accept the performance hit, or lessen security, or accept the higher memory usage? I wouldn't want to decide...

blocktrron, 2024-10-21:
Not quite, the non-executable pages are always 8MB aligned, see the description of the selected option:

```
config STRICT_KERNEL_RWX
	bool "Make kernel text and rodata read-only" if ARCH_OPTIONAL_KERNEL_RWX
	depends on ARCH_HAS_STRICT_KERNEL_RWX
	default !ARCH_OPTIONAL_KERNEL_RWX || ARCH_OPTIONAL_KERNEL_RWX_DEFAULT
	help
	  If this is set, kernel text and rodata memory will be made read-only,
	  and non-text memory will be made non-executable. This provides
	  protection against certain security exploits (e.g. executing the heap
	  or modifying text)

	  These features are considered standard security practice these days.
	  You should say Y here in almost all cases.
```

Without the option in question set, the  pages are not split and the complete memory range is marked executable / writable, so the kernel becomes vulnerable to specific exploits.

You can disable this by settings

```
# CONFIG_STRICT_KERNEL_RWX is not set
```

in the kernel configuration.

As I've said, the option was not active in the past, so we are only losing security compared to a release where it was activated.

I don't think it impacts the usability of the system (albeit still providing less usable memory) even on a WDR4900, the biggest issue is available allocation for decompression.

Shine-, 2024-10-21:
Thanks again for clarifying!

Using the zImage loader works fine on my T10, and it's reported to also work fine on the T30 (see the in-progress PR). Though I have both, a T10(-W) and a WDR4900, I'm currently using neither of them in production, so I wouldn't be able to report any changed usability that may be caused by the kernel doubling (save for the unbootability).