openwrt/openwrt #16732: Hysteria 2 can't cimpile

apollogeek, 2024-10-19:
### Describe the bug

Hello dear team.
I want to compile Hysteria 2 and Xray-core with 23.05.5 version but it show this error:
make[3]: *** [Makefile:62: /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/hysteria-app-v2.5.2/.built] Error 1
make[3]: Leaving directory '/workdir/openwrt/feeds/passwall_packages/hysteria'
time: package/feeds/passwall_packages/hysteria/compile#0.14#0.10#0.23
    ERROR: package/feeds/passwall_packages/hysteria failed to build.
make[2]: *** [package/Makefile:129: package/feeds/passwall_packages/hysteria/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:123: /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:233: world] Error 2

### OpenWrt version

23.05.5

### OpenWrt release

stable

### OpenWrt target/subtarget

Mips ARM

### Device

Xiaomi Redmi AX6000

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2024-10-19:
Do: make -j1 V=s

apollogeek, 2024-10-19:
> Do: make -j1 V=s and you will get more information about the error.

I know about it, but it's problem by openwrt firmware

apollogeek, 2024-10-20:
/home/apollo/Desktop/openwrt-23.05.5/staging_dir/target-aarch64_cortex-a53_musl/usr/share/gocode/src does not exist, skipping symlinks

touch /home/apollo/Desktop/openwrt-23.05.5/build_dir/target-aarch64_cortex-a53_musl/hysteria-app-v2.5.2/.configured_68b329da9893e34099c7d8ad5cb9c940
rm -f /home/apollo/Desktop/openwrt-23.05.5/build_dir/target-aarch64_cortex-a53_musl/hysteria-app-v2.5.2/.built
touch /home/apollo/Desktop/openwrt-23.05.5/build_dir/target-aarch64_cortex-a53_musl/hysteria-app-v2.5.2/.built_check
CONFIG_GOLANG_MOD_CACHE_WORLD_READABLE="" GO_BUILD_CACHE_DIR="/home/apollo/Desktop/openwrt-23.05.5/tmp/go-build" GO_MOD_CACHE_DIR="/home/apollo/Desktop/openwrt-23.05.5/dl/go-mod-cache" GO_MOD_ARGS="-modcacherw" GO_PKG="github.com/apernet/hysteria" GO_INSTALL_EXTRA="" GO_INSTALL_ALL="" GO_SOURCE_ONLY="" GO_BUILD_PKG="github.com/apernet/hysteria/app/v2" GO_EXCLUDES="" GO_GO_GENERATE="" GO_INSTALL_BIN_PATH="/usr/bin" BUILD_DIR="/home/apollo/Desktop/openwrt-23.05.5/build_dir/target-aarch64_cortex-a53_musl/hysteria-app-v2.5.2" GO_BUILD_DIR="/home/apollo/Desktop/openwrt-23.05.5/build_dir/target-aarch64_cortex-a53_musl/hysteria-app-v2.5.2/.go_work/build" GO_BUILD_BIN_DIR="/home/apollo/Desktop/openwrt-23.05.5/build_dir/target-aarch64_cortex-a53_musl/hysteria-app-v2.5.2/.go_work/build/bin/linux_arm64" GO_BUILD_DEPENDS_PATH="/usr/share/gocode" GO_BUILD_DEPENDS_SRC="/home/apollo/Desktop/openwrt-23.05.5/staging_dir/target-aarch64_cortex-a53_musl/usr/share/gocode/src" GOOS="linux" GOARCH="arm64" GO386="" GOAMD64="" GOARM="" GOMIPS="" GOMIPS64="" GOPPC64="" CGO_ENABLED=1 CC="aarch64-openwrt-linux-musl-gcc" CXX="aarch64-openwrt-linux-musl-g++" CGO_CFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/apollo/Desktop/openwrt-23.05.5/build_dir/target-aarch64_cortex-a53_musl/hysteria-app-v2.5.2=hysteria-app-v2.5.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_CPPFLAGS="-I/home/apollo/Desktop/openwrt-23.05.5/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/usr/include -I/home/apollo/Desktop/openwrt-23.05.5/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/include/fortify -I/home/apollo/Desktop/openwrt-23.05.5/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/include" CGO_CXXFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/apollo/Desktop/openwrt-23.05.5/build_dir/target-aarch64_cortex-a53_musl/hysteria-app-v2.5.2=hysteria-app-v2.5.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_LDFLAGS="-L/home/apollo/Desktop/openwrt-23.05.5/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/usr/lib -L/home/apollo/Desktop/openwrt-23.05.5/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/lib -znow -zrelro" GOPATH="/home/apollo/Desktop/openwrt-23.05.5/build_dir/target-aarch64_cortex-a53_musl/hysteria-app-v2.5.2/.go_work/build" GOCACHE="/home/apollo/Desktop/openwrt-23.05.5/tmp/go-build" GOMODCACHE="/home/apollo/Desktop/openwrt-23.05.5/dl/go-mod-cache" GOENV=off GOTOOLCHAIN=local /usr/bin/env bash /home/apollo/Desktop/openwrt-23.05.5/feeds/packages/lang/golang//golang-build.sh build -v -buildvcs=false -trimpath -ldflags "all=-buildid '1727094886' -linkmode external -extldflags '-L/home/apollo/Desktop/openwrt-23.05.5/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/usr/lib -L/home/apollo/Desktop/openwrt-23.05.5/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/lib -Wl,-z,now -Wl,-z,relro'"        -ldflags " -X github.com/apernet/hysteria/app/v2/cmd.appVersion=v2.5.2 -X github.com/apernet/hysteria/app/v2/cmd.appType=release -X github.com/apernet/hysteria/app/v2/cmd.appPlatform=linux -X github.com/apernet/hysteria/app/v2/cmd.appArch=arm64 -buildid '1727094886' -linkmode external -extldflags '-L/home/apollo/Desktop/openwrt-23.05.5/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/usr/lib -L/home/apollo/Desktop/openwrt-23.05.5/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/lib -Wl,-z,now -Wl,-z,relro'"  
Finding targets
go: ../../go.work requires go >= 1.22 (running go 1.21.13; GOTOOLCHAIN=local)

Building targets
go: ../../go.work requires go >= 1.22 (running go 1.21.13; GOTOOLCHAIN=local)

make[3]: *** [Makefile:62: /home/apollo/Desktop/openwrt-23.05.5/build_dir/target-aarch64_cortex-a53_musl/hysteria-app-v2.5.2/.built] Error 1
make[3]: Leaving directory '/home/apollo/Desktop/openwrt-23.05.5/feeds/passwall_packages/hysteria'
time: package/feeds/passwall_packages/hysteria/compile#0.53#0.30#45.19
    ERROR: package/feeds/passwall_packages/hysteria failed to build.
make[2]: *** [package/Makefile:129: package/feeds/passwall_packages/hysteria/compile] Error 1
make[2]: *** Waiting for unfinished jobs....
100 2972k  100 2972k    0     0  77356      0  0:00:39  0:00:39 --:--:--  163k          00    00::0010::0403    00::0000::4344    00::0000:16  148k:09  155k
touch /home/apollo/Desktop/openwrt-23.05.5/build_dir/target-aarch64_cortex-a53_musl/readline-8.2/.prepared_3d9418506351334c41075ff6e4924fce_6664517399ebbbc92a37c5bb081b5c53_check
. /home/apollo/Desktop/openwrt-23.05.5/include/shell.sh; /home/apollo/Desktop/openwrt-23.05.5/staging_dir/host/bin/libdeflate-gzip -dc /home/apollo/Desktop/openwrt-23.05.5/dl/readline-8.2.tar.gz | tar -C /home/apollo/Desktop/openwrt-23.05.5/build_dir/target-aarch64_cortex-a53_musl/readline-8.2/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /home/apollo/Desktop/openwrt-23.05.5/build_dir/target-aarch64_cortex-a53_musl/readline-8.2

apollogeek, 2024-10-20:
I think you need update go version to 0.47 or newer to get success compile, please take attention for this

closed 2024-10-24 by apollogeek