openwrt/openwrt #16735: The gods are seeking help

a834003831, 2024-10-19:
### Describe the bug

After using R4S to write OpenWRT, I inserted a USB network card, but it couldn't start properly. I need to unplug the USB network card to start up properly. The chip is RTL8153, so I am learning how to compile OpenWRT and hope that experts can give me some solutions to this problem

### OpenWrt version

6.1

### OpenWrt release

6.1

### OpenWrt target/subtarget

6.1

### Device

r4s

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-10-19:
You should ask in forum.
If the adapter works with likes of android it is supported by cdc_ether driver.
You can close your fake bug report as there is no actionable bug info included.

tapper82, 2024-10-19:
Hi can you take this to the forums pleas. Thanks.

namiltd, 2024-10-19:
Forum -> https://forum.openwrt.org