openwrt/openwrt #16736: DIR-2150 A1 stoped work as a switch after sysupgrade.

groszexxx, 2024-10-19:
### Describe the bug

My OpenWRT works as a switch. To LAN1 i have cable with internet from main router. LAN2 has cable to PC. 
I had this version, it works correctly: 4/28/2024 12:35 AM      7363651 [openwrt-ramips-mt7621-dlink_dir-2150-a1-squashfs-sysupgrade.bin](http://luckys.onmypc.net/openwrt/DIR-2150-A1/openwrt-ramips-mt7621-dlink_dir-2150-a1-squashfs-sysupgrade.bin)  from http://luckys.onmypc.net/openwrt/DIR-2150-A1/ 

After flash the newest sysupgrade openWRT has no internet. 

### OpenWrt version

6.6.56

### OpenWrt release

OpenWrt SNAPSHOT r27798-dbc2923cbe

### OpenWrt target/subtarget

6.6.56

### Device

DIR-2150 A1

### Image kind

Official downloaded image

### Steps to reproduce

# uci set network.lan.ipaddr=192.168.1.2
    # uci set network.lan.gateway=192.168.1.1
    # uci set network.lan.dns=192.168.1.1
    # uci set dhcp.lan.ignore='1'
    # uci commit
    # reboot
Put cable with internet to LAN1 

### Actual behaviour

Router has no internet.

### Expected behaviour

The router should have internet access.

### Additional info

root@OpenWrt:~# uci show network
network.loopback=interface
network.loopback.device='lo'
network.loopback.proto='static'
network.loopback.ipaddr='127.0.0.1'
network.loopback.netmask='255.0.0.0'
network.globals=globals
network.globals.ula_prefix='fdd3:e4db:eef0::/48'
network.globals.packet_steering='1'
network.@device[0]=device
network.@device[0].name='br-lan'
network.@device[0].type='bridge'
network.@device[0].ports='lan1' 'lan2' 'lan3' 'lan4'
network.lan=interface
network.lan.device='br-lan'
network.lan.proto='static'
network.lan.ipaddr='192.168.1.2'
network.lan.netmask='255.255.255.0'
network.lan.ip6assign='60'
network.lan.gateway='192.168.1.1'
network.lan.dns='192.168.1.1'
network.wan=interface
network.wan.device='wan'
network.wan.proto='dhcp'
network.wan6=interface
network.wan6.device='wan'
network.wan6.proto='dhcpv6'

root@OpenWrt:~# route -n
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         192.168.1.1     0.0.0.0         UG    0      0        0 br-lan
192.168.1.0     0.0.0.0         255.255.255.0   U     0      0        0 br-lan
root@OpenWrt:~#

root@OpenWrt:~# cat /etc/os-release
NAME="OpenWrt"
VERSION="SNAPSHOT"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt SNAPSHOT"
VERSION_ID="snapshot"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r27798-dbc2923cbe"
OPENWRT_BOARD="ramips/mt7621"
OPENWRT_ARCH="mipsel_24kc"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt SNAPSHOT r27798-dbc2923cbe"

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-10-19:
Save config backup as good as you can and try resetting, posted config looks like DSA already and should not have any problems.

groszexxx, 2024-10-19:
I have nothing to save. Old version works instantly after flash. When i flash the newone - router has no internet. I tried 4 sysupgrades files. The result is the same. When I back to old one version which i mentioned - everything start works instantly after flash.  What can i do or which informations should i provide to better explain the situation ?

groszexxx, 2024-10-19:
I downloaded it from https://firmware-selector.openwrt.org/?version=SNAPSHOT&target=ramips%2Fmt7621&id=dlink_dir-2150-a1 
previously i tried few versions from here: 
https://mirror-03.infra.openwrt.org/snapshots/targets/ramips/mt7621/

lexmark3200, 2024-10-19:
Try to install new snapshot without saving the config. Just fresh install, after restart try to upload the config backup.

groszexxx, 2024-10-20:
I collected informations from different scenarios: 
1) Ver 6.1.86 which works ok (as Dummy AP)
3) Newest snapshot as Dummy AP 
4) Newest snapshot internet cable in WAN

Scneario 1) Fresh 6.1.86

uci set network.lan.ipaddr=192.168.1.2
uci set network.lan.gateway=192.168.1.1
uci set network.lan.dns=192.168.1.1
uci set dhcp.lan.ignore='1'
uci commit
reboot

root@OpenWrt:/# ping 8.8.8.8
PING 8.8.8.8 (8.8.8.8): 56 data bytes
[   36.794988] mt7530-mdio mdio-bus:1f lan1: Link is Up - 100Mbps/Full - flow control off
[   36.810871] br-lan: port 1(lan1) entered blocking state
[   36.821307] br-lan: port 1(lan1) entered forwarding state
64 bytes from 8.8.8.8: seq=0 ttl=55 time=1303.710 ms
64 bytes from 8.8.8.8: seq=1 ttl=55 time=314.367 ms
64 bytes from 8.8.8.8: seq=2 ttl=55 time=18.051 ms

root@OpenWrt:/# uci show network
network.loopback=interface
network.loopback.device='lo'
network.loopback.proto='static'
network.loopback.ipaddr='127.0.0.1'
network.loopback.netmask='255.0.0.0'
network.globals=globals
network.globals.ula_prefix='fd18:1918:7fcf::/48'
network.globals.packet_steering='1'
network.@device[0]=device
network.@device[0].name='br-lan'
network.@device[0].type='bridge'
network.@device[0].ports='lan1' 'lan2' 'lan3' 'lan4'
network.lan=interface
network.lan.device='br-lan'
network.lan.proto='static'
network.lan.ipaddr='192.168.1.2'
network.lan.netmask='255.255.255.0'
network.lan.ip6assign='60'
network.lan.gateway='192.168.1.1'
network.lan.dns='192.168.1.1'
network.wan=interface
network.wan.device='wan'
network.wan.proto='dhcp'
network.wan6=interface
network.wan6.device='wan'
network.wan6.proto='dhcpv6'
root@OpenWrt:/#

root@OpenWrt:/# route -n
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         192.168.1.1     0.0.0.0         UG    0      0        0 br-lan
192.168.1.0     0.0.0.0         255.255.255.0   U     0      0        0 br-lan

root@OpenWrt:/# ifconfig
br-lan    Link encap:Ethernet  HWaddr 34:0A:33:7D:FA:6C
          inet addr:192.168.1.2  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::360a:33ff:fe7d:fa6c/64 Scope:Link
          inet6 addr: fd18:1918:7fcf::1/60 Scope:Global
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:664 errors:0 dropped:29 overruns:0 frame:0
          TX packets:422 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:91917 (89.7 KiB)  TX bytes:35169 (34.3 KiB)

eth0      Link encap:Ethernet  HWaddr 34:0A:33:7D:FA:6C
[... 543 lines cut ...]
Sat Oct 19 12:14:29 2024 daemon.err odhcpd[1765]: Failed to send to fe80::be72:fdec:21aa:1a62%lan@br-lan (Address not available)
Sat Oct 19 12:14:29 2024 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 0 names
Sat Oct 19 12:14:29 2024 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c.2131 - 0 names
Sat Oct 19 12:14:29 2024 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 0 names
Sat Oct 19 12:14:30 2024 daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP no-DHCPv6 no-Lua TFTP no-conntrack no-ipset no-nftset no-auth no-cryptohash no-DNSSEC no-ID loop-detect inotify dumpfile
Sat Oct 19 12:14:30 2024 daemon.info dnsmasq-dhcp[1]: DHCP, IP range 192.168.1.100 -- 192.168.1.249, lease time 12h
Sat Oct 19 12:14:30 2024 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 4 names
Sat Oct 19 12:14:30 2024 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 0 names
Sat Oct 19 12:14:30 2024 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Sat Oct 19 12:14:31 2024 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 4 names
Sat Oct 19 12:14:31 2024 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c.2437 - 0 names
Sat Oct 19 12:14:31 2024 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 2 names
Sat Oct 19 12:14:31 2024 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Sat Oct 19 12:14:31 2024 daemon.notice netifd: wan (2292): udhcpc: broadcasting discover
Sat Oct 19 12:14:31 2024 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 4 names
Sat Oct 19 12:14:31 2024 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 2 names
Sat Oct 19 12:14:31 2024 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Sat Oct 19 12:14:32 2024 daemon.warn odhcpd[1765]: No default route present, overriding ra_lifetime to 0!
Sat Oct 19 12:14:34 2024 daemon.notice netifd: wan (2292): udhcpc: broadcasting discover
Sat Oct 19 12:14:36 2024 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) 94:c6:91:59:bd:dc
Sat Oct 19 12:14:36 2024 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.1.155 94:c6:91:59:bd:dc
Sat Oct 19 12:14:36 2024 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.1.155 94:c6:91:59:bd:dc
Sat Oct 19 12:14:36 2024 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.1.155 94:c6:91:59:bd:dc DESKTOP-90KE4MS
Sat Oct 19 12:14:36 2024 daemon.warn odhcpd[1765]: No default route present, overriding ra_lifetime to 0!
Sat Oct 19 12:14:45 2024 daemon.warn odhcpd[1765]: No default route present, overriding ra_lifetime to 0!

groszexxx, 2024-10-20:
I would like to add that regarding the hypotheses about the settings, I tried uploading working settings from the functioning version to the one with issues, but without any results. I saved the settings from the non-working version and imported them into the version that was fine – it remained fine. I think this completely contradicts your initial suspicions that the problem lies in some settings, which a factory reset would help with.

groszexxx, 2024-10-22:
False alarm. 
Ashamed situation. 
Cable has broken one wire. Probably my cat has done it. 
Now everything is ok.

closed 2024-10-22 by groszexxx