openwrt/openwrt #16742: Loss of Static Route After Addition

bsdcpp, 2024-10-20:
### Describe the bug

I added 3 static routes like below:
```bash
config route
	option interface 'wan1'
	option target '20.254.1.0/24'

config route
	option interface 'wan1'
	option target '20.253.1.0/24'

config route
	option interface 'wan1'
	option target '10.223.3.0/24'
```

but with command `ip r` got nothing about these just added.

### OpenWrt version

r28061-399f9a1db3

### OpenWrt release

23.05.4

### OpenWrt target/subtarget

x86/64

### Device

INTEL Corporation Q3XXG4-P

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-10-21:
Your OpenWRT version tags do not match up. In general "main" routing table will be automatically selected but might be overriden by multi-wan, pbr and the likes.

bsdcpp, 2024-10-23:
@brada4 
Thank you. I’m currently using luci_mwan3 to achieve multi-WAN policy routing, but it’s based on fw3. I'm unsure if there will be any issues when mixing it with fw4.

brada4, 2024-10-23:
First specify routing table for routes as auto-selection did not work. Likely pbr has better place for them.