openwrt/openwrt #16751: do_page_fault(): sending SIGSEGV to pidof for invalid read access

guo652917087, 2024-10-21:
### Describe the bug

root@Gateway:~# [138846.005947] do_page_fault(): sending SIGSEGV to pidof for invalid read access from 00000ea1
[138846.014883] epc = 77e1f034 in libc.so[77d98000+b2000]
[138846.020327] ra  = 77e1f3f8 in libc.so[77d98000+b2000]
[159958.445507] do_page_fault(): sending SIGSEGV to ps for invalid read access from 00213a90
[159958.454269] epc = 77e24928 in libc.so[77e00000+b2000]
[159958.459703] ra  = 0043b69b in busybox[400000+55000]
[164042.845228] do_page_fault(): sending SIGSEGV to ps for invalid read access from 00239590
[164042.853955] epc = 77d72928 in libc.so[77d4e000+b2000]
[164042.859323] ra  = 0043b69b in busybox[400000+55000]
[189595.026300] do_page_fault(): sending SIGSEGV to ps for invalid read access from 00291708
[189595.035016] epc = 77daa928 in libc.so[77d86000+b2000]
[189595.040400] ra  = 0043b69b in busybox[400000+55000]
[230009.100057] do_page_fault(): sending SIGSEGV to ps for invalid read access from 0031bdc8
[230009.108799] epc = 77db0928 in libc.so[77d8c000+b2000]
[230009.114181] ra  = 0043b69b in busybox[400000+55000]
[233220.952243] do_page_fault(): sending SIGSEGV to ps for invalid read access from 0032e9b0
[233220.960972] epc = 77e00928 in libc.so[77ddc000+b2000]
[233220.966472] ra  = 0043b69b in busybox[400000+55000]
[277206.600910] do_page_fault(): sending SIGSEGV to ps for invalid read access from 003bf59c
[277206.609548] epc = 77daa928 in libc.so[77d86000+b2000]
[277206.615010] ra  = 0043b69b in busybox[400000+55000]
[318395.018967] do_page_fault(): sending SIGSEGV to pidof for invalid read access from 00000787
[318395.028228] epc = 77e9b034 in libc.so[77e14000+b2000]
[318395.033629] ra  = 77e9b3f8 in libc.so[77e14000+b2000]
[336604.529255] do_page_fault(): sending SIGSEGV to ps for invalid read access from 004887c0
[336604.537921] epc = 77d62928 in libc.so[77d3e000+b2000]
[336604.543382] ra  = 0043b69b in busybox[400000+55000]

### OpenWrt version

r27578-23ac1ad951

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ath79/generic

### Device

GL-X300B

### Image kind

Self-built image

### Steps to reproduce

run a shell

```
#!/bin/sh
tf_log="/tmp/usb_mount/gateway/gateway_log/user_log.txt"
flash_log="/root/user_log.txt"
now=$(date +"%Y-%m-%d %H:%M:%S")

        pid=0
        # kill active clients
        pid_mine="$$"
        pids=$(ps -w| grep "cro_check.sh" | grep -v "grep" | awk '{print $1}')
        for pid in $pids; do
                [ "$pid" = "$pid_mine" ] && continue
                [ -e "/proc/$pid/stat" ] && kill $pid
        done

while [ true ]
[... 550 lines cut ...]
  43bb3c:	722d      	cmpi	v0,45
  43bb3e:	61c9      	btnez	43bad2 <procps_scan+0x476>
  43bb40:	9855      	lw	v0,84(s0)
  43bb42:	d123      	sw	s1,140(sp)
  43bb44:	e229      	addu	v0,s1
  43bb46:	d855      	sw	v0,84(s0)
  43bb48:	17c4      	b	43bad2 <procps_scan+0x476>
  43bb4a:	c865      	sh	v1,10(s0)
  43bb4c:	1a00 2c29 	jal	40b0a4 <xstrdup>
  43bb50:	0432      	addiu	a0,sp,200
  43bb52:	1717      	b	43b982 <procps_scan+0x326>

0043bb54 <__pool_procps_scan_2903>:
  43bb54:	0044 2328 0040 0000 0047 0580 0044 6ba4     .D#(.@...G...Dk.
  43bb64:	0044 6bb4 000c 7e2e 0044 08b0 000f ff00     .Dk...~..D......
  43bb74:	0044 6bd4 0044 6c3c 0001 0080 0044 5dc0     .Dk..Dl<.....D].
  43bb84:	0001 0000 0044 6bc8 0044 6be4 0044 6bf4     .....Dk..Dk..Dk.
  43bb94:	0044 6c04 0044 6c14 0044 11fc 0044 6c24     .Dl..Dl..D...Dl$
  43bba4:	0044 6c30                                   .Dl0

```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

neheb, 2024-10-22:
ping @981213

DragonBluep, 2025-04-27:
It seems that you enabled `BUSYBOX_CONFIG_FEATURE_SHOW_THREADS`? If you encounter this warning again, please attach the full `system` log.

```
struct dirent *readdir(DIR *dir)
{
   24908:	3c1c000b 	lui	gp,0xb
   2490c:	279c3988 	addiu	gp,gp,14728
   24910:	0399e021 	addu	gp,gp,t9
   24914:	27bdffd8 	addiu	sp,sp,-40
   24918:	afb0001c 	sw	s0,28(sp)
   2491c:	afbc0010 	sw	gp,16(sp)
   24920:	afbf0024 	sw	ra,36(sp)
   24924:	afb10020 	sw	s1,32(sp)
	struct dirent *de;
	
	if (dir->buf_pos >= dir->buf_end) {
   24928:	8c87000c 	lw	a3,12(a0) <-- crashed here in musl
   2492c:	8c830010 	lw	v1,16(a0)
   24930:	00e3182a 	slt	v1,a3,v1
   24934:	10600010 	beqz	v1,24978 <readdir+0x70>
   24938:	00808025 	move	s0,a0
```

MBPotier, 2025-04-27:
Here is a few sections of the system log from around when the issue happens.
```
Sat Apr 26 10:16:48 2025 daemon.notice netifd: wan (28260): udhcpc: sending renew to server 69.252.95.198
Sat Apr 26 10:16:48 2025 daemon.notice netifd: wan (28260): udhcpc: lease of $PUBLIC_IP obtained from 69.252.95.198, lease time 7200
Sat Apr 26 10:16:52 2025 user.notice firewall: Reloading firewall due to ifupdate of wan (eth1)
Sat Apr 26 10:16:52 2025 daemon.info procd: Instance https-dns-proxy::instance1 s in a crash loop 8 crashes, 1913 seconds since last crash
Sat Apr 26 10:16:55 2025 user.notice https-dns-proxy [29663]: Starting https-dns-proxy 2025.03.13-r1 instances on_interface_trigger ✓
Sat Apr 26 10:16:56 2025 user.notice https-dns-proxy [29663]: Setting trigger for wan ✓
Sat Apr 26 10:18:56 2025 kern.info kernel: [ 5479.109560] do_page_fault(): sending SIGSEGV to pgrep for invalid read access from 0000000c
Sat Apr 26 10:18:56 2025 kern.info kernel: [ 5479.118155] epc = 77db6928 in libc.so[77d92000+b2000]
Sat Apr 26 10:18:56 2025 kern.info kernel: [ 5479.123441] ra  = 00437ab9 in busybox[400000+4f000]
Sat Apr 26 10:18:57 2025 daemon.err uhttpd[2381]: Segmentation fault
```

```
Sat Apr 26 12:04:59 2025 kern.info kernel: [11841.838128] do_page_fault(): sending SIGSEGV to pgrep for invalid read access from 0000000c
Sat Apr 26 12:04:59 2025 kern.info kernel: [11841.846952] epc = 77dd0928 in libc.so[77dac000+b2000]
Sat Apr 26 12:04:59 2025 kern.info kernel: [11841.852272] ra  = 00437ab9 in busybox[400000+4f000]
Sat Apr 26 12:16:30 2025 daemon.err uhttpd[2381]: cat: can't open '/proc/11270/cmdline': No such file or directory
Sat Apr 26 12:16:49 2025 daemon.notice netifd: wan (28260): udhcpc: sending renew to server 69.252.95.198
Sat Apr 26 12:16:49 2025 daemon.notice netifd: wan (28260): udhcpc: lease of $PUBLIC_IP obtained from 69.252.95.198, lease time 3599
Sat Apr 26 12:16:55 2025 user.notice firewall: Reloading firewall due to ifupdate of wan (eth1)
Sat Apr 26 12:16:56 2025 user.notice https-dns-proxy [15005]: Starting https-dns-proxy 2025.03.13-r1 instances on_interface_trigger ✓
Sat Apr 26 12:16:57 2025 user.notice https-dns-proxy [15005]: Setting trigger for wan ✓
Sat Apr 26 12:20:11 2025 daemon.err uhttpd[2381]: cat: can't open '/proc/23241/cmdline': No such file or directory
Sat Apr 26 12:23:13 2025 kern.info kernel: [12936.030849] do_page_fault(): sending SIGSEGV to pgrep for invalid read access from 0000000c
Sat Apr 26 12:23:13 2025 kern.info kernel: [12936.039499] epc = 77e48928 in libc.so[77e24000+b2000]
Sat Apr 26 12:23:13 2025 kern.info kernel: [12936.044703] ra  = 00437ab9 in busybox[400000+4f000]
Sat Apr 26 12:41:59 2025 kern.info kernel: [14061.323667] do_page_fault(): sending SIGSEGV to pgrep for invalid read access from 0000000c
Sat Apr 26 12:41:59 2025 kern.info kernel: [14061.332314] epc = 77d7c928 in libc.so[77d58000+b2000]
Sat Apr 26 12:41:59 2025 kern.info kernel: [14061.337589] ra  = 00437ab9 in busybox[400000+4f000]
Sat Apr 26 12:46:50 2025 daemon.notice netifd: wan (28260): udhcpc: sending renew to server 69.252.95.198
Sat Apr 26 12:46:50 2025 daemon.notice netifd: wan (28260): udhcpc: lease of $PUBLIC_IP obtained from 69.252.95.198, lease time 7200
Sat Apr 26 12:46:55 2025 user.notice firewall: Reloading firewall due to ifupdate of wan (eth1)
Sat Apr 26 12:46:56 2025 user.notice https-dns-proxy [10481]: Starting https-dns-proxy 2025.03.13-r1 instances on_interface_trigger ✓
Sat Apr 26 12:46:56 2025 user.notice https-dns-proxy [10481]: Setting trigger for wan ✓
```

```
Sat Apr 26 16:32:02 2025 kern.info kernel: [27864.534213] do_page_fault(): sending SIGSEGV to pgrep for invalid read access from 0000000c
Sat Apr 26 16:32:02 2025 kern.info kernel: [27864.543269] epc = 77d6e928 in libc.so[77d4a000+b2000]
Sat Apr 26 16:32:02 2025 kern.info kernel: [27864.548754] ra  = 00437ab9 in busybox[400000+4f000]
Sat Apr 26 16:32:39 2025 daemon.err uhttpd[2381]: cat: read error: No such process
```