openwrt/openwrt #16760: Can't compile 22.03.7

apollogeek, 2024-10-23:
Here is log:

install -d -m0755 /home/apollo/Desktop/openwrt-22.03.7/tmp/stage-curl/host/bin /home/apollo/Desktop/openwrt-22.03.7/tmp/stage-curl/usr/bin /home/apollo/Desktop/openwrt-22.03.7/tmp/stage-curl/usr/include /home/apollo/Desktop/openwrt-22.03.7/tmp/stage-curl/usr/lib /home/apollo/Desktop/openwrt-22.03.7/tmp/stage-curl/usr/lib/pkgconfig
install -m0755 /home/apollo/Desktop/openwrt-22.03.7/build_dir/target-mipsel_24kc_musl/curl-8.6.0/ipkg-install/usr/bin/curl-config /home/apollo/Desktop/openwrt-22.03.7/tmp/stage-curl/usr/bin/
cp -fpR /home/apollo/Desktop/openwrt-22.03.7/build_dir/target-mipsel_24kc_musl/curl-8.6.0/ipkg-install/usr/include/curl /home/apollo/Desktop/openwrt-22.03.7/tmp/stage-curl/usr/include/
cp -fpR /home/apollo/Desktop/openwrt-22.03.7/build_dir/target-mipsel_24kc_musl/curl-8.6.0/ipkg-install/usr/lib/libcurl.{a,so*} /home/apollo/Desktop/openwrt-22.03.7/tmp/stage-curl/usr/lib/
cp -fpR /home/apollo/Desktop/openwrt-22.03.7/build_dir/target-mipsel_24kc_musl/curl-8.6.0/libcurl.pc /home/apollo/Desktop/openwrt-22.03.7/tmp/stage-curl/usr/lib/pkgconfig/
/home/apollo/Desktop/openwrt-22.03.7/staging_dir/host/bin/sed -i -e 's,^\(prefix\|exec_prefix\)=.*,\1=/home/apollo/Desktop/openwrt-22.03.7/staging_dir/target-mipsel_24kc_musl/usr,g' /home/apollo/Desktop/openwrt-22.03.7/tmp/stage-curl/usr/bin/curl-config
[ -n "-L/home/apollo/Desktop/openwrt-22.03.7/staging_dir/toolchain-mipsel_24kc_gcc-11.2.0_musl/usr/lib -L/home/apollo/Desktop/openwrt-22.03.7/staging_dir/toolchain-mipsel_24kc_gcc-11.2.0_musl/lib -L/home/apollo/Desktop/openwrt-22.03.7/staging_dir/target-mipsel_24kc_musl/usr/lib/libiconv-stub/lib -Wl,-rpath-link=/home/apollo/Desktop/openwrt-22.03.7/staging_dir/target-mipsel_24kc_musl/usr/lib/libiconv-stub/lib -L/home/apollo/Desktop/openwrt-22.03.7/staging_dir/target-mipsel_24kc_musl/usr/lib/libintl-stub/lib -Wl,-rpath-link=/home/apollo/Desktop/openwrt-22.03.7/staging_dir/target-mipsel_24kc_musl/usr/lib/libintl-stub/lib -znow -zrelro -Wl,--gc-sections" ] && /home/apollo/Desktop/openwrt-22.03.7/staging_dir/host/bin/sed -i -e 's#-L/home/apollo/Desktop/openwrt-22.03.7/staging_dir/toolchain-mipsel_24kc_gcc-11.2.0_musl/usr/lib -L/home/apollo/Desktop/openwrt-22.03.7/staging_dir/toolchain-mipsel_24kc_gcc-11.2.0_musl/lib -L/home/apollo/Desktop/openwrt-22.03.7/staging_dir/target-mipsel_24kc_musl/usr/lib/libiconv-stub/lib -Wl,-rpath-link=/home/apollo/Desktop/openwrt-22.03.7/staging_dir/target-mipsel_24kc_musl/usr/lib/libiconv-stub/lib -L/home/apollo/Desktop/openwrt-22.03.7/staging_dir/target-mipsel_24kc_musl/usr/lib/libintl-stub/lib -Wl,-rpath-link=/home/apollo/Desktop/openwrt-22.03.7/staging_dir/target-mipsel_24kc_musl/usr/lib/libintl-stub/lib -znow -zrelro -Wl,--gc-sections##g' /home/apollo/Desktop/openwrt-22.03.7/tmp/stage-curl/usr/lib/pkgconfig/libcurl.pc || true
ln -sf /home/apollo/Desktop/openwrt-22.03.7/staging_dir/target-mipsel_24kc_musl/usr/bin/curl-config /home/apollo/Desktop/openwrt-22.03.7/tmp/stage-curl/host/bin/
find /home/apollo/Desktop/openwrt-22.03.7/tmp/stage-curl -name '*.la' | xargs -r rm -f; 
if [ -f /home/apollo/Desktop/openwrt-22.03.7/staging_dir/target-mipsel_24kc_musl/packages/curl.list ]; then /home/apollo/Desktop/openwrt-22.03.7/scripts/clean-package.sh "/home/apollo/Desktop/openwrt-22.03.7/staging_dir/target-mipsel_24kc_musl/packages/curl.list" "/home/apollo/Desktop/openwrt-22.03.7/staging_dir/target-mipsel_24kc_musl"; fi
if [ -d /home/apollo/Desktop/openwrt-22.03.7/tmp/stage-curl ]; then (cd /home/apollo/Desktop/openwrt-22.03.7/tmp/stage-curl; find ./ > /home/apollo/Desktop/openwrt-22.03.7/tmp/stage-curl.files);      SHELL= flock /home/apollo/Desktop/openwrt-22.03.7/tmp/.staging-dir.flock -c ' mv /home/apollo/Desktop/openwrt-22.03.7/tmp/stage-curl.files /home/apollo/Desktop/openwrt-22.03.7/staging_dir/target-mipsel_24kc_musl/packages/curl.list && cp -fpR /home/apollo/Desktop/openwrt-22.03.7/tmp/stage-curl/* /home/apollo/Desktop/openwrt-22.03.7/staging_dir/target-mipsel_24kc_musl/; '; fi
rm -rf /home/apollo/Desktop/openwrt-22.03.7/tmp/stage-curl
touch /home/apollo/Desktop/openwrt-22.03.7/staging_dir/target-mipsel_24kc_musl/stamp/.curl_installed
rm -rf /home/apollo/Desktop/openwrt-22.03.7/build_dir/target-mipsel_24kc_musl/curl-8.6.0/.pkgdir/curl.installed /home/apollo/Desktop/openwrt-22.03.7/build_dir/target-mipsel_24kc_musl/curl-8.6.0/.pkgdir/curl
mkdir -p /home/apollo/Desktop/openwrt-22.03.7/build_dir/target-mipsel_24kc_musl/curl-8.6.0/.pkgdir/curl
install -d -m0755 /home/apollo/Desktop/openwrt-22.03.7/build_dir/target-mipsel_24kc_musl/curl-8.6.0/.pkgdir/curl/usr/bin
install -m0755 /home/apollo/Desktop/openwrt-22.03.7/build_dir/target-mipsel_24kc_musl/curl-8.6.0/ipkg-install/usr/bin/curl /home/apollo/Desktop/openwrt-22.03.7/build_dir/target-mipsel_24kc_musl/curl-8.6.0/.pkgdir/curl/usr/bin/
touch /home/apollo/Desktop/openwrt-22.03.7/build_dir/target-mipsel_24kc_musl/curl-8.6.0/.pkgdir/curl.installed
mkdir -p /home/apollo/Desktop/openwrt-22.03.7/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp
SHELL= flock /home/apollo/Desktop/openwrt-22.03.7/tmp/.root-copy.flock -c 'cp -fpR /home/apollo/Desktop/openwrt-22.03.7/build_dir/target-mipsel_24kc_musl/curl-8.6.0/.pkgdir/curl/. /home/apollo/Desktop/openwrt-22.03.7/staging_dir/target-mipsel_24kc_musl/root-ramips/'
touch /home/apollo/Desktop/openwrt-22.03.7/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.curl_installed
echo "curl" >> /home/apollo/Desktop/openwrt-22.03.7/staging_dir/target-mipsel_24kc_musl/pkginfo/curl.default.install
make[3]: Leaving directory '/home/apollo/Desktop/openwrt-22.03.7/feeds/packages/net/curl'
time: package/feeds/packages/curl/compile#46.05#6.76#31.86
make[2]: Leaving directory '/home/apollo/Desktop/openwrt-22.03.7'
make[1]: *** [package/Makefile:114: /home/apollo/Desktop/openwrt-22.03.7/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/apollo/Desktop/openwrt-22.03.7'
make: *** [/home/apollo/Desktop/openwrt-22.03.7/include/toplevel.mk:230: world] Error 2

pkgadd, 2024-10-24:
openwrt-22.03 has been EOL since June, no changes are being done for it anymore. Please switch to openwrt-23.05 or main.

apollogeek, 2024-10-24:
> openwrt-22.03 has been EOL since June, no changes are being done for it anymore. Please switch to openwrt-23.05 or main.

I understand, but if I use 23.05 version for Xiaomi Redmi Ac2100 it loads and hangs, I installed Passwall+Xray-core and sometimes 1% is left in memory and the router reboots itself at this point

brada4, 2024-10-24:
Could you make a new issue respecting the template?

apollogeek, 2024-10-24:
> Could you make a new issue respecting the template?

Done, open new issue

brada4, 2024-10-24:
And close this  useless one?

closed 2024-10-24 by apollogeek