openwrt/openwrt #16762: MT7530 transmit queue timeout

yogo1212, 2024-10-23:
### Describe the bug

Today at about 9 UTC, my Cudy WR2100 stopped transmitting traffic on the wan port after about 2 days uptime (without any outside interference). Other ethernet ports also don't work.

### OpenWrt version

r27798-dbc2923cbe

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ramips/mt7621

### Device

Cudy WR2100

### Image kind

Self-built image

### Steps to reproduce

Unsure what led to this.
Now it happens on every reboot.

### Actual behaviour

No traffic. Error messages in dmesg. Ingress works (for instance, ARP table is filled).

```
[  913.080368] mtk_soc_eth 1e100000.ethernet wan: NETDEV WATCHDOG: CPU: 0: transmit queue 1 timed out 845010 ms
[  918.120365] mtk_soc_eth 1e100000.ethernet wan: NETDEV WATCHDOG: CPU: 0: transmit queue 1 timed out 850050 ms
[  923.080363] mtk_soc_eth 1e100000.ethernet wan: NETDEV WATCHDOG: CPU: 0: transmit queue 1 timed out 855010 ms
[  928.120419] mtk_soc_eth 1e100000.ethernet wan: NETDEV WATCHDOG: CPU: 0: transmit queue 1 timed out 860050 ms
[  933.080366] mtk_soc_eth 1e100000.ethernet wan: NETDEV WATCHDOG: CPU: 0: transmit queue 1 timed out 865010 ms
[  938.120422] mtk_soc_eth 1e100000.ethernet wan: NETDEV WATCHDOG: CPU: 0: transmit queue 1 timed out 870050 ms
[  943.080362] mtk_soc_eth 1e100000.ethernet wan: NETDEV WATCHDOG: CPU: 0: transmit queue 1 timed out 875010 ms
[  948.120367] mtk_soc_eth 1e100000.ethernet wan: NETDEV WATCHDOG: CPU: 0: transmit queue 1 timed out 880050 ms
```

After plugging the ethernet cable into a lan port, the message changes:

```
[ 1310.050338] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 2: transmit queue 6 timed out 5210 ms
[ 1315.000448] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 2: transmit queue 6 timed out 10160 ms
[ 1320.050331] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 2: transmit queue 6 timed out 15210 ms
[ 1325.000342] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 2: transmit queue 6 timed out 20160 ms
```

Unlike #14167, there is no stack trace.

### Expected behaviour

Normal egress :-)
For "cheerful, bright, and gay", it would also be nice if bugs didn't appear sporadically :-D

### Diffconfig

```text
CONFIG_TARGET_ramips=y
CONFIG_TARGET_ramips_mt7621=y
CONFIG_TARGET_MULTI_PROFILE=y
CONFIG_TARGET_DEVICE_ramips_mt7621_DEVICE_dual-q_h721=y
CONFIG_BUSYBOX_CUSTOM=y
CONFIG_BATMAN_ADV_BATMAN_V=y
CONFIG_BATMAN_ADV_BLA=y
CONFIG_BATMAN_ADV_DAT=y
CONFIG_BATMAN_ADV_MCAST=y
# CONFIG_BUSYBOX_CONFIG_IP is not set
CONFIG_BUSYBOX_CONFIG_NC_110_COMPAT=y
CONFIG_BUSYBOX_CONFIG_NC_SERVER=y
# CONFIG_BUSYBOX_CONFIG_SWAPOFF is not set
# CONFIG_BUSYBOX_CONFIG_SWAPON is not set
CONFIG_BUSYBOX_CONFIG_TIMEOUT=y
CONFIG_BUSYBOX_CONFIG_USLEEP=y
CONFIG_FASTD_ENABLE_CIPHER_NULL=y
CONFIG_FASTD_ENABLE_CIPHER_SALSA2012=y
CONFIG_FASTD_ENABLE_MAC_UHASH=y
CONFIG_FASTD_ENABLE_METHOD_COMPOSED_UMAC=y
CONFIG_FASTD_ENABLE_METHOD_GENERIC_UMAC=y
CONFIG_FASTD_ENABLE_METHOD_NULL=y
CONFIG_FASTD_ENABLE_METHOD_NULL_L2TP=y
CONFIG_FASTD_WITH_OFFLOAD_L2TP=y
CONFIG_FASTD_WITH_STATUS_SOCKET=y
CONFIG_IMAGEOPT=y
CONFIG_OPENSSL_ENGINE=y
CONFIG_OPENSSL_PREFER_CHACHA_OVER_GCM=y
CONFIG_OPENSSL_WITH_ASM=y
CONFIG_OPENSSL_WITH_CHACHA_POLY1305=y
CONFIG_OPENSSL_WITH_CMS=y
CONFIG_OPENSSL_WITH_DEPRECATED=y
CONFIG_OPENSSL_WITH_ERROR_MESSAGES=y
CONFIG_OPENSSL_WITH_IDEA=y
CONFIG_OPENSSL_WITH_MDC2=y
CONFIG_OPENSSL_WITH_PSK=y
CONFIG_OPENSSL_WITH_SEED=y
CONFIG_OPENSSL_WITH_SRP=y
CONFIG_OPENSSL_WITH_TLS13=y
CONFIG_OPENSSL_WITH_WHIRLPOOL=y
CONFIG_PACKAGE_batctl-default=y
CONFIG_PACKAGE_block-mount=y
CONFIG_PACKAGE_bridge_mac_table=y
CONFIG_PACKAGE_fastd=y
# CONFIG_PACKAGE_firewall4 is not set
CONFIG_PACKAGE_ip-full=y
CONFIG_PACKAGE_kmod-batman-adv=y
CONFIG_PACKAGE_kmod-crypto-kpp=y
CONFIG_PACKAGE_kmod-crypto-lib-chacha20=y
CONFIG_PACKAGE_kmod-crypto-lib-chacha20poly1305=y
CONFIG_PACKAGE_kmod-crypto-lib-curve25519=y
CONFIG_PACKAGE_kmod-crypto-lib-poly1305=y
CONFIG_PACKAGE_kmod-fs-vfat=y
CONFIG_PACKAGE_kmod-hid=y
CONFIG_PACKAGE_kmod-hid-generic=y
CONFIG_PACKAGE_kmod-input-core=y
CONFIG_PACKAGE_kmod-input-evdev=y
CONFIG_PACKAGE_kmod-l2tp=y
CONFIG_PACKAGE_kmod-l2tp-eth=y
# CONFIG_PACKAGE_kmod-lib-crc-ccitt is not set
CONFIG_PACKAGE_kmod-lib-crc16=y
CONFIG_PACKAGE_kmod-mii=y
# CONFIG_PACKAGE_kmod-nft-fib is not set
CONFIG_PACKAGE_kmod-nls-cp437=y
CONFIG_PACKAGE_kmod-nls-iso8859-1=y
CONFIG_PACKAGE_kmod-nls-utf8=y
# CONFIG_PACKAGE_kmod-ppp is not set
CONFIG_PACKAGE_kmod-tun=y
CONFIG_PACKAGE_kmod-udptunnel4=y
CONFIG_PACKAGE_kmod-udptunnel6=y
CONFIG_PACKAGE_kmod-usb-hid=y
CONFIG_PACKAGE_kmod-usb-net=y
CONFIG_PACKAGE_kmod-usb-net-cdc-ether=y
CONFIG_PACKAGE_kmod-usb-net-cdc-mbim=y
CONFIG_PACKAGE_kmod-usb-net-cdc-ncm=y
CONFIG_PACKAGE_kmod-usb-net-qmi-wwan=y
CONFIG_PACKAGE_kmod-usb-ohci=y
CONFIG_PACKAGE_kmod-usb-serial=y
CONFIG_PACKAGE_kmod-usb-serial-option=y
CONFIG_PACKAGE_kmod-usb-serial-qualcomm=y
CONFIG_PACKAGE_kmod-usb-serial-wwan=y
CONFIG_PACKAGE_kmod-usb-storage=y
CONFIG_PACKAGE_kmod-usb-uhci=y
CONFIG_PACKAGE_kmod-usb-wdm=y
CONFIG_PACKAGE_kmod-veth=y
CONFIG_PACKAGE_kmod-wireguard=y
CONFIG_PACKAGE_libatomic=y
CONFIG_PACKAGE_libbase64=y
CONFIG_PACKAGE_libbpf=y
CONFIG_PACKAGE_libdnscache=y
CONFIG_PACKAGE_libelf=y
CONFIG_PACKAGE_libethmac=y
CONFIG_PACKAGE_libevent2=y
CONFIG_PACKAGE_libevent2-core=y
CONFIG_PACKAGE_libevent2-extra=y
CONFIG_PACKAGE_libevent2-openssl=y
CONFIG_PACKAGE_libevmqtt=y
CONFIG_PACKAGE_libevtsock=y
CONFIG_PACKAGE_libevtssl=y
CONFIG_PACKAGE_libevubus=y
CONFIG_PACKAGE_libncurses=y
CONFIG_PACKAGE_libopenssl=y
CONFIG_PACKAGE_libopenssl-conf=y
CONFIG_PACKAGE_libopenssl-legacy=y
CONFIG_PACKAGE_libpcap=y
CONFIG_PACKAGE_libpcre2=y
CONFIG_PACKAGE_libringbuffer=y
CONFIG_PACKAGE_librt=y
# CONFIG_PACKAGE_libuclient is not set
CONFIG_PACKAGE_libuecc=y
# CONFIG_PACKAGE_opkg is not set
# CONFIG_PACKAGE_ppp is not set
CONFIG_PACKAGE_resolveip=y
CONFIG_PACKAGE_screen=y
CONFIG_PACKAGE_tcpdump=y
CONFIG_PACKAGE_terminfo=y
# CONFIG_PACKAGE_uclient-fetch is not set
CONFIG_PACKAGE_ucode-mod-uloop=y
CONFIG_PACKAGE_umbim=y
CONFIG_PACKAGE_uqmi=y
CONFIG_PACKAGE_utftp=y
CONFIG_PACKAGE_uthash=y
CONFIG_PACKAGE_vpn-routes=y
CONFIG_PACKAGE_wget-ssl=y
CONFIG_PACKAGE_wireguard-tools=y
CONFIG_PACKAGE_wpad-openssl=y
CONFIG_PACKAGE_wwan=y
CONFIG_PACKAGE_zlib=y
CONFIG_PREINITOPT=y
CONFIG_STRIP_ARGS="--strip-all"
CONFIG_STRIP_EXTRACT_DEBUGINFO=y
CONFIG_TARGET_PREINIT_DISABLE_FAILSAFE=y
# CONFIG_USE_SSTRIP is not set
CONFIG_USE_STRIP=y
```
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

yogo1212, 2024-10-24:
this is how it magically disappeared again this morning:

```
[74548.300059] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 2: transmit queue 6 timed out 32584140 ms
[74553.259821] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 2: transmit queue 6 timed out 32589100 ms
[74558.299764] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 2: transmit queue 6 timed out 32594140 ms
[74563.259570] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 2: transmit queue 6 timed out 32599100 ms
[79907.093015] mt7530-mdio mdio-bus:1f lan4: Link is Down
[79907.098316] br-n_1: port 4(lan4) entered disabled state
[79910.349400] mt7530-mdio mdio-bus:1f lan4: Link is Up - 1Gbps/Full - flow control rx/tx
[79910.349857] br-n_1: port 4(lan4) entered blocking state
[79910.362590] br-n_1: port 4(lan4) entered forwarding state
```

lukasz1992, 2024-10-26:
I think there were some patches for filogic not backported to ramips

yogo1212, 2024-10-26:
@lukasz1992 it has been working since my last past but just minutes after reading your message it stopped working again :-D

i would happily test some patches - do you have some details?

lukasz1992, 2024-10-27:
There would be many - imho easiest would be to copy mtk_eth_soc* and files from filogic target.
Right now you can these two: https://github.com/lukasz1992/openwrt/commit/1ef4d4f128737132778246924d2ec4f7acdc2c38 https://github.com/lukasz1992/openwrt/commit/0c6118938c9e017f54d128f40a476145c682f356

yogo1212, 2024-10-27:
thanks for checking back! indeed, these commits look very promising.

turns out, both changes are in main (and in my build) already:
https://github.com/openwrt/openwrt/blob/main/target/linux/generic/pending-6.6/735-net-ethernet-mtk_eth_soc-fix-memory-corruption-durin.patch
https://github.com/openwrt/openwrt/commit/852ed0379a170ca031bc238ec34a4a8768526a45

i didn't have any such issues before my rebase (from before these patches).

if i find some more time (after tracing pakets through net/bridge/* day and night two weeks ago), i'll start by looking at dma/queueing.

khanjui, 2024-10-30:
It also happens very rarely on my MT7621 too.

yogo1212, 2024-10-31:
@khanjui my cudy is also using a mt7621 chipset, the mt7530 is the on-board switch.

which device are you using?

yogo1212, 2024-11-04:
@romanovj interesting!
i read "100m" and immediately tried that as well but i just got hit again with a 100m switch before the router.

yogo1212, 2024-11-13:
this will be horrible to bisect because of its sporadic nature but i'm nearing the point where i see no alternative.

target/linux/generic/pending-6.6/730-net-ethernet-mtk_eth_soc-reset-all-TX-queues-on-DMA-.patch is applied.
i don't see many related changes from the "previously working state" (?) by glimpsing at the commit log.
maybe it's a patch in the kernel fixes

yogo1212, 2024-11-26:
at the risk of exposing sensitive information, sometimes some frames do get through:

```
12:25:11.715487 22:8b:1d:ea:fb:3e > ff:fd:00:80:02:00, ethertype Unknown (0xedff), length 5384: 
	0x0000:  c0a8 0001 5920 0804 0000 0000 0204 0001  ....Y...........
	0x0010:  0000 0000 5920 c2a8 0048 6285 0000 0000  ....Y....Hb.....
	0x0020:  8028 0024 0080 239c 0000 2046 000a 00b2  .(.$..#....F....
	0x0030:  0445 4c46 4345 5245 4044 4746 4146 4145  .ELFCERE@DGFAFAE
	0x0040:  4041 4141 4543 4143 4141 4102 4143 4143  @AAAECACAAA.ACAC
	0x0050:  4144 4345 2046 4845 4245 5044 4c45 4846  ADCE.FHEBEPDLEHF
	0x0060:  4141 4143 4246 4143 4143 4142 4143 4343  AAACBFACACABACCC
	0x0070:  4d42 2502 4f00 ff53 0100 0000 0000 0000  MB%.O..S........
	0x0080:  0000 0000 0000 0000 0000 0000 0000 0000  ................
	0x0090:  0018 0000 0000 1100 0000 0000 0000 0000  ................
	0x00a0:  0000 0000 0000 0000 0001 0001 0018 0056  ...............V
	0x00b0:  0029 0054 0001 0002 024c 4f54 4d41 4b4c  .).T.....LOTMAKL
	0x00c0:  5751 4500 5842 524f 0d00 0000 0801 070f  WQE.XBRO........
	0x00d0:  0000 0001 0120 05a6 0000 fffd 0080 0200  ................
	0x00e0:  2289 1dea ff7f ffff 0001 0800 5b28 0806  "...........[(..
	0x00f0:  2289 1de8 0204 0001 000e 0000 5b2a c0a8  "...........[*..
	0x0100:  80a8 0001 0000 0000 0000 0000 0000 0000  ................
	0x0110:  0000 0000 0000 0200 00c8 6285 0000 219c  ..........b...!.
	0x0120:  8088 0024 110a 68cf 0000 2046 000a 00b2  ...$..h....F....
	0x0130:  0445 4c46 4345 5245 4044 4746 4146 4345  .ELFCERE@DGFAFCE
	0x0140:  4041 4141 4543 4143 4141 4102 4143 4143  @AAAECACAAA.ACAC
	0x0150:  4144 4345 2046 4845 4245 5044 4c45 4846  ADCE.FHEBEPDLEHF
	0x0160:  4141 4143 4646 4143 4143 4142 4143 4343  AAACFFACACABACCC
	0x0170:  4942 0502 4f00 ff53 0100 0000 0000 0000  IB..O..S........
	0x0180:  0000 0000 0000 0000 0000 0000 0000 0000  ................
	0x0190:  0018 0000 0000 1100 0000 0000 0000 0000  ................
	0x01a0:  0000 0000 0000 0000 0001 0001 0018 0056  ...............V
	0x01b0:  0029 005c 0001 0002 024c 4f54 4d41 4b4c  .).\.....LOTMAKL
	0x01c0:  5751 4500 5c42 524f 0d00 0000 0881 070f  WQE.\BRO........
	0x01d0:  0000 0001 0120 05a6 0000 fff1 0080 0200  ................
	0x01e0:  2289 1dea fb3f fdff 0001 0800 5b2a 0806  "....?......[*..
	0x01f0:  2289 1de8 0604 0001 000e 0000 5b2a c0a8  "...........[*..
	0x0200:  8028 0001 0000 0000 0000 0000 0000 0000  .(..............
	0x0210:  0000 0000 0000 0200 00c8 6285 0000 219c  ..........b...!.
	0x0220:  8088 0024 118a 68cf 0000 2046 000a 00b2  ...$..h....F....
	0x0230:  0445 4c46 4345 5245 4044 4746 4146 4345  .ELFCERE@DGFAFCE
	0x0240:  4041 4141 4143 4143 4141 4102 4143 4143  @AAAACACAAA.ACAC
	0x0250:  4144 4345 2046 4845 4245 5044 4c45 4846  ADCE.FHEBEPDLEHF
	0x0260:  4141 4143 4646 4143 4143 4142 4101 4343  AAACFFACACABA.CC
	0x0270:  4d42 2502 4f00 ff53 0100 0000 0000 0000  MB%.O..S........
	0x0280:  0000 0000 0000 0000 0000 0000 0000 0000  ................
	0x0290:  0018 0000 0000 1100 0000 0000 0000 0000  ................
	0x02a0:  0000 0000 0000 0000 0001 0001 0018 0056  ...............V
	0x02b0:  0009 0050 0001 0002 024c 4f54 4d41 4b4c  ...P.....LOTMAKL
	0x02c0:  5751 4500 5c42 524f 0d00 0000 0801 070f  WQE.\BRO........
	0x02d0:  0000 0001 0120 05a6 0000 fff1 0000 0200  ................
	0x02e0:  2289 1dea fb7f ffff 0001 0800 5b2a 0806  "...........[*..
	0x02f0:  0289 1de0 0604 0001 002e 0000 5b2a c0a8  ............[*..
	0x0300:  8028 0001 0000 0000 0000 0000 0000 0000  .(..............
	0x0310:  0000 0000 0000 0200 0048 6285 0000 219c  .........Hb...!.
	0x0320:  80a8 0024 118a 68cf 0000 2046 000a 00b2  ...$..h....F....
	0x0330:  0445 4c46 4345 5245 4044 4746 4146 4345  .ELFCERE@DGFAFCE
	0x0340:  4041 4141 4543 4143 4141 4102 4143 4143  @AAAECACAAA.ACAC
	0x0350:  4144 4345 2046 4845 4245 5044 4c45 4846  ADCE.FHEBEPDLEHF
	0x0360:  4141 4143 4646 4143 4143 4142 4143 4343  AAACFFACACABACCC
	0x0370:  4d42 2502 4f00 ff53 0100 0000 0000 0000  MB%.O..S........
	0x0380:  0000 0000 0000 0000 0000 0000 0000 0000  ................
	0x0390:  0018 0000 0000 1100 0000 0000 0000 0000  ................
	0x03a0:  0000 0000 0000 0000 0001 0001 0018 0056  ...............V
	0x03b0:  0009 0050 0001 0002 024c 4f54 4d41 4b4c  ...P.....LOTMAKL
	0x03c0:  5751 4500 5c42 524f 0d00 0000 0801 070f  WQE.\BRO........
	0x03d0:  0000 0001 0120 05a6 0000 fffd 0080 0200  ................
	0x03e0:  2289 1dea ff7f ffff 0001 0800 5b28 0806  "...........[(..
	0x03f0:  0209 0de0 0604 0001 002e 0000 5b28 c0a8  ............[(..
	0x0400:  8028 0001 0000 0000 0000 0000 0000 0000  .(..............
	0x0410:  0000 0000 0000 0200 0008 6281 0000 239c  ..........b...#.
[... 270 lines cut ...]
12:25:12.775368  [|llc]
12:25:13.815431 22:8b:1d:ea:fb:3f > ff:fd:00:80:02:00, ethertype Unknown (0xfdff), length 164: 
	0x0000:  0001 0800 5b2a 0806 628b 1de8 0204 0001  ....[*..b.......
	0x0010:  8028 0001 0000 0000 0000 0000 0000 0000  .(..............
	0x0020:  0000 0000 0000 239c 00c8 6285 0000 219c  ......#...b...!.
	0x0030:  80a8 002c 118a 68cf 0000 2046 000a 00b2  ...,..h....F....
	0x0040:  0445 4c46 4345 5245 4044 4746 4146 4145  .ELFCERE@DGFAFAE
	0x0050:  4041 4141 4543 4143 4141 4102 4143 4143  @AAAECACAAA.ACAC
	0x0060:  4144 4345 2046 4845 4245 5044 4c45 4846  ADCE.FHEBEPDLEHF
	0x0070:  4141 4143 4246 4143 4143 4142 4143 4343  AAACBFACACABACCC
	0x0080:  4d42 2502 4f00 ff53 0100 0000 0000 0000  MB%.O..S........
	0x0090:  0000 0000 0000                           ......
[... cut ...]
	0x0020:  0000 0000 0000 239c 00c8 6285 0000 219c  ......#...b...!.
	0x0030:  80a8 002c 118a 68cf 0000 2046 0008 00b2  ...,..h....F....
	0x0040:  0445 4c46 4345 5245 4044 4746 4146 4145  .ELFCERE@DGFAFAE
	0x0050:  4041 4141 4143 4143 4141 4102 4143 4143  @AAAACACAAA.ACAC
	0x0060:  4144 4345 2044 4845 4345 5044 4c45 4846  ADCE.DHECEPDLEHF
	0x0070:  4141 4143 4246 4143 4143 4142 4143 4343  AAACBFACACABACCC
	0x0080:  4d42 2502 4b00 fd53 0100 0000 0000 0000  MB%.K..S........
	0x0090:  0000 0000 0000                           ......
```

yogo1212, 2024-11-26:
since i got the bug again, my last 'thought to be working' state turns out to be a broken state....

yogo1212, 2024-11-27:
i noticed that waiting for more than just 10 seconds between unplugging and re-powering the router also helps.
30 seconds seems to do it fairly reliable.

it looks as if the switch chip has a stroke, starts behaving erratically and, maybe, it recovers by its own after waiting long enough - sometimes.

yogo1212, 2024-11-27:
with this now likely being a switch-internal problem, it is out of my league :-(

best i can do is bisect and test. but right now, i don't even know which range to bisect.

yogo1212, 2024-11-28:
@msquirogac there's no link down for me. might still be the same cause, though

yogo1212, 2024-12-09:
@romanovj 

hi :-)

i don't tink i get it :sweat_smile: 
you detect the tx queue hang and perform a reset?

yogo1212, 2024-12-12:
@romanovj thanks for providing more details!
i'm still unsure about how your solution turns out in code (mainly because i lack knowledge).
do you think you could turn it into a commit?

yogo1212, 2024-12-19:
@romanovj sure - thanks! :slightly_smiling_face: 
build is running

we'll have to wait a few weeks, though, for a positive result to be meaningful :-(

yogo1212, 2024-12-27:
@romanovj just to be sure: the `force_reset` module parameter is disabled for the test, right?

it looks to me as if it would only work once anyway

yogo1212, 2025-01-03:
@romanovj still running :-)
i'd say we have another look at the end of january - but you can always ping me here if your curious!

yogo1212, 2025-01-21:
@romanovj i still haven't noticed any outages 🙂

yogo1212, 2025-01-21:
nothing that looks unusual:

```
[432873.646610] mt7530-mdio mdio-bus:1f lan1: Link is Down
[432873.652135] br-n_1: port 1(lan1) entered disabled state
[432877.775866] mt7530-mdio mdio-bus:1f lan1: Link is Up - 1Gbps/Full - flow control rx/tx
[432877.775982] br-n_1: port 1(lan1) entered blocking state
[432877.789257] br-n_1: port 1(lan1) entered forwarding state
[471323.050413] mt7530-mdio mdio-bus:1f lan1: Link is Down
[471323.055801] br-n_1: port 1(lan1) entered disabled state
[526946.891656] mt7530-mdio mdio-bus:1f lan1: Link is Up - 1Gbps/Full - flow control off
[526946.899635] br-n_1: port 1(lan1) entered blocking state
[526946.904991] br-n_1: port 1(lan1) entered forwarding state
[526972.853147] mt7530-mdio mdio-bus:1f lan1: Link is Down
[526972.858642] br-n_1: port 1(lan1) entered disabled state
[526977.235436] mt7530-mdio mdio-bus:1f lan1: Link is Up - 1Gbps/Full - flow control rx/tx
[526977.243371] br-n_1: port 1(lan1) entered blocking state
[526977.248752] br-n_1: port 1(lan1) entered forwarding state
[527228.446681] mt7530-mdio mdio-bus:1f lan1: Link is Down
[527228.452050] br-n_1: port 1(lan1) entered disabled state
[527244.601822] mt7530-mdio mdio-bus:1f lan1: Link is Up - 1Gbps/Full - flow control off
[527244.606207] br-n_1: port 1(lan1) entered blocking state
[527244.614960] br-n_1: port 1(lan1) entered forwarding state
[527265.339964] mt7530-mdio mdio-bus:1f lan1: Link is Down
[527265.345363] br-n_1: port 1(lan1) entered disabled state
[527269.653467] mt7530-mdio mdio-bus:1f lan1: Link is Up - 1Gbps/Full - flow control rx/tx
[527269.661628] br-n_1: port 1(lan1) entered blocking state
[527269.666994] br-n_1: port 1(lan1) entered forwarding state
[564505.485279] mt7530-mdio mdio-bus:1f lan1: Link is Down
[564505.490724] br-n_1: port 1(lan1) entered disabled state
```

could it be there haven't been any events that would have 'crashed' the chip?

Critter74, 2025-05-16:
Hello,
I try compile OpenWRT v24.10.1 with actual mt76 drivers from https://git01.mediatek.com/plugins/gitiles/openwrt/feeds/mtk-openwrt-feeds/
After installing it to my BPI-R4, I have no network and the issue with timed out. Does this patch 
tx_timeout.patch fix it too?
Thx

alanswanson, 2025-08-05:
@romanovj Would you be able to split your patch into parts for testing? Was looking to apply separate patches for the warm reset fix and the tx timeout reset (without the force_reset) but was unsure of untangling the combined patch.

Also had a look on the OpenWrt forum for any of your posts discussing this patch but didn't find any.

Vazgen005, 2025-08-15:
I'm getting same problem in my 2 openwrt routers:
https://openwrt.org/toh/routerich/ax3000
https://openwrt.org/toh/asus/rt-ax53u

There is solution for this problem?