openwrt/openwrt #16763: Some OpenWrt DNS Query requests appear to be damaged

dkachel7258, 2024-10-23:
### Describe the bug

When using the Pi-hole applicationto resolve and monitor DNS requests from OpenWrt, I see 
things like:

When looking at the test log files in "/var/log/pihole":

 - Short addresses (i.e., 20 hex digits instead of 32) when doing an IPv6 reverse lookup. The part
 missing is last 12 characters of the lookup address and the ".ip6.arpa" suffix.

      Jul 14 08:41:53 dnsmasq[22366]: query b.8.7.0.0.0.0.0.0.0.0.0.0.0.0.0.2.c.2.0 from 192.168.1.1

- "Name unprintable" messages for A, AAAA, and HTTPS name lookups.

        Jul 14 11:56:10 dnsmasq[24079]: query <name unprintable> from 192.168.1.1

- Damaged queries are shown without a query type. (e.g., " query " instead of " query[HTTPS]".)

When looking through a Query Log in the Pi-hole web interface:

 - Damaged queries have a strange query type, like "TYPE0" or "TYPE126".
 
 - Damaged queries that have an unprintable name show up in the Query Log as something 
   like "musicstatus-us-w.edge-itunes-apple.com."
   where the question marks display as rectangles with an X in them (in the Pi-hole web 
   interface).

### OpenWrt version

r27798-dbc2923cbe

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

bcm27xx/bcm2712

### Device

Raspberry Pi 5 Model B v1.0

### Image kind

Official downloaded image

### Steps to reproduce

  1. Install and set up OpenWrt.
  2. Install and set up Pi-hole.
  3. Configure the OpenWrt LAN interface to handle all local network DNS requests.
  4. Configure the OpenWrt LAN interface to use Pi-hole as the DNS resolver.

Let run for while and then check the /var/log/pihole/pihole.log file for query entries without
a query type.

If desired, the timestamp information of those entries can be used to help locate those records
in the Pi-hole web interface query log. (FWIW)

### Additional info

Behavior happens on SNAPSHOT release and not on release 23.05.5.

Behavior also happens on a Raspberry Pi 4 Model B v1.1.  (Not tested on others)

I once got a FORMERR error and a packet dump from an "nslookup" command.

In cases where OpenWrt isn't handling local DNS lookups for LAN hosts, only the Pi-hole 
log entries generated by OpenWrt itself show this behavior. 

Behavior has been present for months.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-10-23:
itunes is not made  by openwrt. What if you point your itunes directly ar pihole?

dkachel7258, 2024-10-23:
The string with "itunes" in it is an example of a corrupted FQDN that was sent by OpenWrt to Pi-hole to be forwarded to an external DNS resolver.

This might be a clearer example - it's a screen capture of a Pi-hole Query Log that shows a corrupted name lookup request.

[image]

Also note that in addition to the problem with the name, the query type is shown as "N/A" instead of A, AAAA, or HTTPS.

FWIW, I suspect that part of the DNS packet sent to Pi-hole is being overwritten with zeroes.  This is based on a hex dump from an nslookup request for a global IPv6 address for one of my systems.  In the middle of the lookup string is 16 zeroes

brada4, 2024-10-24:
None of that originates from openwrt, tcpdump in both sides and get query log from dnsmasq.

dkachel7258, 2024-10-24:
Neither tcpdump nor syslog are in my skillset. Without those, this problem report seems incomplete.

Withdrawing problem submission.

closed 2024-10-24 by dkachel7258