openwrt/openwrt #16772: [Build] How to resolve `recursive dependency` error?

jjm2473, 2024-10-24:
This may not a BUG.

I want to add `atlantic.ko` to openwrt, so I add these code to `package/kernel/linux/modules/netdevices.mk`:

```Makefile

define KernelPackage/atlantic
  SUBMENU:=$(NETWORK_DEVICES_MENU)
  TITLE:=aQuantia AQtion(tm) Ethernet card support
  DEPENDS:=@PCI_SUPPORT @TARGET_x86_64 +kmod-ptp +(PACKAGE_kmod-macsec||PACKAGE_kmod-macsec=m):kmod-macsec
  KCONFIG:= \
	CONFIG_AQTION
  FILES:=$(LINUX_DIR)/drivers/net/ethernet/aquantia/atlantic/atlantic.ko
  AUTOLOAD:=$(call AutoProbe,atlantic)
endef

$(eval $(call KernelPackage,atlantic))

```

Then I run `make menuconfig`, it has a error:
```
tmp/.config-package.in:16587:error: recursive dependency detected!
tmp/.config-package.in:16587:   symbol PACKAGE_kmod-macsec is selected by PACKAGE_kmod-macsec
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

```

How to resolve that error?

My goal is to make `kmod-atlantic` depend on `kmod-macsec` when `PACKAGE_kmod-macsec=y` or `PACKAGE_kmod-macsec=m`.

brada4, 2024-10-24:
Issue template is there for a reason.

jjm2473, 2024-10-25:
@brada4 So which device and which openwrt version do you think this issue should be related to?

raenye, 2024-11-04:
He probably means you should ask your question in the OpenWrt forum, or over IRC (#openwrt-devel), rather than on the issue tracker.