openwrt/openwrt #16773: Can't compile 22.03.7

apollogeek, 2024-10-24:
### Describe the bug

Here is log:

install -d -m0755 /home/apollo/Desktop/openwrt-22.03.7/tmp/stage-curl/host/bin /home/apollo/Desktop/openwrt-22.03.7/tmp/stage-curl/usr/bin /home/apollo/Desktop/openwrt-22.03.7/tmp/stage-curl/usr/include /home/apollo/Desktop/openwrt-22.03.7/tmp/stage-curl/usr/lib /home/apollo/Desktop/openwrt-22.03.7/tmp/stage-curl/usr/lib/pkgconfig
install -m0755 /home/apollo/Desktop/openwrt-22.03.7/build_dir/target-mipsel_24kc_musl/curl-8.6.0/ipkg-install/usr/bin/curl-config /home/apollo/Desktop/openwrt-22.03.7/tmp/stage-curl/usr/bin/
cp -fpR /home/apollo/Desktop/openwrt-22.03.7/build_dir/target-mipsel_24kc_musl/curl-8.6.0/ipkg-install/usr/include/curl /home/apollo/Desktop/openwrt-22.03.7/tmp/stage-curl/usr/include/
cp -fpR /home/apollo/Desktop/openwrt-22.03.7/build_dir/target-mipsel_24kc_musl/curl-8.6.0/ipkg-install/usr/lib/libcurl.{a,so*} /home/apollo/Desktop/openwrt-22.03.7/tmp/stage-curl/usr/lib/
cp -fpR /home/apollo/Desktop/openwrt-22.03.7/build_dir/target-mipsel_24kc_musl/curl-8.6.0/libcurl.pc /home/apollo/Desktop/openwrt-22.03.7/tmp/stage-curl/usr/lib/pkgconfig/
/home/apollo/Desktop/openwrt-22.03.7/staging_dir/host/bin/sed -i -e 's,^(prefix|exec_prefix)=.,\1=/home/apollo/Desktop/openwrt-22.03.7/staging_dir/target-mipsel_24kc_musl/usr,g' /home/apollo/Desktop/openwrt-22.03.7/tmp/stage-curl/usr/bin/curl-config
[ -n "-L/home/apollo/Desktop/openwrt-22.03.7/staging_dir/toolchain-mipsel_24kc_gcc-11.2.0_musl/usr/lib -L/home/apollo/Desktop/openwrt-22.03.7/staging_dir/toolchain-mipsel_24kc_gcc-11.2.0_musl/lib -L/home/apollo/Desktop/openwrt-22.03.7/staging_dir/target-mipsel_24kc_musl/usr/lib/libiconv-stub/lib -Wl,-rpath-link=/home/apollo/Desktop/openwrt-22.03.7/staging_dir/target-mipsel_24kc_musl/usr/lib/libiconv-stub/lib -L/home/apollo/Desktop/openwrt-22.03.7/staging_dir/target-mipsel_24kc_musl/usr/lib/libintl-stub/lib -Wl,-rpath-link=/home/apollo/Desktop/openwrt-22.03.7/staging_dir/target-mipsel_24kc_musl/usr/lib/libintl-stub/lib -znow -zrelro -Wl,--gc-sections" ] && /home/apollo/Desktop/openwrt-22.03.7/staging_dir/host/bin/sed -i -e 's#-L/home/apollo/Desktop/openwrt-22.03.7/staging_dir/toolchain-mipsel_24kc_gcc-11.2.0_musl/usr/lib -L/home/apollo/Desktop/openwrt-22.03.7/staging_dir/toolchain-mipsel_24kc_gcc-11.2.0_musl/lib -L/home/apollo/Desktop/openwrt-22.03.7/staging_dir/target-mipsel_24kc_musl/usr/lib/libiconv-stub/lib -Wl,-rpath-link=/home/apollo/Desktop/openwrt-22.03.7/staging_dir/target-mipsel_24kc_musl/usr/lib/libiconv-stub/lib -L/home/apollo/Desktop/openwrt-22.03.7/staging_dir/target-mipsel_24kc_musl/usr/lib/libintl-stub/lib -Wl,-rpath-link=/home/apollo/Desktop/openwrt-22.03.7/staging_dir/target-mipsel_24kc_musl/usr/lib/libintl-stub/lib -znow -zrelro -Wl,--gc-sections##g' /home/apollo/Desktop/openwrt-22.03.7/tmp/stage-curl/usr/lib/pkgconfig/libcurl.pc || true
ln -sf /home/apollo/Desktop/openwrt-22.03.7/staging_dir/target-mipsel_24kc_musl/usr/bin/curl-config /home/apollo/Desktop/openwrt-22.03.7/tmp/stage-curl/host/bin/
find /home/apollo/Desktop/openwrt-22.03.7/tmp/stage-curl -name '.la' | xargs -r rm -f;
if [ -f /home/apollo/Desktop/openwrt-22.03.7/staging_dir/target-mipsel_24kc_musl/packages/curl.list ]; then /home/apollo/Desktop/openwrt-22.03.7/scripts/clean-package.sh "/home/apollo/Desktop/openwrt-22.03.7/staging_dir/target-mipsel_24kc_musl/packages/curl.list" "/home/apollo/Desktop/openwrt-22.03.7/staging_dir/target-mipsel_24kc_musl"; fi
if [ -d /home/apollo/Desktop/openwrt-22.03.7/tmp/stage-curl ]; then (cd /home/apollo/Desktop/openwrt-22.03.7/tmp/stage-curl; find ./ > /home/apollo/Desktop/openwrt-22.03.7/tmp/stage-curl.files); SHELL= flock /home/apollo/Desktop/openwrt-22.03.7/tmp/.staging-dir.flock -c ' mv /home/apollo/Desktop/openwrt-22.03.7/tmp/stage-curl.files /home/apollo/Desktop/openwrt-22.03.7/staging_dir/target-mipsel_24kc_musl/packages/curl.list && cp -fpR /home/apollo/Desktop/openwrt-22.03.7/tmp/stage-curl/* /home/apollo/Desktop/openwrt-22.03.7/staging_dir/target-mipsel_24kc_musl/; '; fi
rm -rf /home/apollo/Desktop/openwrt-22.03.7/tmp/stage-curl
touch /home/apollo/Desktop/openwrt-22.03.7/staging_dir/target-mipsel_24kc_musl/stamp/.curl_installed
rm -rf /home/apollo/Desktop/openwrt-22.03.7/build_dir/target-mipsel_24kc_musl/curl-8.6.0/.pkgdir/curl.installed /home/apollo/Desktop/openwrt-22.03.7/build_dir/target-mipsel_24kc_musl/curl-8.6.0/.pkgdir/curl
mkdir -p /home/apollo/Desktop/openwrt-22.03.7/build_dir/target-mipsel_24kc_musl/curl-8.6.0/.pkgdir/curl
install -d -m0755 /home/apollo/Desktop/openwrt-22.03.7/build_dir/target-mipsel_24kc_musl/curl-8.6.0/.pkgdir/curl/usr/bin
install -m0755 /home/apollo/Desktop/openwrt-22.03.7/build_dir/target-mipsel_24kc_musl/curl-8.6.0/ipkg-install/usr/bin/curl /home/apollo/Desktop/openwrt-22.03.7/build_dir/target-mipsel_24kc_musl/curl-8.6.0/.pkgdir/curl/usr/bin/
touch /home/apollo/Desktop/openwrt-22.03.7/build_dir/target-mipsel_24kc_musl/curl-8.6.0/.pkgdir/curl.installed
mkdir -p /home/apollo/Desktop/openwrt-22.03.7/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp
SHELL= flock /home/apollo/Desktop/openwrt-22.03.7/tmp/.root-copy.flock -c 'cp -fpR /home/apollo/Desktop/openwrt-22.03.7/build_dir/target-mipsel_24kc_musl/curl-8.6.0/.pkgdir/curl/. /home/apollo/Desktop/openwrt-22.03.7/staging_dir/target-mipsel_24kc_musl/root-ramips/'
touch /home/apollo/Desktop/openwrt-22.03.7/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.curl_installed
echo "curl" >> /home/apollo/Desktop/openwrt-22.03.7/staging_dir/target-mipsel_24kc_musl/pkginfo/curl.default.install
make[3]: Leaving directory '/home/apollo/Desktop/openwrt-22.03.7/feeds/packages/net/curl'
time: package/feeds/packages/curl/compile#46.05#6.76#31.86
make[2]: Leaving directory '/home/apollo/Desktop/openwrt-22.03.7'
make[1]: *** [package/Makefile:114: /home/apollo/Desktop/openwrt-22.03.7/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/apollo/Desktop/openwrt-22.03.7'
make: *** [/home/apollo/Desktop/openwrt-22.03.7/include/toplevel.mk:230: world] Error 2

### OpenWrt version

22.03.7

### OpenWrt release

22.03.7

### OpenWrt target/subtarget

ramips/mt7621

### Device

Xiaomi Redmi Ac2100

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-10-24:
Any chance you ever format logs into code blocks? Your log does not show any error which very likely preceeds actual missing file error by half million output lines.

apollogeek, 2024-10-24:
> Any chance you ever format logs into code blocks? Your log does not show any error which very likely preceeds actual missing file error by half million output lines.

- Really? You maybe can't see that error but I see that

brada4, 2024-10-24:
Like gcc or ld error not producing output?

apollogeek, 2024-10-24:
> Like gcc or ld error not producing output?

I found problem, it's Hysteria can't compile, I tried without it and it's done, with 23.05.5 version the same problem, can't compile with Hysteria, I don't know why but without it working

closed 2024-10-24 by apollogeek

brada4, 2024-10-24:
You need to ask party giving you histeria and stop flooding bug tracker with non-bugs
https://github.com/openwrt/openwrt/issues/16732