openwrt/openwrt #16792: No VLAN for TP-Link TL-WR802N

adrai, 2024-10-26:
### Describe the bug

Is it just me, or can't we make a VLAN setup with the TP-Link TL-WR802N? I would like to have different SSIDs, mapped to different VLANs.

[image]

Also only VLAN ids between 0-15 are accepted:
[image]

### OpenWrt version

r24012-d8dd03c46f

### OpenWrt release

23.05.4

### OpenWrt target/subtarget

ramips/mt76x8

### Device

TP-Link TL-WR802N v4

### Image kind

Official downloaded image

### Steps to reproduce

Try to define a VLAN and map that to a SSID...?

### Actual behaviour

There seems to be no way right now to accomplish this.

There is no checkbox to enable VLAN functionality in the Switch menu.

### Expected behaviour

Would be nice if configuring VLANs and mapping them to different SSIDs would work.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-10-27:
Initial image is likely luci issue, the other is that it correctly exposes hw limitation.

adrai, 2024-10-27:
Do you want me to open a new issue in the luci repository?
Or there anything else I need to provide?

brada4, 2024-10-27:
Better open, core system functions withour web ui.

adrai, 2024-10-28:
https://github.com/openwrt/luci/issues/7359

closed 2024-11-11 by adrai