openwrt/openwrt #16794: apk fails to read its own database: FDB format error

hnyman, 2024-10-26:
### Describe the bug

I have compiled an apk enabled version of main/master snapshot r27902-6835ff8cbc (with apk-openssl) for mediatek MT6000, but using apk fails as it seems unable to read its own database of already installed packages (in the image itself).

```
root@router6000:/lib/apk/db# apk update
ERROR: FDB format error (line 6860, entry 'F')
ERROR: Unable to read database: v2 database format error
ERROR: Failed to open apk database: v2 database format error

root@router6000:/lib/apk/db# apk info
ERROR: FDB format error (line 6860, entry 'F')
ERROR: Unable to read database: v2 database format error
ERROR: Failed to open apk database: v2 database format error
```

Looking at the line number, it seems to fail on parsing package info for openssl, roughly at the middle of the 11300 lines in the file:
```
root@router6000:~# less -N /lib/apk/db/installed
   6854 U:https://www.openssl.org/
   6855 L:Apache-2.0
   6856 o:feeds/base/libs/openssl
   6857 m:Eneas U de Queiroz <cotequeiroz@gmail.com>
   6858 c:
   6859 D:libc libopenssl3
   6860 F:etc
   6861 F:etc/config
   6862 R:openssl
   6863 Z:Q247DEQpj8HBSa+/TImW+5JCeuQeRkm5NMpJWZG3hSuFU=
   6864 F:etc/init.d
   6865 R:openssl
   6866 a:0:0:755
   6867 Z:Q2K3nUsjzYTwbSQIAcrvqq7gqnCnrl1bW4fk9q3gw1Zv8=
```

But apk is able to e.g. verify the package index downloaded from apracar's test site, so in general, it should be ok?
```
root@router6000:~# apk verify packages.adb
packages.adb: OK
```

I am using apk-openssl:
```
root@router6000:/etc# grep apk build.config
# CONFIG_PACKAGE_apk-mbedtls is not set
CONFIG_PACKAGE_apk-openssl=y
```

The same happens also with RT3200 at the same place:
```
 OpenWrt SNAPSHOT, r27909-d7ed4c77b5
 -----------------------------------------------------
root@router4:~# apk info
ERROR: FDB format error (line 6904, entry 'F')
ERROR: Unable to read database: v2 database format error
ERROR: Failed to open apk database: v2 database format error

   6901 m:Eneas U de Queiroz <cotequeiroz@gmail.com>
   6902 c:
   6903 D:libc libopenssl3
   6904 F:etc
   6905 F:etc/config
   6906 R:openssl
```

### OpenWrt version

r27902-6835ff8cbc

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic  and mediatek/7622

### Device

GL.iNet GL-MT6000   and Belkin RT3200

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

aparcar, 2024-10-26:
@Ansuel wdyt

Ansuel, 2024-10-26:
Feel like a bug with APK but the bisect from @hnyman made me understand the problem better... I also had this error but I hoped it was caused by my downstream patch and not from that.

By the looks of it, it seems empty entry with .apk file.

hnyman, 2024-10-26:
> it seems empty entry with .apk file.

If you refer to the "c:" field, it seems to be empty for all packages, not only for openssl. 

By looking at the apk-tools sources, the field seems to be for a commit hash. 
 `MAP('c', ADBI_PI_REPO_COMMIT),`
`ADB_FIELD(ADBI_PI_REPO_COMMIT,	"repo-commit",	scalar_hexblob),`

Ansuel, 2024-10-26:
No the error talks about entry F. 

Entry can be empty... Problem is if they are apk_blob... They still needs to be allocated and properly init. 

Or maybe someone in APK fkup up and entry F was dropped but never actually dropped from the schema. Can you check what should be the usage of entry F ?

hnyman, 2024-10-26:
> Can you check what should be the usage of entry F ?

Well, looking at the apk sources for first time, it seems pretty hard to find.
At least here it is said that it is about directories:
https://github.com/alpinelinux/apk-tools/blob/master/src/app_convdb.c#L154

And indeed, looking at the "installed" file contents, F is about defining directories for the files to be installed (and subsequent R is then the file).
```
   1319 m:Jo-Philipp Wich <jo@mein.io>, Hannu Nyman <hannu.nyman@iki.fi>
   1320 c:
   1321 D:collectd libc
   1322 F:lib
   1323 F:lib/apk
   1324 F:lib/apk/packages
   1325 R:collectd-mod-interface.list
   1326 Z:Q2Vq/ENS/oHulypWu7+3ZR061rPWXZ5YhnC3PpSGryECE=
   1327 F:usr
   1328 F:usr/lib
   1329 F:usr/lib/collectd
   1330 R:interface.so
   1331 a:0:0:755
   1332 Z:Q29yfaitEuJplzvXaV4+bA+1A8tgnAfdujZhzjnF+u/sw=
```

Ansuel, 2024-10-26:
It's in database.c that is just an app to convert the db

https://github.com/alpinelinux/apk-tools/blob/master/src/database.c

But why the format error.... strange....

hnyman, 2024-10-27:
I wonder if we are encountering the problem just because we are using apk from a commit in a feature branch, which has then later been merged into the mainline apk, maybe in a slightly different way or with some earlier/later fixes?

We are using the apk commit `54caa31b` (from early August) as our apk source, but that commit is not in the official apk commit history.  (Apparently `7972a780` of 21 August is the same commit in the official apk history)

@aparcar refers to branch "index-trust" in his commit ef8c1adb `"apk: switch to index-trust branch"` here, but there is no such branch in apk any more, as the feature has been adopted into the official apk.

Maybe we are encountering something that has already been fixed in apk? 
Should we bump our apk to be from the official apk mainline, and maybe something newer than the current 6 Aug 2024 commit from a feature branch?

(I will test the most recent apk commit)

Ansuel, 2024-10-27:
@hnyman ok I discovered the culprit :D 

If you notice openssl adds a new line and destroy the database... Check the T entry... This is totally a fragility of APK and I will produce a patch to handle this.

Aka
- yes it's a problem caused by us
- yes it's actually an APK bug

hnyman, 2024-10-27:
Great that you figured it out.
Sounds fragile also for upstream if the description text is not sanitized when it is entered into the database.

Ansuel, 2024-10-27:
@hnyman well problem was something totally different and wasn't us adding new line char... More a too long entry getting corrupted.

If you want further reading :D 
https://gitlab.alpinelinux.org/alpine/apk-tools/-/merge_requests/236/diffs

hnyman, 2024-10-27:
Great that you found the real reason.

I read your explanation. And funnily enough, I had noticed from less/hexedit that the long descriptions ended in a non-printable char instead of the dot, but as none of occurences earlier in the file seemed to cause problems, I just thought that it may be an intended end marker for the longest descriptions... Sounds like that was a wrong guess. 
(Apparently other chars like 0x03 etc. are handled ok, but in this case the last wrong char was newline, breaking the database encoding.)

I will apply and test with your patches.
Are you committing them now to OpenWrt repo as patches for our apk?

Ansuel, 2024-10-27:
> Are you committing them now to OpenWrt repo as patches for our apk?

They are in the gigantic APK wip pr.

pko76, 2024-10-27:
I have a similar problem:
[LuCI Master (24.298.52515~e34c268)](https://github.com/openwrt/luci) / [OpenWrt SNAPSHOT (r27913-a2aabc9a7a)]
root@OpenWrt:~# apk verify packages.adb
packages.adb: No such file or directory
root@OpenWrt:~# grep apk build.config
grep: build.config: No such file or directory
root@OpenWrt:~# apk info
ERROR: Unable to read database: No such file or directory
ERROR: Failed to open apk database: No such file or directory
root@OpenWrt:~# apk --version
apk-tools 3.0.0_pre20240806, compiled for armv7.

hnyman, 2024-10-28:
@Ansuel 's fix for this has now been merged upstream at apk:

* https://gitlab.alpinelinux.org/alpine/apk-tools/-/commits/a1ad8568a1378d71c1f6354c428ebb1baa46fa05
* https://gitlab.alpinelinux.org/alpine/apk-tools/-/merge_requests/236

So, might it make sense to update our apk to that upstream HEAD and avoid taking the local patch separately to our marginally older apk version.

I close this bug, as we have the fix locally as patch in the apk WIP PR, and it has been fixed upstream.

closed 2024-10-28 by hnyman