openwrt/openwrt #16796: Kernel driver bug breaks some w25q128 spi-nor flash chips

RussellSenior, 2024-10-27:
### Describe the bug

Recently discovered that kernel 6.6.x with some wistron w25q128 (chip id 0xef4018) cause boot loops. Described here:

https://lore.kernel.org/all/CALxbwRo_-9CaJmt7r7ELgu+vOcgk=xZcGHobnKf=oT2=u4d4aA@mail.gmail.com/

Bug introduced in linux kernel commit 83e824a4a595132f9bd7ac4f5afff857bfc5991e, effects all v6.6.x kernels. Some SPI-NOR, including parts marked 25Q128BVFG, generate dmesg:

```
[    0.738493] spi-nor spi0.0: BFPT parsing failed. Please consider using SPI_NOR_SKIP_SFDP when declaring the flash
[    0.748779] spi-nor: probe of spi0.0 failed with error -22
[    0.755114] spi-nor spi0.1: w25q128 (16384 Kbytes)
```
The failure to successfully talk to the chip results in a boot loop. This was observed on Buffalo WZR600DHP, but the report above suggests other devices are similarly effected.

### OpenWrt version

r27865-caf29ee97b45

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ath79/generic

### Device

Buffalo WZR-600DHP

### Image kind

Self-built image

### Steps to reproduce

Flash any build with a v6.6.x kernel on affected hardware.

### Actual behaviour

```

BUFFALO U-BOOT Ver 1.00
  == CPU:680MHz, DDR:340MHz, AHB:170MHz ==
AP96 (ar7100) U-boot 0.0.1
DRAM:  128 MB
Top of RAM usable for U-Boot at: 84000000
Reserving 266k for U-Boot at: 83fbc000
Reserving 192k for malloc() at: 83f8c000
Reserving 44 Bytes for Board Info at: 83f8bfd4
Reserving 36 Bytes for Global Data at: 83f8bfb0
Reserving 128k for boot params() at: 83f6bfb0
Stack Pointer at: 83f6bf98
Now running in RAM - U-Boot at: 83fbc000
flash bank #0 found 16 MB flash [W25Q128BV, blk:0x10000, sectors:256]
flash bank #1 found 16 MB flash [W25Q128BV, blk:0x10000, sectors:256]
Flash: 32 MB
In:    serial
Out:   serial
Err:   serial
Memory Test
uboot use  83F6BFB0 - 84000000
Memory Test start(80000000) end(83F00000) size(03F00000)
Pattern 00000000  Writing...  Reading...
Memory Test start(84000000) end(88000000) size(04000000)
Pattern 00000000  Writing...  Reading...
Memory Test OK
### buf_ver=[1.00] U-Boot Ver.=[1.00]
### build_date(env)=[May 21 2012 - 08:29:08] build_date(bin)=[May 21 2012 - 08:29:08]
[... 359 lines cut ...]
Please press Enter to activate this console.
```

### Additional info

https://lore.kernel.org/linux-mtd/D1W69FY9MUZL.33B8G1877MPBL@walle.cc/T/

In the email thread, the reporter, e9hack, provides a patch that fixes it for me. There is an upstream kernel commit (d35df77707bf5ae1221b5ba1c8a88cf4fcdd4901) that fixes it, but it has not been back-ported to v6.6.x. The upstream patch can't be applied directly because the format of the spi-nor parameters changed in 348d772d04fa36d453587c8a9a2720c77eb242c9.

This is kind of URGENT to correct as devices, particularly when flashed remotely, will be bricked, requiring invasive recovery steps. (says a person who remotely bricked a device).

### Diffconfig

```text
CONFIG_TARGET_ath79=y
CONFIG_TARGET_ath79_generic=y
CONFIG_TARGET_ath79_generic_DEVICE_buffalo_wzr-600dhp=y
CONFIG_DEVEL=y
CONFIG_BUSYBOX_CUSTOM=y
CONFIG_BUILD_LOG=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

RussellSenior, 2024-10-27:
Here's my local patch:

```
From 505ca9fa4e7332c4af0b5f4ca6b90981badf371e Mon Sep 17 00:00:00 2001
From: Russell Senior <russell@personaltelco.net>
Date: Sat, 26 Oct 2024 06:00:36 -0700
Subject: [PATCH] kernel: fix-w25q128 per e9hack

---
 target/linux/generic/hack-6.6/422-fix-w25q128.patch | 13 +++++++++++++
 1 file changed, 13 insertions(+)
 create mode 100644 target/linux/generic/hack-6.6/422-fix-w25q128.patch

diff --git a/target/linux/generic/hack-6.6/422-fix-w25q128.patch b/target/linux/generic/hack-6.6/422-fix-w25q128.patch
new file mode 100644
index 000000000000..78e16d494764
--- /dev/null
+++ b/target/linux/generic/hack-6.6/422-fix-w25q128.patch
@@ -0,0 +1,13 @@
+--- a/drivers/mtd/spi-nor/winbond.c
++++ b/drivers/mtd/spi-nor/winbond.c
+@@ -120,8 +120,8 @@ static const struct flash_info winbond_n
+               NO_SFDP_FLAGS(SECT_4K) },
+       { "w25q80bl", INFO(0xef4014, 0, 64 * 1024,  16)
+               NO_SFDP_FLAGS(SECT_4K) },
+-      { "w25q128", INFO(0xef4018, 0, 0, 0)
+-              PARSE_SFDP
++      { "w25q128", INFO(0xef4018, 0, 64 * 1024, 256)
++              NO_SFDP_FLAGS(SECT_4K | SPI_NOR_DUAL_READ | SPI_NOR_QUAD_READ)
+               FLAGS(SPI_NOR_HAS_LOCK | SPI_NOR_HAS_TB) },
+       { "w25q256", INFO(0xef4019, 0, 64 * 1024, 512)
+               NO_SFDP_FLAGS(SECT_4K | SPI_NOR_DUAL_READ | SPI_NOR_QUAD_READ)
-- 
2.47.0

```

KanjiMonster, 2024-10-27:
The upstream fix is in https://github.com/torvalds/linux/commit/d35df77707bf5ae1221b5ba1c8a88cf4fcdd4901 but is only in 6.11+ and hasn't been backported to any stable kernels. Have you tried that one?

KanjiMonster, 2024-10-27:
Ah, just saw that info. Took the liberty and updated your commit hashes with github links to the commits.

KanjiMonster, 2024-10-27:
The change was also backported to 6.1, but not further (https://github.com/gregkh/linux/commit/7c6ba20a0b9aeb82a6c097c74ccbecdda8e9fc25).

RussellSenior, 2024-10-27:
> The change was also backported to 6.1, but not further ([gregkh/linux@7c6ba20](https://github.com/gregkh/linux/commit/7c6ba20a0b9aeb82a6c097c74ccbecdda8e9fc25)).

That's the change that broke it.

KanjiMonster, 2024-10-27:
> > The change was also backported to 6.1, but not further ([gregkh/linux@7c6ba20](https://github.com/gregkh/linux/commit/7c6ba20a0b9aeb82a6c097c74ccbecdda8e9fc25)).
> 
> That's the change that broke it.

Right, that's the backport of it to 6.1 (in 6.1.54), to show that 6.1 is also affected (and not just 6.6, where it was originally introduced).

bam80, 2024-10-27:
[Kroks KNdRt31R19](https://openwrt.org/toh/hwdata/kroks/kroks_rt-pot-mxw) is probably also affected:
https://forum.openwrt.org/t/snapshot-bootloop-on-mt7628/213673?u=bam80
I can test it if get the fixed build.
Thanks.

linusw, 2024-10-27:
I made a backport of the fix to v6.6/v6.1, can someone test if it solves the issue?
https://git.kernel.org/pub/scm/linux/kernel/git/linusw/linux-integrator.git/commit/?h=b4/w25q128-fix-backport&id=07ca9174d974ea8a30d5224d0b6c3dfef2adb696

RussellSenior, 2024-10-27:
@linusw still getting:

`[    0.747531] spi-nor spi0.0: BFPT parsing failed. Please consider using SPI_NOR_SKIP_SFDP when declaring the flash`

and boot looping with your patch.

linusw, 2024-10-27:
I pushed a version which drops PARSE_SFDP which makes it equivalent to my old v1 patch:
https://lore.kernel.org/linux-mtd/20230712-spi-nor-winbond-w25q128-v1-1-f78f3bb42a1c@linaro.org/

So does that work then?

RussellSenior, 2024-10-28:
My local test patch works:

```
--- a/drivers/mtd/spi-nor/winbond.c
+++ b/drivers/mtd/spi-nor/winbond.c
@@ -120,9 +120,10 @@ static const struct flash_info winbond_n
                NO_SFDP_FLAGS(SECT_4K) },
        { "w25q80bl", INFO(0xef4014, 0, 64 * 1024,  16)
                NO_SFDP_FLAGS(SECT_4K) },
-       { "w25q128", INFO(0xef4018, 0, 0, 0)
-               PARSE_SFDP
-               FLAGS(SPI_NOR_HAS_LOCK | SPI_NOR_HAS_TB) },
+       { "w25q128", INFO(0xef4018, 0, 64 * 1024, 256)
+               FLAGS(SPI_NOR_HAS_LOCK | SPI_NOR_HAS_TB)
+               NO_SFDP_FLAGS(SECT_4K | SPI_NOR_DUAL_READ |
+                             SPI_NOR_QUAD_READ) },
        { "w25q256", INFO(0xef4019, 0, 64 * 1024, 512)
                NO_SFDP_FLAGS(SECT_4K | SPI_NOR_DUAL_READ | SPI_NOR_QUAD_READ)
                .fixups = &w25q256_fixups },

```

linusw, 2024-10-28:
I added some commit messages and sent the patch for stable inclusion let's see if it works out.

RussellSenior, 2024-11-04:
Fixed with 7da9d6bfe017bc243ac70b83dde1761dad3d3a02

closed 2024-11-04 by RussellSenior