openwrt/openwrt #16797: It doesn't support the ASUS AX4200Q, hoping it will be supported.

xiaoxiudog, 2024-10-27:
### Describe the bug

It doesn't support the ASUS AX4200Q, hoping it will be supported.

### OpenWrt version

*

### OpenWrt release

*

### OpenWrt target/subtarget

*

### Device

AX4200Q

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-10-27:
You have to discuss that in forums. There is no wrong in existing software in your report. Google denies knowledge of such device.

xiaoxiudog, 2024-10-27:
> You have to discuss that in forums. There is no wrong in existing software in your report. Google denies knowledge of such device.

There has been discussion on the forum, but no conclusion. The AX4200 is supported, and the AX4200Q only adds a 2.5G network port. Is it possible to modify the OpenWrt configuration and compile it myself? If so, please let me know what to modify.

brada4, 2024-10-27:
Forum, please, sure it is possible to add code submitted by you, but not by claiming an issue in empty place.