openwrt/openwrt #16805: Can't Find Balia, Lia, Olia congestion control selection in menuconfig

iamdivyanshverma, 2024-10-28:
### Describe the bug

Hi there, I have build a custom OpenWrt image where i would like to test Congestion control like Balia, Lia, Olia mentioned on "https://multipath-tcp.org/pmwiki.php/Users/ConfigureMPTCP" but in kernel_menuconfig only these congestion controls are available

### OpenWrt version

snapshot

### OpenWrt release

snapshot

### OpenWrt target/subtarget

bcm2711

### Device

rpi4

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Spudz76, 2024-10-28:
If you search (/) for "olia" or such it will tel;l you which other options must be set (dependencies) before it shows up.

Or perhaps it's not in mainstream 6.1.71 and requires patches.

Spudz76, 2024-10-28:
Well it's not in 6.6.x either so it's probably patch-only for now (still under testing/development)

Spudz76, 2024-10-28:
See also https://openwrt.org/docs/guide-user/network/mptcp

Spudz76, 2024-10-28:
And the page you linked is for v5.4 and older kernels.

https://mptcp.dev/ is for the v5.6 and newer stuff, and I can't find anything about those congestion controls.

brada4, 2024-10-29:
I see no reason for such options appearing in openwrt
https://github.com/search?q=repo%3Atorvalds%2Flinux%20balia&type=code
Ask whomever told about them to you to mainstream them.

iamdivyanshverma, 2024-11-03:
Thank you everyone for the insights on congestion control algorithms. We are working in an environment where multiple 4G connections are used for multipath aggregation, and that the link latency and bandwidth fluctuate rapidly occur in our setup. Because of the varying latency and bw, we are currently experiencing overall throughput lower than the capacity of individual links at times. We are optimistic that CCA like OLIA will help address this issue by dynamically adjusting the congestion window according to each link's latency and bandwidth. Could you point me in the right direction for building the congestion control algorithm for MPTCP on the Mainstream kernel Any guidance on resources or steps would be greatly appreciated!
Thanks.