openwrt/openwrt #16818: crippled LAN to LAN file transfer (MR8300 with snapshot)

badulesia, 2024-10-29:
### Describe the bug

With snapshot, LAN to LAN file transfers capped at 100 MB/s, with a huge load average (>0.80). LAN to LAN file transfers should reach 115 MB/s (full Gb capability) with a low load average.

### OpenWrt version

main

### OpenWrt release

r27935-bcd95cb9c4

### OpenWrt target/subtarget

ipq40xx

### Device

Linksys MR8300

### Image kind

Self-built image

### Steps to reproduce

Run a snapshot with default settings. WAN is not connected. Perform file tranfers between two wired computers (LAN <-> LAN). I used two high-end PCs running Win11 24H2.

### Actual behaviour

Huge load average >0.80
bandwidth capped at 100 MB/s

### Expected behaviour

Low load average <0.10
bandwith at 115 MB/s (full Gb/s capability)
This is what the same device gives when running on 23.05.5/

### Additional info

Image is build using with packages : luci luci-theme-material luci-app-advanced-reboot -luci-proto-ppp
also non-CT wifi drivers, which are not involved in the test.
I'm a monitoring this behavior for several months now, it was the same back in spring 2024 when snapshots where running kernel 6.1. Performances were even worst at this time.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

raenye, 2024-10-29:
Do you see the same behaviour with `iperf3` to/from the router?
Is there any routing involved? (L2 switch should not involve the CPU IMHO)

badulesia, 2024-10-30:
Hi.
I have made some tests with iperf3. PC to router give 945 Mb/s, which is plain gigabit capability. For each PCs. PC1 to PC2 gives 840 Mb/s (load average 0.40). 

There is no routing involved, just plain LAN to LAN. WAN is disconnected anyhow.
I also agree that the switch should not involve the CPU. While running 23.05 (kernel 5.15) there is plain LAN Gb/s capability between PCs and low CPU usage. 

This misbehavior was already present with kernel 6.1 and so 6.6. Something deep has changed. I wonder if other ipq40xx devices are involved.

badulesia, 2024-10-30:
23.05 already uses DSA.

raenye, 2024-10-30:
> PC to router gives 945 Mb/s, which is plain gigabit capability

Did you also try the reverse direction (`-R`) and duplex (`--bidir`)?
840 vs. 945 is 13% less, in line with the file transfer performance decrease.

Is there a difference in the rate of interrupts between kernels 5.15 and 6.6?
(compare `/proc/interrupts` before and after transfers)

badulesia, 2024-10-30:
With reverse direction, speed is 820 Mb/s.

Interrupts results are very different !  c080000.ethernet, function call interrupts
These results are monitored while doing iperf3 between two PCs.
Kernel 5.15
```
root@MR8300:~# cat /proc/interrupts
           CPU0       CPU1       CPU2       CPU3
 26:       3479      10136      16134       4562     GIC-0  20 Level     arch_timer
 30:          3          0          0          0     GIC-0 270 Level     bam_dma
 31:          0          0          0          0     GIC-0 239 Level     bam_dma
 32:          2          0          0          0     GIC-0 139 Level     msm_serial0
 34:      46495          0          0          0     GIC-0 133 Level     bam_dma
 51:       5570          0          0          0     GIC-0 200 Level     ath10k_ahb
 68:       3291          0          0          0     GIC-0 201 Level     ath10k_ahb
 69:        354          0          0          0     GIC-0  97 Edge      c080000.ethernet:txq0
 73:        802          0          0          0     GIC-0 101 Edge      c080000.ethernet:txq4
 77:        332          0          0          0     GIC-0 105 Edge      c080000.ethernet:txq8
 81:        434          0          0          0     GIC-0 109 Edge      c080000.ethernet:txq12
 85:        377          0          0          0     GIC-0 272 Edge      c080000.ethernet:rxq0
 87:        258          0          0          0     GIC-0 274 Edge      c080000.ethernet:rxq2
 89:        222          0          0          0     GIC-0 276 Edge      c080000.ethernet:rxq4
 91:        782          0          0          0     GIC-0 278 Edge      c080000.ethernet:rxq6
102:          0          0          0          0   PCI-MSI   0 Edge      aerdrv
103:          0          0          0          0   msmgpio  50 Edge      keys
104:          0          0          0          0   msmgpio  18 Edge      keys
105:          0          0          0          0     GIC-0 164 Level     xhci-hcd:usb1
106:          0          0          0          0     GIC-0 168 Level     xhci-hcd:usb3
107:       4583          0          0          0   PCI-MSI 524288 Edge      ath10k_pci
IPI0:          0          0          0          0  CPU wakeup interrupts
IPI1:          0          0          0          0  Timer broadcast interrupts
IPI2:        738        932        711        889  Rescheduling interrupts
IPI3:       3689       6640       9466       6309  Function call interrupts
IPI4:          0          0          0          0  CPU stop interrupts
IPI5:          1          0          0          0  IRQ work interrupts
IPI6:          0          0          0          0  completion interrupts
Err:          0

```
Kernel 6.6
```
root@OpenWrt:~# cat /proc/interrupts
           CPU0       CPU1       CPU2       CPU3
 26:      18655       9167      11196      27420 GIC-0  20 Level     arch_timer
 29:          5          0          0          0 GIC-0 270 Level     bam_dma
 30:          0          0          0          0 GIC-0 239 Level     bam_dma
 31:      51238          0          0          0 GIC-0 133 Level     bam_dma
 32:          3          0          0          0 GIC-0 139 Level     msm_serial0
 35:          0          0          0          0   PCI-MSI   0 Edge      aerdrv
 36:      20649          0          0          0 GIC-0  97 Edge      c080000.ethernet:txq0
 40:      72375          0          0          0 GIC-0 101 Edge      c080000.ethernet:txq4
 44:      20410          0          0          0 GIC-0 105 Edge      c080000.ethernet:txq8
 48:     134762          0          0          0 GIC-0 109 Edge      c080000.ethernet:txq12
 52:     128310          0          0          0 GIC-0 272 Edge      c080000.ethernet:rxq0
 54:      31221          0          0          0 GIC-0 274 Edge      c080000.ethernet:rxq2
 56:     101437          0          0          0 GIC-0 276 Edge      c080000.ethernet:rxq4
 58:     155102          0          0          0 GIC-0 278 Edge      c080000.ethernet:rxq6
 60:          0          0          0          0   msmgpio  50 Edge      keys
 61:          0          0          0          0   msmgpio  18 Edge      keys
 62:          0          0          0          0 GIC-0 164 Level     xhci-hcd:usb1
 63:          0          0          0          0 GIC-0 168 Level     xhci-hcd:usb3
 64:         39          0          0          0   PCI-MSI 524288 Edge      ath10k_pci
 65:         21          0          0          0 GIC-0 200 Level     ath10k_ahb
 66:         22          0          0          0 GIC-0 201 Level     ath10k_ahb
IPI0:          0          0          0          0  CPU wakeup interrupts
IPI1:          0          0          0          0  Timer broadcast interrupts
IPI2:       1291       1287       1260       1125  Rescheduling interrupts
IPI3:     180603     226583     177650     215985  Function call interrupts
IPI4:          0          0          0          0  CPU stop interrupts
IPI5:          0          0          0          0  IRQ work interrupts
IPI6:          0          0          0          0  completion interrupts
```

raenye, 2024-10-30:
Seems to be the source of CPU load, but sadly I don't know why it happens or how to solve it :(
Not a solution but perhaps a workaround: try spreading the load over all cores/threads(*) with `irqbalance`.

(*) IIRC this is a 2-core 4-thread CPU.

badulesia, 2024-10-31:
Hi.
I figured it at myself while reading the result. Good idea thought to have a look at this.
c080000.ethernet IRQs are indeed all done on CPU0. I don't know if IRQ balance will distribute them, I'll perform a test later today.
Real question now is why these values are much greater than with kernel 5.15?

brada4, 2024-10-31:
4 interrupts each direction probably are intended to be irqbalanced on 4 CPU-s, at leat thats how PC servers work

raenye, 2024-11-01:
@Ansuel @robimarko Any idea here? (see also related discussion on #14121)

badulesia, 2024-11-08:
#14121 seems to be solved. Last time I checked I had no drop to 100Mb/s. Since them I have upgraded WAN to a Gb fiber, so I can't reproduce the issue anymore. On that occasion, I discovered the LAN to LAN issue. The device was still on kernel 6.1:  I waited for kernel 6.6 to be well established to reperform LAN to LAN tests.

badulesia, 2025-04-17:
Solved in 24.10.1 (by a commit since 24.10.0).