openwrt/openwrt #16821: Unknown beug

tacal666, 2024-10-30:
### Describe the bug

hi, please i have this bug on my system log
can some one explain it ?

### OpenWrt version

r24106-10cc5fcd00

### OpenWrt release

23.05.5

### OpenWrt target/subtarget

x86/64

### Device

Intel Geminilake

### Image kind

Official downloaded image

### Additional info

Wed Oct 30 14:05:06 2024 kern.warn kernel: [   40.976285] ------------[ cut here ]------------
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   40.985839] WARNING: CPU: 1 PID: 0 at 0xffffffffa0483659
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   40.996126] Modules linked in: ath9k ath9k_common pppoe ppp_async nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet mt7921u mt7921s mt7921e mt7921_common ath9k_hw ath10k_pci ath10k_core ath pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_ct nft_counter nft_compat nft_chain_nat nf_tables mt792x_usb mt792x_lib mt76x2e mt76x2_common mt76x02_lib mt7603e mt76_usb mt76_sdio mt76_connac_lib mt76 mac80211 ipt_REJECT cfg80211 zstd xt_time xt_tcpudp xt_state xt_quota xt_pkttype xt_physdev xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_cgroup xt_addrtype xt_TCPMSS xt_TARPIT xt_REDIRECT xt_MASQUERADE xt_LOG ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda slhc r8169 nfnetlink nf_reject_ipv4 nf_log_syslog nf_flow_table lzo_rle lzo iptable_nat iptable_mangle
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   40.996319]  iptable_filter ip_tables igc forcedeth e1000e crc_ccitt compat_xtables compat br_netfilter bnx2 i2c_dev snd_pcsp ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ixgbe e1000 amd_xgbe mdio_devres mdio veth tun snd_rawmidi snd_seq_device snd_pcm_oss snd_mixer_oss snd_hwdep snd_compress snd_pcm snd_timer snd soundcore nls_utf8 ena xxhash_generic sha512_ssse3 sha512_generic seqiv jitterentropy_rng drbg hmac cmac blake2b_generic crypto_acompress nls_iso8859_1 nls_cp437 uas xhci_plat_hcd fsl_mph_dr_of ehci_platform ehci_fsl igb vfat fat btrfs zstd_decompress zstd_compress xxhash xor raid6_pq lzo_decompress lzo_compress libcrc32c button_hotplug tg3 realtek libphy
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   41.209242] CPU: 1 PID: 0 Comm: swapper/1 Not tainted 5.15.167 #0
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   41.218725] Hardware name: Intel Geminilake/Geminilake, BIOS 5.13 08/05/2022
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   41.229437] RIP: 0010:0xffffffffa0483659
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   41.237032] Code: ff ff 41 be 04 00 00 00 be 04 00 00 00 eb 8d 0f 0b e9 9d fe ff ff 0f 0b e9 e9 fe ff ff 4c 89 e7 e8 fc b1 49 e1 e9 71 fe ff ff <0f> 0b eb 8a 0f 1f 00 bf 01 00 00 00 f6 46 7f 01 74 22 48 89 f0 48
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   41.263607] RSP: 0018:ffffc900000f8968 EFLAGS: 00010246
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   41.275797] RAX: 0000000000000190 RBX: ffff888104b1b200 RCX: 0000000000000102
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   41.290004] RDX: 0000000000000000 RSI: 0000000000000000 RDI: ffff888104b1b200
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   41.304256] RBP: ffffc900000f8988 R08: 0000000000000000 R09: 00000000fb0000e0
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   41.318503] R10: fffffffffffffff0 R11: 0000000000000002 R12: ffff8881064a1400
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   41.332699] R13: 0000000000000000 R14: 0000000000000000 R15: ffff888101842000
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   41.346845] FS:  0000000000000000(0000) GS:ffff88817bc80000(0000) knlGS:0000000000000000
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   41.362014] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   41.374661] CR2: 000000c000849018 CR3: 000000000340a000 CR4: 0000000000350ee0
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   41.388838] Call Trace:
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   41.397956]  <IRQ>
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   41.406552]  ? show_regs.part.0+0x1e/0x24
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   41.417266]  ? show_regs.cold+0x8/0xd
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   41.427547]  ? __warn+0x74/0xf0
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   41.437255]  ? 0xffffffffa0483659
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   41.447126]  ? report_bug+0x86/0xa0
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   41.457250]  ? handle_bug+0x38/0x90
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   41.467374]  ? exc_invalid_op+0x18/0x70
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   41.477842]  ? asm_exc_invalid_op+0x1b/0x20
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   41.488718]  ? 0xffffffffa0483659
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   41.498685]  ? 0xffffffffa04835d3
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   41.508665]  nf_hook_slow+0x3c/0xc0
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   41.518829]  br_pass_frame_up+0x101/0x180
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   41.529611]  ? br_port_flags_change+0x90/0x90
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   41.540731]  br_handle_frame_finish+0x360/0x4f0
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   41.552011]  ? br_pass_frame_up+0x180/0x180
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   41.562918]  br_nf_hook_thresh+0xb4/0x14d0 [br_netfilter]
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   41.575136]  ? br_pass_frame_up+0x180/0x180
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   41.586047]  br_nf_hook_thresh+0x244/0x14d0 [br_netfilter]
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   41.598411]  ? br_pass_frame_up+0x180/0x180
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   41.609304]  ? nf_nat_ipv6_unregister_fn+0xd6/0x240 [nf_nat]
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   41.621757]  ? nf_hook_slow+0x3c/0xc0
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   41.632015]  br_nf_hook_thresh+0x9fa/0x14d0 [br_netfilter]
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   41.644201]  ? br_nf_hook_thresh+0xe0/0x14d0 [br_netfilter]
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   41.656475]  br_handle_frame+0x1ac/0x400
[... 13 lines cut ...]
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   41.797710]  irq_exit_rcu+0x60/0x90
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   41.807438]  common_interrupt+0x86/0xa0
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   41.817478]  </IRQ>
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   41.825651]  <TASK>
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   41.833769]  asm_common_interrupt+0x27/0x40
Wed Oct 30 14:05:06 2024 kern.warn kernel: [   41.844133] RIP: 0010:cpuidle_enter_state+0xbb/0x2d0
Wed Oct 30 14:05:07 2024 kern.warn kernel: [   41.855147] Code: 4f 8a ff 65 8b 3d 75 e1 7c 7e e8 a0 4d 8a ff 31 ff 49 89 c6 e8 86 5a 8a ff 80 7d d7 00 0f 85 ab 01 00 00 fb 66 0f 1f 44 00 00 <45> 85 ff 0f 88 dc 00 00 00 49 63 cf 4c 8b 55 c8 48 8d 04 49 48 8d
Wed Oct 30 14:05:07 2024 kern.warn kernel: [   41.880932] RSP: 0018:ffffc900000b3e70 EFLAGS: 00000246
Wed Oct 30 14:05:07 2024 kern.warn kernel: [   41.892059] RAX: ffff88817bca2580 RBX: ffff88817bcaab00 RCX: 000000000000001f
Wed Oct 30 14:05:07 2024 kern.warn kernel: [   41.905069] RDX: 000000098a5dfb6d RSI: 00000000401a41a4 RDI: 0000000000000000
Wed Oct 30 14:05:07 2024 kern.warn kernel: [   41.917831] RBP: ffffc900000b3ea8 R08: 0000000000000000 R09: 000000000000000c
Wed Oct 30 14:05:07 2024 kern.warn kernel: [   41.930370] R10: 000000098a04d9eb R11: 0000000000000000 R12: 0000000000000002
Wed Oct 30 14:05:07 2024 kern.warn kernel: [   41.942673] R13: ffffffff824cd820 R14: 000000098a5dfb6d R15: 0000000000000002
Wed Oct 30 14:05:07 2024 kern.warn kernel: [   41.954693]  cpuidle_enter+0x2f/0x50
Wed Oct 30 14:05:07 2024 kern.warn kernel: [   41.962743]  call_cpuidle+0x1e/0x40
Wed Oct 30 14:05:07 2024 kern.warn kernel: [   41.970654]  do_idle+0x15c/0x180
Wed Oct 30 14:05:07 2024 kern.warn kernel: [   41.978221]  cpu_startup_entry+0x18/0x20
Wed Oct 30 14:05:07 2024 kern.warn kernel: [   41.986533]  start_secondary+0xf9/0x110
Wed Oct 30 14:05:07 2024 kern.warn kernel: [   41.994568]  secondary_startup_64_no_verify+0xb0/0xbb
Wed Oct 30 14:05:07 2024 kern.warn kernel: [   42.003750]  </TASK>
Wed Oct 30 14:05:07 2024 kern.warn kernel: [   42.009821] ---[ end trace 0e066ccaf72ddc57 ]---

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-10-31:
br_nf_hook is not reachable in default OpenWRT config - what is your change that triggers the warning?

tacal666, 2024-11-03:
@brada4 i made only some port forward rules, however yesterday i swap to SNAPSHOT version and until now no bug reported !