openwrt/openwrt #16823: The wan and wan6 interfaces cannot obtain IP addresses

small-white-rabbit, 2024-10-30:
### Describe the bug

Updated the latest code from the openwrt library, from 5.15 to 6.6. After compiling and upgrading, it was found that WAN and WAN6 could not obtain IP addresses. The reset or img file flashing problem was the same.

[image]
[image]

log
```
l
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
Wed Oct 30 10:15:38 2024 kern.notice kernel: [    0.000000] Linux version 6.6.58 (yh@) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r27970-36834ea340) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 SMP Wed Oct 30 10:15:34 2024
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.000000] Machine model: Netgear WAX206
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.000000] earlycon: uart8250 at MMIO32 0x0000000011002000 (options '')
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.000000] printk: bootconsole [uart8250] enabled
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.000000] OF: reserved mem: 0x0000000042ff0000..0x0000000042ffffff (64 KiB) map non-reusable ramoops@42ff0000
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.000000] OF: reserved mem: 0x0000000043000000..0x000000004302ffff (192 KiB) nomap non-reusable secmon@43000000
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.000000] Zone ranges:
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.000000]   DMA      [mem 0x0000000040000000-0x000000005fffffff]
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.000000]   DMA32    empty
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.000000]   Normal   empty
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.000000] Movable zone start for each node
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.000000] Early memory node ranges
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000040000000-0x0000000042ffffff]
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000043000000-0x000000004302ffff]
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000043030000-0x000000005fffffff]
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x000000005fffffff]
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.000000] psci: probing for conduit method from DT.
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.000000] psci: PSCIv0.2 detected in firmware.
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.000000] psci: Using standard PSCI v0.2 function IDs
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.000000] psci: Trusted OS migration not required
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.000000] percpu: Embedded 18 pages/cpu s34984 r8192 d30552 u73728
Wed Oct 30 10:15:38 2024 kern.debug kernel: [    0.000000] pcpu-alloc: s34984 r8192 d30552 u73728 alloc=18*4096
Wed Oct 30 10:15:38 2024 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.000000] Detected VIPT I-cache on CPU0
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.000000] CPU features: kernel page table isolation disabled by kernel configuration
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.000000] CPU features: detected: ARM erratum 843419
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.000000] alternatives: applying boot alternatives
Wed Oct 30 10:15:38 2024 kern.notice kernel: [    0.000000] Kernel command line: earlycon=uart8250,mmio32,0x11002000 console=ttyS0,115200n8 swiotlb=512
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.000000] Dentry cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.000000] Inode-cache hash table entries: 32768 (order: 6, 262144 bytes, linear)
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129024
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.000000] software IO TLB: area num 2.
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.000000] software IO TLB: mapped [mem 0x000000005f4c0000-0x000000005f5c0000] (1MB)
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.000000] Memory: 499336K/524288K available (8896K kernel code, 942K rwdata, 2548K rodata, 448K init, 295K bss, 24952K reserved, 0K cma-reserved)
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.000000] 	Tracing variant of Tasks RCU enabled.
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.000000] Root IRQ handler: gic_handle_irq
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.000000] GIC: Using split EOI/Deactivate mode
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.000000] arch_timer: cp15 timer(s) running at 12.50MHz (phys).
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x2e2049cda, max_idle_ns: 440795202628 ns
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.000000] sched_clock: 56 bits at 13MHz, resolution 80ns, wraps every 4398046511080ns
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.008271] Calibrating delay loop (skipped), value calculated using timer frequency.. 25.00 BogoMIPS (lpj=125000)
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.018669] pid_max: default: 32768 minimum: 301
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.026093] Mount-cache hash table entries: 1024 (order: 1, 8192 bytes, linear)
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.033440] Mountpoint-cache hash table entries: 1024 (order: 1, 8192 bytes, linear)
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.043839] RCU Tasks Trace: Setting shift to 1 and lim to 1 rcu_task_cb_adjust=1.
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.051575] rcu: Hierarchical SRCU implementation.
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.056386] rcu: 	Max phase no-delay instances is 1000.
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.062051] smp: Bringing up secondary CPUs ...
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.067012] Detected VIPT I-cache on CPU1
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.067089] CPU1: Booted secondary processor 0x0000000001 [0x410fd034]
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.067155] smp: Brought up 1 node, 2 CPUs
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.081838] SMP: Total of 2 processors activated.
Wed Oct 30 10:15:38 2024 kern.info kernel: [    0.086554] CPU features: detected: 32-bit EL0 Support
[... 649 lines cut ...]
`

### OpenWrt version

24.298

### OpenWrt release

24.298

### OpenWrt target/subtarget

24.298

### Device

wax206

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

M95D, 2024-10-31:
WAN port looks disconnected.

small-white-rabbit, 2024-10-31:
> WAN 端口看上去已断开连接。

As long as you change back to the original firmware, you can use it normally. I don't know how to solve this problem. I also tried it in the early 6.1 version, but the problem was the same. I thought it was because the version had just been upgraded so some bugs were normal, but I tried it again in 6.6 and it still had the same problem. I guessed that there might be a problem with the firmware compilation.

small-white-rabbit, 2024-11-02:
> WAN is RTL8221B 2.5 GbE. Did something change in the Realtek driver?
> 
> Could you provide the output of `ethtool -m` on the WAN interface? (from `ethtool-full` package)

```
**5.15**
root@OpenWrt:~# ethtool eth0
Settings for eth0:
        Supported ports: [ MII ]
        Supported link modes:   2500baseT/Full
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  2500baseT/Full
        Advertised pause frame use: Symmetric Receive-only
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Link partner advertised link modes:  2500baseT/Full
        Link partner advertised pause frame use: Symmetric
        Link partner advertised auto-negotiation: No
        Link partner advertised FEC modes: Not reported
        Speed: 2500Mb/s
        Duplex: Full
        Auto-negotiation: on
        Port: MII
        PHYAD: 0
        Transceiver: internal
        Current message level: 0x000000ff (255)
                               drv probe link timer ifdown ifup rx_err tx_err
        Link detected: yes

root@OpenWrt:~# ethtool -i eth0
driver: mtk_soc_eth
version: 5.15.138
firmware-version: 
expansion-rom-version: 
bus-info: 1b100000.ethernet
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: yes
supports-priv-flags: no
root@OpenWrt:~# 

root@OpenWrt:~# ethtool -m
ethtool: bad command line argument(s)
For more information run ethtool -h
root@OpenWrt:~# ethtool -m eth0
netlink error: Not supported
root@OpenWrt:~# 

**6.6**
root@OpenWrt:~# ethtool eth0
Settings for eth0:
        Supported ports: [ MII ]
        Supported link modes:   2500baseT/Full
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  2500baseT/Full
        Advertised pause frame use: Symmetric Receive-only
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Link partner advertised link modes:  2500baseT/Full
        Link partner advertised pause frame use: Symmetric
        Link partner advertised auto-negotiation: No
        Link partner advertised FEC modes: Not reported
        Speed: 2500Mb/s
        Duplex: Full
        Auto-negotiation: on
        Port: MII
        PHYAD: 0
        Transceiver: internal
        Current message level: 0x000000ff (255)
                               drv probe link timer ifdown ifup rx_err tx_err
        Link detected: yes

root@OpenWrt:~# ethtool -i eth0
driver: mtk_soc_eth
version: 6.6.58
firmware-version: 
expansion-rom-version: 
bus-info: 1b100000.ethernet
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: yes
supports-priv-flags: no
root@OpenWrt:~# 
root@OpenWrt:~# ethtool -m
ethtool: bad command line argument(s)
For more information run ethtool -h
root@OpenWrt:~# 

``` 
I compared them and they seem to be the same, but I don't know why 6.6 doesn't work.

raenye, 2024-11-02:
Hmm, no idea :(

M95D, 2024-11-03:
WAN may or may not be eth0 in the new version. I can't tell. Did you keep the configuration when you updated? If so, try a reset to defaults. 

If that doesn't work, `ifconfig -a` would tell us more information. Also, from Luci / Network / Interfaces / Devices tab, show us the popups that appears over wan and eth0 icons.

Other things you may try:
- connect and disconnect the cable and see if Luci WAN port status changes, also if ethool report changes "Link detected: yes / no"
- install and run tcpdump to see if there is traffic from wan or eth0
- while tcpdump is running, also try to run a dhcp client manually (busybox udhcpc should be available) and see if the request is sent via wan and/or eth0 and if there is a reply from the server
- connect WAN to a PC/laptop, assign static IP addresses to both, and see if ping works (pay attention to the Windows firewall; it might block ping)

small-white-rabbit, 2024-11-03:
> WAN may or may not be eth0 in the new version. I can't tell. Did you keep the configuration when you updated? If so, try a reset to defaults.
>
> If that doesn't work, `ifconfig -a` would tell us more [...]
[image]
[image]
[image]

I reset the system but the result is the same. I also tried to switch the lede firmware. Although the result is the same, there is a prompt at the bottom saying that the device is not recognized [I forgot to take a screenshot, probably because it is not recognized or the device does not exist]. So I think it may be caused by the driver problem. I set the port to a static IP of 192.168.22.1, and it can be pinged through wireless and wired devices. Because it cannot connect to the Internet, no other tests were performed later.

M95D, 2024-11-04:
OK, so this device has an integrated switch connected to eth0 (SoC interface, internal) and lan1-lan4 and wan as external ports. As you are connecting to it via the same switch and internal interface, it can't be a driver problem.

eth0 is an internal port, so it shouldn't be configured. Only wan configuration matters. Show us `ethtool wan`.

raenye, 2024-11-04:
Does it make sense to have the same MAC for WAN and LAN? in my devices they're different.

M95D, 2024-11-04:
Yes, it does. Lan1-lan4 and wan are not etherent interfaces. They are just DSA switch ports, so they don't have MACs at all. Linux DSA creates virtual interfaces for each port and gives them the MAC of the real interface (SoC's eth0) to which the switch is connected.

raenye, 2024-11-04:
> Yes, it does. Lan1-lan4 and wan are not ethernet interfaces. They are just switch ports. So they don't have MACs at all. Linux DSA creates virtual interfaces for each port and gives them the MAC of the real interface (SoC's eth0) to which the switch is connected.

But in many devices the MAC address assigned to the WAN port is changed (see, e.g., `/etc/board.json`) even when all physical ports are backed by the same real interface.

raenye, 2024-11-04:
> and the problem was solved.

Hooray!

> Now I don't know which compilation file should be modified to change the default settings so that it can work normally after compilation.

I think this might involve changes to the Device Tree; somehow in 6.6 auto-negotiation is no longer active.

M95D, 2024-11-04:
That's weird. Why would auto-negotiation be disabled only on wan port. The lan seems to be working.
@small-white-rabbit, please show us `ethtool lan1`, or the port where your PC is connected.

small-white-rabbit, 2024-11-04:
> 我认为这可能涉及设备树的变化；不知何故，6.6 中的自动协商不再处于活动状态。

Thanks, thanks to your reply I was able to try and finally solve it.

small-white-rabbit, 2024-11-04:
> 这很奇怪。为什么只在 WAN 端口上禁用自动协商。LAN 似乎正在运行。 @small-white-rabbit，请向我们显示`ethtool lan1`，或者您的电脑所连接的端口。

```
root@OpenWrt:~# ethtool lan1
Settings for lan1:
        Supported ports: [ TP    MII ]
        Supported link modes:   10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
        Advertised pause frame use: Symmetric Receive-only
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Speed: Unknown!
        Duplex: Unknown! (255)
        Auto-negotiation: on
        master-slave cfg: preferred slave
        master-slave status: unknown
        Port: Twisted Pair
        PHYAD: 1
        Transceiver: external
        MDI-X: Unknown
        Supports Wake-on: d
        Wake-on: d
        Link detected: no
root@OpenWrt:~# 
``` 
Yes, it looks like the LAN port is set to open automatically.

M95D, 2024-11-04:
Nevermind that.
It seems that the 2500 Gbps is a known problem of this router:
https://openwrt.org/toh/netgear/wax206

raenye, 2024-11-04:
(If you're happy with the result, please close the issue)

M95D, 2024-11-04:
No, I think the title should be changed to better describe the bug and the issue should be left open until autonegotiation is fixed.
I propose this title:
`Netgear WAX206: autonegotiation not available on wan port with kernel 6.6`

M95D, 2024-11-06:
The devicetree only has two commits:
This one, https://github.com/openwrt/openwrt/commit/3221128639e43d5bcf3a06eda769350a75b3dc72 by @dangowrt, which certainly seems related, but it's already included 23.05 branch. So it can't be it.
And this one: https://github.com/openwrt/openwrt/commit/9dae0ac0bc61385b77639fa1397c0fc302a0abeb by @namiltd.

abajk, 2024-11-07:
I confirm that the bug exists. From my investigation, it turned out that the cause is a [commit](https://github.com/openwrt/openwrt/commit/ca4469045f6a4845dbf8aeabd4b04b421209f66e). This was reported in the [thread](https://github.com/openwrt/openwrt/commit/ca4469045f6a4845dbf8aeabd4b04b421209f66e#commitcomment-147891558). The last working revision is https://github.com/openwrt/openwrt/commit/f20987c161bde4f0d9713de9e0c8514fb0cdb341.

abajk, 2024-11-07:
By the way. There is no need to force speed. A temporary workaround is to enable autonegotiation:
~~~
ethtool -s wan autoneg on
~~~

small-white-rabbit, 2024-11-16:
I found a new problem. Because the hard disk space is too small, I deleted the previous compilation environment. After re-git clone and compile, the wan port disappeared. The device does not exist below the interface. It seems that the driver is not installed this time. I don’t know why. How to solve it?

small-white-rabbit, 2024-11-16:
> The devicetree only has two commits
I found a new problem. Because the hard disk space is too small, I deleted the previous compilation environment. After re-git clone and compile, the wan port disappeared. The device does not exist below the interface. It seems that the driver is not installed this time. I don’t know why. How to solve it?

small-white-rabbit, 2024-11-16:
> The devicetree only has two commits: This one, [3221128](https://github.com/openwrt/openwrt/commit/3221128639e43d5bcf3a06eda769350a75b3dc72) by @dangowrt, which seems related, but it's already included 23.05 branch. So it can't be it. And this one: [9dae0ac](https://github.com/openwrt/openwrt/commit/9dae0ac0bc61385b77639fa1397c0fc302a0abeb) by @namiltd, which doesn't affect wan port alone, so it's less likely to be the cause.

I found a new problem. Because the hard disk space is too small, I deleted the previous compilation environment. After re-git clone and compile, the wan port disappeared. The device does not exist below the interface. It seems that the driver is not installed this time. I don’t know why. How to solve it?

raenye, 2024-11-16:
Maybe you dropped the Realtek driver by mistake? You should compare `.config` between the builds.

small-white-rabbit, 2024-11-16:
> Maybe you dropped the Realtek driver by mistake? You should compare `.config` between the builds.

I tried to compile with the config files that I had backed up in versions 6.1 and 5.4, but the problem was the same. I remember that I had guessed that it might be a driver problem, so I copied and pasted the driver file from version 5.4 into 6.6 to compile [it might also be extracted from an immortalwrtARM that was previously repaired specifically for wax206], but because of my bad memory, I forgot how to do it, and only successfully compiled version 6.6.58. I mentioned before that I tried lede successfully but it also prompted that the device did not exist. Now I don’t know how to add the driver.

commit 730db6b8 (2025-01-26): generic: fix probe issues with RealTek RTL8221B PHYs

Import patch "net: phy: realtek: mark existing MMDs as present"

    When using Clause-45 mode to access RealTek RTL8221B 2.5G PHYs some
    versions of the PHY fail to report the MMDs present on the PHY.
    Mark MMDs PMAPMD, PCS and AN which are always existing according to
    the datasheet as present to fix that.

Fixes: #16823, #17183, #17232

Tested-by: Aleksander Jan Bajkowski <olek2@wp.pl>
Tested-by: Juan Pedro Paredes Caballero <juanpedro.paredes@gmail.com>
commit d30773ab (2025-01-26): generic: fix probe issues with RealTek RTL8221B PHYs

Import patch "net: phy: realtek: mark existing MMDs as present"

    When using Clause-45 mode to access RealTek RTL8221B 2.5G PHYs some
    versions of the PHY fail to report the MMDs present on the PHY.
    Mark MMDs PMAPMD, PCS and AN which are always existing according to
    the datasheet as present to fix that.

Fixes: #16823, #17183, #17232

Tested-by: Aleksander Jan Bajkowski <olek2@wp.pl>
Tested-by: Juan Pedro Paredes Caballero <juanpedro.paredes@gmail.com>
(cherry picked from commit 730db6b893748517949b1d3a0c0cb8b6c1288040)
closed 2025-01-26 by openwrt-bot