openwrt/openwrt #16836: TP-Link Archer C2 v1 unstable Wi-Fi (twice per year)

George-TL, 2024-11-01:
### Describe the bug

Few times during the year wireless in AP mode stops working.
All devices (tablets, phones, laptops) are unable to connect.
SSID is broadcasted but connection fails without any error message.
Associated Stations shows list of devices which are trying to connect.
I'm using radio1 (Generic 802.11bgn) only.
Radio0 (MediaTek MT7610E 802.11acbgn) is disabled.
Restarting radio1 interface from WEB-UI solves the problem.

dmesg is full of:
```
ieee80211 phy1: rt2800_wait_bbp_rf_ready: Error - BBP/RF register access failed, aborting
ieee80211 phy1: rt2800_config_alc: Warning - RF busy while configuring ALC
```
Sparse errors:
```
ieee80211 phy1: rt2x00lib_rxdone_read_signal: Warning - Frame received with unrecognized signal, mode=0x0001, signal=0x010e, type=4
```
This might be important:
```
[629222.300961] ------------[ cut here ]------------
[629222.305797] WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:467 0x804265b0
[629222.313097] NETDEV WATCHDOG: eth0 (mtk_soc_eth): transmit queue 0 timed out
[629222.320284] Modules linked in: rt2800soc rt2800mmio rt2800lib pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet mt76x0e mt76x0_common rt2x00soc rt2x00mmio rt2x00lib rndis_host pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt76x02_lib mt76 mac80211 cfg80211 cdc_ether usbnet slhc nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_ipv6 nf_log_ipv4 nf_log_common nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c crc_ccitt compat ledtrig_usbport mii sha256_generic libsha256 seqiv jitterentropy_rng drbg hmac cmac leds_gpio ohci_platform ohci_hcd rtl8367b rtl8366_smi fsl_mph_dr_of ehci_platform ehci_fsl ehci_hcd gpio_button_hotplug usbcore nls_base usb_common crc32c_generic
[629222.404622] CPU: 0 PID: 0 Comm: swapper Not tainted 5.10.221 #0
[629222.410751] Stack : 806c0c20 800615b4 806d0000 805cc2b8 00000000 00000000 00000000 00000000
[629222.419401]         00000000 00000000 00000000 00000000 00000000 00000001 80c09db0 c3098ca0
[629222.428050]         80c09e48 00000000 00000000 80c09c58 00000038 802cef64 00000000 ffffffea
[629222.436698]         00000000 80c09c64 000002b5 8064f0c8 80c09d90 805a48a0 00000000 804265b0
[629222.445347]         00000009 00000000 08a7e5f4 806c0c34 00000000 8031e900 00000000 80810000
[629222.453997]         ...
[629222.456584] Call Trace:
[629222.456597] [<800615b4>] 0x800615b4
[629222.462831] [<802cef64>] 0x802cef64
[629222.466479] [<804265b0>] 0x804265b0
[629222.470124] [<8031e900>] 0x8031e900
[629222.473781] [<80006d28>] 0x80006d28
[629222.477429] [<80006d30>] 0x80006d30
[629222.481092] [<80026aa8>] 0x80026aa8
[629222.484769] [<804265b0>] 0x804265b0
[629222.488414] [<80026ba0>] 0x80026ba0
[629222.492074] [<804265b0>] 0x804265b0
[629222.495743] [<81ada21c>] 0x81ada21c [mac80211@e41af491+0x7faa0]
[629222.501883] [<80426344>] 0x80426344
[629222.505531] [<80071f94>] 0x80071f94
[629222.509179] [<800722d0>] 0x800722d0
[629222.512844] [<80064380>] 0x80064380
[629222.516496] [<805907e8>] 0x805907e8
[629222.520141] [<80068d60>] 0x80068d60
[629222.523799] [<80063c70>] 0x80063c70
[629222.527447] [<806e0000>] 0x806e0000
[629222.531107] [<80002af0>] 0x80002af0
[629222.534752] 
[629222.536361] ---[ end trace 8557444aa16bb616 ]---
```

### OpenWrt version

r20341-591b7e93d3

### OpenWrt release

22.03.7

### OpenWrt target/subtarget

ramips/mt7620

### Device

TP-Link Archer C2 v1

### Image kind

Official downloaded image

### Steps to reproduce

Just sit and wait.

### Actual behaviour

Wireless stops working.

### Expected behaviour

OpenWRT is a rock solid solution.

### Additional info

Have a nice day.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2024-11-01:
Unfortunately, the problem has been present for a long time on other devices with MT7620.
This script saves me:
```
#!/bin/sh

# A simple script to workaround MT7620 wireless hang issue via
# crontab schedule:
# * * * * * /sbin/check_radio.sh 

minilog=`logread 2>/dev/null | grep "Hardware restart was requested"`

(echo $minilog | grep -q "phy0: Hardware restart was requested") && phy0=true 

(echo $minilog | grep -q "phy1: Hardware restart was requested") && phy1=true

if [ "$phy0" == "true" ] || [ "$phy1" == "true" ]; then
    /etc/init.d/log restart

    [ "$phy0" == "true" ] && wifi down radio0
    [ "$phy1" == "true" ] && wifi down radio1

    sleep 2

    [ "$phy0" == "true" ] && wifi up radio0
    [ "$phy1" == "true" ] && wifi up radio1
fi
```

brada4, 2024-11-01:
Suggesting upgrade to 23.05.5, mt76 wifi got zillion of fixes by now.

George-TL, 2024-11-02:
@brada4
I've successfully updated to 23.05.
Will make new ticked when problem returns.
Thanks

closed 2024-11-02 by George-TL