openwrt/openwrt #16855: [24.10] Parallel image build error with initramfs

obsy, 2024-11-04:
### Describe the bug

When building with -j16, the image building process is stuck on initramfs 

### OpenWrt version

24.10-SNAPSHOT

### OpenWrt release

24.10-SNAPSHOT

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

all

### Image kind

Official downloaded image

### Steps to reproduce

```make -j16```

### Actual behaviour

When building with -j16, the image building process is stuck on initramfs 

### Expected behaviour

Generated images

### Additional info

I described qualcommax/ipq807x, but the same thing happens for others with initramfs, e.g. ath79/generic. 
Make -j1 works, but -j16 hangs.  Starts multiple copy processes that get stuck and leave temporary directories.
```
$ ls  build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6* | grep build_dir
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.03363680:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.057cd404:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.06646ecf:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.19ababe6:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.2458ae93:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.27076f27:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.4e2912fd:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.4f8e80a0:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.5310f19a:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.563bb6d5:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.6ac55ba6:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.9509eee9:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.a17c5372:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.b7ceb6df:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.c30e5112:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.ef8cd2b4:
```

```
$ ps ax | grep flock
2633022 pts/0    S+     0:00 flock /opt/openwrt-24.10/qualcommax/ipq807x/tmp/.gen-initramfs.4e2912fd.flock -c { ?{ [ ! -d "/opt/openwrt-24.10/qualcommax/ipq807x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.4e2912fd" ] || rm -rf /opt/openwrt-24.10/qualcommax/ipq807x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.4e2912fd; mkdir /opt/openwrt-24.10/qualcommax/ipq807x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.4e2912fd && cp -fpR -T /opt/openwrt-24.10/qualcommax/ipq807x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58 /opt/openwrt-24.10/qualcommax/ipq807x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.4e2912fd; touch /opt/openwrt-24.10/qualcommax/ipq807x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.4e2912fd/.config; rm -rf /opt/openwrt-24.10/qualcommax/ipq807x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.4e2912fd/usr/initramfs_data.cpio*; }; ??{ rm -f /opt/openwrt-24.10/qualcommax/ipq807x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.4e2912fd/.config.prev; mv /opt/openwrt-24.10/qualcommax/ipq807x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.4e2912fd/.config /opt/openwrt-24.10/qualcommax/ipq807x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.4e2912fd/.config.old; ?{ grep -v -e INITRAMFS -e CONFIG_RD_ -e CONFIG_BLK_DEV_INITRD /opt/openwrt-24.10/qualcommax/ipq807x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.4e2912fd/.config.old > /opt/openwrt-24.10/qualcommax/ipq807x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.4e2912fd/.config; echo 'CONFIG_BLK_DEV_INITRD=y' >> /opt/openwrt-24.10/qualcommax/ipq807x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.4e2912fd/.config; echo 'CONFIG_INITRAMFS_SOURCE="/opt/openwrt-24.10/qualcommax/ipq807x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/target-dir-4e2912fd /opt/openwrt-24.10/qualcommax/ipq807x/target/linux/generic/image/initramfs-base-files.txt"' >> /opt/openwrt-24.10/qualcommax/ipq807x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.4e2912fd/.config; }; echo "# CONFIG_INITRAMFS_PRESERVE_MTIME is not set" >> /opt/openwrt-24.10/qualcommax/ipq807x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.4e2912fd/.config; echo "CONFIG_INITRAMFS_ROOT_UID=1000" >> /opt/openwrt-24.10/qualcommax/ipq807x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.4e2912fd/.config; echo "CONFIG_INITRAMFS_ROOT_GID=1000" >> /opt/openwrt-24.10/qualcommax/ipq807x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.4e2912fd/.config;   echo "# CONFIG_INITRAMFS_FORCE is not set" >> /opt/openwrt-24.10/qualcommax/ipq807x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.4e2912fd/.config;  echo "# CONFIG_INITRAMFS_COMPRESSION_NONE is not set" >> /opt/openwrt-24.10/qualcommax/ipq807x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.4e2912fd/.config;    echo "# CONFIG_INITRAMFS_COMPRESSION_GZIP is not set" >> /opt/openwrt-24.10/qualcommax/ipq807x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.4e2912fd/.config;   echo "# CONFIG_RD_GZIP is not set" >> /opt/openwrt-24.10/qualcommax/ipq807x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.4e2912fd/.config;     echo "# CONFIG_INITRAMFS_COMPRESSION_BZIP2 is not set" >> /opt/openwrt-24.10/qualcommax/ipq807x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.4e2912fd/.config;   echo "# CONFIG_RD_BZIP2 is not set" >> /opt/openwrt-24.10/qualcommax/ipq807x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.4e2912fd/.config;     echo "# CONFIG_INITRAMFS_COMPRESSION_LZMA is not set" >> /opt/openwrt-24.10/qualcommax/ipq807x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.4e2912fd/.config;   echo "# CONFIG_RD_LZMA is not set" >> /opt/openwrt-24.10/qualcommax/ipq807x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.4e2912fd/.config;     echo "# CONFIG_INITRAMFS_COMPRESSION_LZO is not set" >> /opt/openwrt-24.10/qualcommax/ipq807x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.4e2912fd/.config;   echo "# CONFIG_RD_LZO is not set" >> /opt/openwrt-24.10/qualcommax/ipq807x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.4e2912fd/.config;     echo "# CONFIG_INITRAMFS_COMPRESSION_XZ is not set" >> /opt/openwrt-24.10/qualcommax/ipq807x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.4e2912fd/.config;   echo "# CONFIG_RD_XZ is not set" >> /opt/openwrt-24.10/qualcommax/ipq807x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.4e2912fd/.config;     echo "# CONFIG_INITRAMFS_COMPRESSION_LZ4 is not set" >> /opt/openwrt-24.10/qualcommax/ipq807x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.4e2912fd/.config;   echo "# CONFIG_RD_LZ4 is not set" >> /opt/openwrt-24.10/qualcommax/ipq807x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.4e2912fd/.config;     echo "CONFIG_INITRAMFS_COMPRESSION_ZSTD=y" >> /opt/openwrt-24.10/qualcommax/ipq807x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.4e2912fd/.config;   echo "CONFIG_RD_ZSTD=y" >> /opt/openwrt-24.10/qualcommax/ipq807x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.4e2912fd/.config;   }; cp -fpR /opt/openwrt-24.10/qualcommax/ipq807x/target/linux/generic/other-files/init /opt/openwrt-24.10/qualcommax/ipq807x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/target-dir-4e2912fd/init; touch -hcd "@1730714138" /opt/openwrt-24.10/qualcomma [... cut ...]
CONFIG_VERSION_BUG_URL=""
CONFIG_VERSION_CODE=""
CONFIG_VERSION_CODE_FILENAMES=y
CONFIG_VERSION_DIST="OpenWrt"
CONFIG_VERSION_FILENAMES=y
CONFIG_VERSION_HOME_URL=""
CONFIG_VERSION_HWREV=""
CONFIG_VERSION_MANUFACTURER=""
CONFIG_VERSION_MANUFACTURER_URL=""
CONFIG_VERSION_NUMBER=""
CONFIG_VERSION_PRODUCT=""
CONFIG_VERSION_REPO="https://downloads.openwrt.org/releases/24.10-SNAPSHOT"
CONFIG_VERSION_SUPPORT_URL=""
```
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

obsy, 2024-11-04:
@Ansuel  related to https://github.com/openwrt/openwrt/commit/d78dec3e19e3f2ab0f14ba6819357d50156b30b8 ?

Ansuel, 2024-11-04:
And how much time was waited to claim it did get stuck?

obsy, 2024-11-04:
I manually removed linux-6.6.58.* and target* and gave make -j16, after about 3 minutes the first temporary directories appear which are no longer deleted.
```
$ ps ax | grep flock | wc -l
15
```

```
$ ls build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.* | grep build_dir
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.03363680:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.057cd404:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.06646ecf:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.19ababe6:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.2458ae93:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.27076f27:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.4e2912fd:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.4f8e80a0:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.5310f19a:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.563bb6d5:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.9509eee9:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.a17c5372:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.b7ceb6df:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.c30e5112:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.ef8cd2b4:
```

Ansuel, 2024-11-04:
please wait more....

obsy, 2024-11-04:
After 15min - 17 processes, the same directories remain. The compilation process still hangs.
```
$ ls build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.* | grep build_dir
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.03363680:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.057cd404:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.06646ecf:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.19ababe6:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.2458ae93:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.27076f27:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.4e2912fd:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.4f8e80a0:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.5310f19a:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.563bb6d5:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.6ac55ba6:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.9509eee9:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.a17c5372:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.b7ceb6df:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.c30e5112:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.ef8cd2b4:

```

obsy, 2024-11-04:
After 45min: 
```
$ ps ax | grep flock | wc -l
17
```
```
$ ls build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.* | grep build_dir
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.03363680:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.057cd404:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.06646ecf:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.19ababe6:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.2458ae93:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.27076f27:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.270fd9b3:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.4651dfb3:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.4a724e2b:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.4f8e80a0:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.5310f19a:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.6ac55ba6:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.9509eee9:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.a17c5372:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.b7ceb6df:
build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.58.ef8cd2b4:

```

23.05-snapshot images with a similar configuration in this time would have become complicated 3 times.

obsy, 2024-11-04:
With make -j16 it reaches 16 processes, which change after a few dozen minutes (timeouts)? The list of directories changes slightly (see previous entries), I gave up after an hour.

aparcar, 2024-11-07:
This seems to slow down the ath79 generic buildbot near infinity. I'll try to investigate this later

obsy, 2024-11-07:
The problem is copying a full kernel (4GB) for each device for initramfs. With many parallel threads, there is a strange slowdown in copying, practically to a few kb/s