openwrt/openwrt #16863: 50-root-passwd: missing object check and syntax error

efahl, 2024-11-04:
### Describe the bug

On first boot, `50-root-passwd` fails to execute due to two errors:
```
WARNING: Variable 'credentials' does not exist or is not an array/object
/etc/uci-defaults/50-root-passwd: line 13: syntax error: unexpected end of file (expecting "}")
```

Easy one to fix, add this line following the jshn load:
```
json_is_a credentials object || exit 0
```
and add the missing semicolon after `passwd root` and its closing `}`

### OpenWrt version

r27695-dae6a871de and earlier

### OpenWrt release

SNAPSHOT and now 24.10

### OpenWrt target/subtarget

all

### Device

all

### Image kind

Official downloaded image

### Steps to reproduce

$ sh /etc/uci-defaults/50-root-passwd
WARNING: Variable 'credentials' does not exist or is not an array/object
/etc/uci-defaults/50-root-passwd: line 13: syntax error: unexpected end of file (expecting "}")

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

RussellSenior, 2024-11-10:
@blogic I have seen this too on all my recent builds across a variety of hardware (e.g. buffalo,wzr-600dhp, pcengines,alix2, meraki,mr24, netgear,wndr3800, cudy,x6-v2, ubnt,edgerouter-x, etc). /etc/uci-defaults/50-root-passwd never exits cleanly, and so it remains in /etc/uci-defaults and continues to fail on every boot.

blogic, 2024-11-13:
@RussellSenior thanks for tagging me, I just fixed the issue in main and the release branch

closed 2024-11-13 by blogic