openwrt/openwrt #16881: Tailscale is not available on 23.05.5/x86_64

spitfire, 2024-11-07:
### Describe the bug

Tailscale is not available on 23.05.5/x86_64. I used to be able to install it on 22.03

```
root@OpenWrt01:~# opkg print-architecture
arch all 1
arch noarch 1
arch x86_64 10
root@OpenWrt01:~# opkg update
Downloading https://downloads.openwrt.org/releases/23.05.5/targets/x86/64/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_core
Downloading https://downloads.openwrt.org/releases/23.05.5/targets/x86/64/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/x86_64/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/x86_64/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/x86_64/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/x86_64/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/x86_64/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/x86_64/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/x86_64/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/x86_64/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/x86_64/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/x86_64/telephony/Packages.sig
Signature check passed.
root@OpenWrt01:~# opkg list | grep tailscale  # Check if tailscale appears in the list of available packages
root@OpenWrt01:~# opkg install tailscale
Unknown package 'tailscale'.
Collected errors:
 * opkg_install_cmd: Cannot install package tailscale.
```

### OpenWrt version

r24106-10cc5fcd00

### OpenWrt release

23.05.5

### OpenWrt target/subtarget

x86/64

### Device

Proxmox VE VM - QEMU Standard PC (i440FX + PIIX, 1996)

### Image kind

Official downloaded image

### Steps to reproduce

```
# opkg update
Downloading https://downloads.openwrt.org/releases/23.05.5/targets/x86/64/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_core
Downloading https://downloads.openwrt.org/releases/23.05.5/targets/x86/64/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/x86_64/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/x86_64/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/x86_64/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/x86_64/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/x86_64/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/x86_64/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/x86_64/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/x86_64/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/x86_64/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/x86_64/telephony/Packages.sig
Signature check passed.
# opkg install tailscale
Unknown package 'tailscale'.
Collected errors:
 * opkg_install_cmd: Cannot install package tailscale.
```

### Actual behaviour

```
# opkg install tailscale
Unknown package 'tailscale'.
Collected errors:
 * opkg_install_cmd: Cannot install package tailscale.
```

### Expected behaviour

Software should install

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

spitfire, 2024-11-07:
> Quote reply

I've reported version `r24106-10cc5fcd00`

> Invalid Release reported. `` Is this from a clean repository?

I've reported release `23.05.5`

> Invalid Target/Subtarget reported. `` Is this from a supported device?

I've reported `Proxmox VE VM - QEMU Standard PC (i440FX + PIIX, 1996)`

brada4, 2024-11-07:
It is built for other platforms, btw wrong place for package bugs.

brada4, 2024-11-09:
https://downloads.openwrt.org/releases/faillogs-23.05/x86_64/packages/tailscale/

spitfire, 2024-11-14:
> It is built for other platforms, btw wrong place for package bugs.

thanks for the reply, but the fact that it builds for other platforms does not help me much;) I’m using it right now with previous version of OpenWRT and while I was setting up the current one for someone else it bummed me out a bit that it didn’t work, thankfully I didn’t do upgrade on my own installation.

> https://downloads.openwrt.org/releases/faillogs-23.05/x86_64/packages/tailscale/

Yeah, I’ve eventually found some issues opened for it with no consensus on what is planned to do about it. I’ve managed to use init scripts from OpenWRT and dropped (pretty big) static builds of Tailscale binaries from their sites (and called it a day for now).