openwrt/openwrt #16889: Huawei AP5030DN reset during OpenWrt boot during install

tim-seoss, 2024-11-08:
### Describe the bug

Failed to install using instructions in commit - tried 23.05.4 23.05.5 and snapshot.  Serial terminal output shows a reset after kernel load.

### OpenWrt version

23.05.5

### OpenWrt release

23.05.5

### OpenWrt target/subtarget

ath79

### Device

Huawei AP5030DN

### Image kind

Official downloaded image

### Steps to reproduce

Follow install instructions at https://openwrt.org/toh/huawei/ap5030dn ( https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=06cdc07f8cc703ef7dcb3e7b329b9abff0806a6e )

### Actual behaviour

Apparent hardware resets (and jump back to boot loader) during early kernel boot, after:
```
Bytes transferred = 5644389 (562065 hex)
OK

Starting kernel ...

Huawei AP5030DN

OpenWrt kernel loader for AR7XXX/AR9XXX
Copyright (C) 2011 Gabor Juhos <juhosg@openwrt.org>
Decompressing kernel... DRAM Parameter:
    Tap(low, high): 0x3-0x1f
    Tap values: 0x11-0x11-0x11-0x11
DRAM:  256 MB
```

Output starting with `DRAM Parameter:` and continuing from there is from the firmware boot loader.

### Expected behaviour

OpenWrt kernel starts

### Additional info

Installed firmware and uboot version on the device is newer than that shown in the boot log on https://openwrt.org/toh/huawei/ap5030dn (this device `U-Boot version-632(Dec 30 2021  17:14:57)` wiki boot log: `U-Boot version-618(Aug 13 2020  14:15:09)`  the `Boot Kernel A Address: 0x9f66d000` is different to the `Boot Kernel A Address: 0x9ec08000` in the wiki.

```

DRAM Parameter:
    Tap(low, high): 0x3-0x1f
    Tap values: 0x11-0x11-0x11-0x11
DRAM:  256 MB
Press CTRL+T for Full Memory Test in 1 Seconds:  0
Start up the normal boot...
DRAM Parameter:
    Tap(low, high): 0x3-0x1f
    Tap values: 0x11-0x11-0x11-0x11
DRAM:  256 MB
Press CTRL+T for Full Memory Test in 1 Seconds:  0

       ************************************************************
       **********************    *********    *********************
       ********************       *******       *******************
       ******************         *******         *****************
       ***********    ***          *****          ***    **********
       *********       ***         *****         ***      *********
       ********          **         ***         **         ********
       ********           **        ***        **          ********
       *********           **       ***       **          *********
       ***********          **      ***      **         ***********
       *****      ***         **    ***    **        ***      *****
       *****          ***      **  *****  **     ***          *****
       ******             ***   **********   ***             ******
       *********                 ********                 *********
       ************************************************************
       **********             **************             **********
       *************  ******************************  *************
       ************************************************************

                        Welcome To HUAWEI AP World

U-Boot version-632(Dec 30 2021  17:14:57)

Board Type: 0x4a
Flash: 32 MB
Boot Rom Version: 632
Hardware version: VER.A
Ethernet Mac: 7C-A2-3E-E2-2E-E0
Ethernet Mac: 7C-A2-3E-E2-2E-E0

Start Up time(s) : 1
Image: Current Bootup is A
Image: Current Bootup is A 
Boot Kernel A Address: 0x9ec08000
Boot Kernel B Address: 0x9fb58000
Press j or J to stop Bus-Test in 1 seconds:   0
POST test:  Begin
Testing PCIE  bus : Passed.
Testing SPI   bus : Passed.
Testing DDR   bus : Passed.
Testing RGMII bus : Passed.
Testing SGMII bus : Passed.
POST test: End

Saving POST results: Done  

Press f or F  to stop Auto-Boot in 3 seconds:  0
f
Info: The password is the same as the default. Please Set New Password.

The password must be a combination of at least two of the following: lowercase letters a to z, uppercase letters A to Z, digits, and special characters!

The length of password must be more than 8 and less than 80 bytes.
New password:
Confirm  password:
ar7240> setenv serverip 192.168.1.10
ar7240> setenv ipaddr 192.168.1.1
ar7240> setenv rambootfile openwrt-ath79-generic-huawei_ap5030dn-initramfs-kernel.bin
ar7240> saveenv
Saving Environment to Flash...
ar7240> run ramboot
Using eth0 device
TFTP from server 192.168.1.10; our IP address is 192.168.1.1
Filename 'openwrt-ath79-generic-huawei_ap5030dn-initramfs-kernel.bin'.
Load address: 0x80800000   
Loading: #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         ###############################################################
done
Bytes transferred = 5644389 (562065 hex)
OK

Starting kernel ...

Huawei AP5030DN

OpenWrt kernel loader for AR7XXX/AR9XXX
Copyright (C) 2011 Gabor Juhos <juhosg@openwrt.org>
Decompressing kernel... DRAM Parameter:
    Tap(low, high): 0x3-0x1f
    Tap values: 0x11-0x11-0x11-0x11
DRAM:  256 MB
Press CTRL+T for Full Memory Test in 1 Seconds:  0
Start up the normal boot...
DRAM Parameter:
    Tap(low, high): 0x3-0x1e
    Tap values: 0x10-0x10-0x10-0x10
DRAM:  256 MB
Press CTRL+T for Full Memory Test in 1 Seconds:  0
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

DragonBluep, 2024-11-17:
Did you try snapshot vetsion? Please try this patch.
https://github.com/DragonBluep/openwrt/commit/54bc7c9dc16a3564a1604e29e8457b07c3aeb5ef

tim-seoss, 2024-11-19:
Hi @DragonBluep  - thanks for your patch.  I tried it, and could not get OpenWRT to start up (tried 10 tftp boot attempts).  All hit the watchdog reset before the kernel started.

I also retried the downloaded image (both snapshot and 23.05.4) which I had retrieved just before opening this issue.

Before opening the issue, I had no successful boots (I think out of approximately 6 attempts).  This time I had some successful boots.  The boot works approx 50% of the time.  Last time I must have just been unlucky I think.

Also it's possible that the new uboot version which I have on this hardware has shortened the watchdog timer period.

tim-seoss, 2024-11-27:
Shall I leave this open?

I'm also seeing resets during boot when installed to flash so the race is not limited to tftp booting.

sdfernand, 2024-12-13:
In my case I'm the same, it always restarts after sending openwrt.

DragonBluep, 2024-12-14:
The min `WDI Pulse Width` for SGM706 watchdog is 70 ns. So the watchdog input clock can not greater than 14 MHz. Usually the AHB clock is 200+ MHz. That's why it doesn't work.

You can hack the lzma loader to mux the watchdog pin to the uart out (tx) signal and manually output some chars to feed the watchdog.

CodingMarco, 2024-12-14:
Yeah I mean they say this in their datasheet, but somehow it seemed to work anyway. But of course if it is not specified it might work for some units and for some not I guess...
If we do it with something like UART we have to make sure that the toggling continues even while the LZMA decompression is happening. So this might be a larger change?

DragonBluep, 2024-12-14:
> Yeah I mean they say this in their datasheet, but somehow it seemed to work anyway. But of course if it is not specified it might work for some units and for some not I guess... If we do it with [...]

Usually decompression function is a for or while loop. we only need to insert uart char output here. I guess we dont need to touch other loader codes because 1.6s is enough to execute them, maybe.

sdfernand, 2024-12-18:
I think I understand the problem described.
Does it happen in all versions of openwrt compatible with this model?
Thanks

CodingMarco, 2024-12-18:
As I said, I didn't have this problem with _any_ of the ~20 AP5030DNs I installed OpenWrt on (which were however bought in a small time frame for our student network, maybe even with the same order).

DragonBluep, 2025-04-14:
Please try this patch:
```
diff --git a/target/linux/ath79/image/lzma-loader/src/board.c b/target/linux/ath79/image/lzma-loader/src/board.c
index 7ed414aa90..5c27551b20 100644
--- a/target/linux/ath79/image/lzma-loader/src/board.c
+++ b/target/linux/ath79/image/lzma-loader/src/board.c
@@ -200,20 +200,7 @@ static inline void huawei_ap_init(void)
 
 	reg = READREG(gpiobase + AR71XX_GPIO_REG_OE);
 	WRITEREG(gpiobase + AR71XX_GPIO_REG_OE,
-			reg & ~(1 << watchdog_gpio));
-
-	/* Set GPIO15 MUX to output CLK_OBS5 (= CPU_CLK/4)
-	 * or CLK_OBS4 (= AHB_CLK/2) to keep the watchdog happy
-	 * until wdt-gpio takes over
-	 */
-	reg = READREG(gpiobase + AR934X_GPIO_REG_OUT_FUNC3);
-#if defined(CONFIG_BOARD_HUAWEI_AP5030DN)
-	WRITEREG(gpiobase + AR934X_GPIO_REG_OUT_FUNC3,
-			reg | (QCA955X_GPIO_OUTSEL_CLK_OBS5 << 24));
-#elif defined(CONFIG_BOARD_HUAWEI_AP6010DN)
-	WRITEREG(gpiobase + AR934X_GPIO_REG_OUT_FUNC3,
-			reg | (AR934X_GPIO_OUTSEL_CLK_OBS4 << 24));
-#endif
+			reg | (1 << watchdog_gpio));
 }
 #else
 static inline void huawei_ap_init(void) {}

```

Shidyfu, 2025-04-25:
Thanks for continuing to look into this team! I've never had a successful boot per the original issue. I was not able to get the most recent patch to work but would like to confirm I've applied the patch correctly. Snippet of code I have used below:

`#if defined(CONFIG_BOARD_HUAWEI_AP5030DN) || defined(CONFIG_BOARD_HUAWEI_AP6010DN)
static inline void huawei_ap_init(void)
{
	const unsigned int watchdog_gpio = 15;
	unsigned int gpiobase, reg;

	gpiobase = KSEG1ADDR(AR71XX_GPIO_BASE);

	printf("Huawei AP p\n");

	reg = READREG(gpiobase + AR71XX_GPIO_REG_OE);
	WRITEREG(gpiobase + AR71XX_GPIO_REG_OE,
			reg | (1 << watchdog_gpio));
}
#else
static inline void huawei_ap_init(void) {}
#endif
`

(I add letters to the printf to confirm the build changes when I recompile)

I have very little experience with coding in c or developing for hardware but would like to contribute anyway I can so please let me know if there's testing or investigation I can do to help resolve this issue.

Nabhan87, 2025-11-01:
> I got the same issue when flashing my AP5030DN. I disassembled my AP and found that the watchdog chip used in my unit is an ADM706 instead of SGM706, which has a 50ns WDI Pulse Width. Hope this information helps.
> 
> I also managed to disable the watchdog by floating the 6th pin of ADM706. The device now has a stable boot to OpenWRT.

Can you help me? Why does the 192.168.1.1 page in my browser show ERR_CONNECTION_REFUSED?