openwrt/openwrt #16896: Exception stack(0xc3a2bfa8 to 0xc3a2bff0)

pko76, 2024-11-09:
### Describe the bug

Exception stack(0xc3a2bfa8 to 0xc3a2bff0) got at the kernel start several times. 

[   48.047521] ------------[ cut here ]------------
[   48.058543] WARNING: CPU: 1 PID: 1481 at backports-6.11.2/net/mac80211/main.c:270 ieee80211_do_open+0x554/0x650 [mac80211]
[   48.063221] Modules linked in: ksmbd pppoe ppp_async option nft_fib_inet nf_flow_table_inet ath10k_pci(O) ath10k_core(O) ath(O) wireguard usb_wwan qmi_wwan pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mac80211(O) libchacha20poly1305 curve25519_neon cfg80211(O) usbserial usbnet slhc poly1305_arm nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcurve25519_generic libcrc32c hwmon crc_ccitt compat(O) chacha_neon cdc_wdm nls_ucs2_utils cifs_arc4 asn1_decoder oid_registry tun ovpn_dco_v2(O) udp_tunnel ip6_udp_tunnel autofs4 netfs nls_utf8 nls_iso8859_1 nls_cp437 nls_cp1250 zram zsmalloc sha512_arm md5 kpp ghash_arm_ce cmac usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_qcom fsl_mph_dr_of ehci_platform ehci_fsl sd_mod scsi_mod scsi_common ehci_hcd
[   48.063749]  gpio_button_hotplug(O) vfat fat mii crc32c_generic [last unloaded: ksmbd]
[   48.163941] CPU: 1 PID: 1481 Comm: hostapd Tainted: G           O       6.6.59 #0
[   48.171631] Hardware name: Generic DT based system
[   48.179185]  unwind_backtrace from show_stack+0x10/0x14
[   48.183868]  show_stack from dump_stack_lvl+0x40/0x4c
[   48.188988]  dump_stack_lvl from __warn+0x94/0xbc
[   48.194197]  __warn from warn_slowpath_fmt+0x158/0x15c
[   48.198885]  warn_slowpath_fmt from ieee80211_do_open+0x554/0x650 [mac80211]
[   48.203943]  ieee80211_do_open [mac80211] from ieee80211_open+0x58/0x7c [mac80211]
[   48.211130]  ieee80211_open [mac80211] from __dev_open+0xd4/0x194
[   48.218507]  __dev_open from __dev_change_flags+0x184/0x204
[   48.224667]  __dev_change_flags from dev_change_flags+0x18/0x54
[   48.230048]  dev_change_flags from devinet_ioctl+0x5ac/0x7dc
[   48.235951]  devinet_ioctl from inet_ioctl+0x16c/0x268
[   48.241855]  inet_ioctl from sock_ioctl+0x240/0x60c
[   48.246802]  sock_ioctl from sys_ioctl+0x25c/0xc0c
[   48.251577]  sys_ioctl from ret_fast_syscall+0x0/0x54
[   48.256436] Exception stack(0xc3a2bfa8 to 0xc3a2bff0)
[   48.261564] bfa0:                   00000010 beb58848 00000010 00008914 beb58848 00000000
[   48.266603] bfc0: 00000010 beb58848 00000010 00000036 00000001 00000001 00000000 00000020
[   48.274758] bfe0: 00528ea4 beb58708 004a2f84 b6ee766c
[   48.282950] ---[ end trace 0000000000000000 ]---
[   48.291494] br-lan: port 4(phy0-ap0) entered blocking state
[   48.292659] br-lan: port 4(phy0-ap0) entered disabled state
[   48.298003] ath10k_ahb a000000.wifi phy0-ap0: entered allmulticast mode
[   48.303969] ath10k_ahb a000000.wifi phy0-ap0: entered promiscuous mode
[   48.310436] br-lan: port 4(phy0-ap0) entered blocking state
[   48.316691] br-lan: port 4(phy0-ap0) entered forwarding state
[   48.325984] br-lan: port 4(phy0-ap0) entered disabled state
[   48.495895] ath10k_ahb a000000.wifi: NOTE:  Firmware DBGLOG output disabled in debug_mask: 0x10000000
[   48.601543] br-lan: port 5(phy1-ap0) entered blocking state
[   48.601637] br-lan: port 5(phy1-ap0) entered disabled state
[   48.606055] ath10k_ahb a800000.wifi phy1-ap0: entered allmulticast mode
[   48.612086] ath10k_ahb a800000.wifi phy1-ap0: entered promiscuous mode
[   50.112079] ath10k_ahb a800000.wifi: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[   50.112167] ath10k_ahb a800000.wifi: msdu-desc: 2500  skid: 32
[   50.159216] ath10k_ahb a800000.wifi: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[   50.162856] ath10k_ahb a800000.wifi: wmi print 'free: 53252 iram: 13432 sram: 35752'
[   50.477137] ath10k_ahb a800000.wifi: rts threshold -1
[   50.477402] ath10k_ahb a800000.wifi: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
[   50.481288] ------------[ cut here ]------------
[   50.492222] WARNING: CPU: 3 PID: 1481 at backports-6.11.2/net/mac80211/main.c:270 ieee80211_do_open+0x554/0x650 [mac80211]
[   50.496926] Modules linked in: ksmbd(+) pppoe ppp_async option nft_fib_inet nf_flow_table_inet ath10k_pci(O) ath10k_core(O) ath(O) wireguard usb_wwan qmi_wwan pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mac80211(O) libchacha20poly1305 curve25519_neon cfg80211(O) usbserial usbnet slhc poly1305_arm nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcurve25519_generic libcrc32c hwmon crc_ccitt compat(O) chacha_neon cdc_wdm nls_ucs2_utils cifs_arc4 asn1_decoder oid_registry tun ovpn_dco_v2(O) udp_tunnel ip6_udp_tunnel autofs4 netfs nls_utf8 nls_iso8859_1 nls_cp437 nls_cp1250 zram zsmalloc sha512_arm md5 kpp ghash_arm_ce cmac usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_qcom fsl_mph_dr_of ehci_platform ehci_fsl sd_mod scsi_mod scsi_common ehci_hcd
[   50.497667]  gpio_button_hotplug(O) vfat fat mii crc32c_generic [last unloaded: ksmbd]
[   50.597717] CPU: 3 PID: 1481 Comm: hostapd Tainted: G        W  O       6.6.59 #0
[   50.605611] Hardware name: Generic DT based system
[   50.613163]  unwind_backtrace from show_stack+0x10/0x14
[   50.617846]  show_stack from dump_stack_lvl+0x40/0x4c
[   50.622969]  dump_stack_lvl from __warn+0x94/0xbc
[   50.628174]  __warn from warn_slowpath_fmt+0x158/0x15c
[   50.632854]  warn_slowpath_fmt from ieee80211_do_open+0x554/0x650 [mac80211]
[   50.637893]  ieee80211_do_open [mac80211] from ieee80211_open+0x58/0x7c [mac80211]
[   50.645111]  ieee80211_open [mac80211] from __dev_open+0xd4/0x194
[   50.652486]  __dev_open from __dev_change_flags+0x184/0x204
[   50.658644]  __dev_change_flags from dev_change_flags+0x18/0x54
[   50.664028]  dev_change_flags from devinet_ioctl+0x5ac/0x7dc
[   50.669930]  devinet_ioctl from inet_ioctl+0x16c/0x268
[   50.675834]  inet_ioctl from sock_ioctl+0x240/0x60c
[   50.680780]  sock_ioctl from sys_ioctl+0x25c/0xc0c
[   50.685558]  sys_ioctl from ret_fast_syscall+0x0/0x54
[   50.690406] Exception stack(0xc3a2bfa8 to 0xc3a2bff0)
[   50.695531] bfa0:                   00000010 beb58848 00000010 00008914 beb58848 00000000
[   50.700569] bfc0: 00000010 beb58848 00000010 00000036 00000001 00000001 00000000 00000020
[   50.708726] bfe0: 00528ea4 beb58708 004a2f84 b6ee766c
[   50.716951] ---[ end trace 0000000000000000 ]---
[   50.722191] br-lan: port 5(phy1-ap0) entered blocking state
[   50.726622] br-lan: port 5(phy1-ap0) entered forwarding state
[   50.732882] ath10k_ahb a800000.wifi phy1-ap0: left allmulticast mode
[   50.737833] ath10k_ahb a800000.wifi phy1-ap0: left promiscuous mode
[   50.744413] br-lan: port 5(phy1-ap0) entered disabled state
[   50.847777] br-lan: port 5(phy1-ap0) entered blocking state
[   50.847964] br-lan: port 5(phy1-ap0) entered disabled state
[   50.852458] ath10k_ahb a800000.wifi phy1-ap0: entered allmulticast mode
[   50.858232] ath10k_ahb a800000.wifi phy1-ap0: entered promiscuous mode
[   51.218888] br-lan: port 4(phy0-ap0) entered blocking state
[   51.218980] br-lan: port 4(phy0-ap0) entered forwarding state
[   52.326877] br-lan: port 5(phy1-ap0) entered blocking state
[   52.326950] br-lan: port 5(phy1-ap0) entered forwarding state

### OpenWrt version

r28034-ca53f2d430

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

 ipq40xx/generic

### Device

ZTE MF286D

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

pko76, 2024-11-22:
Hi brada4, sorry for the ate reply. I have, imho, nothing special installed, just standard:
Powered by [LuCI Master (24.326.69161~3f907ab)](https://github.com/openwrt/luci) / [OpenWrt SNAPSHOT (r28141-52519a59a8)](https://openwrt.org/)

Here is the list of installed packages:  https://github.com/pko76/exchange-files/blob/main/list%20of%20packages.txt

brada4, 2024-11-22:
Bith semi-backtraces point to "nothing special" ksmbd.

pko76, 2024-11-22:
Thank you for the info. Is there anything I could do about it?

ptpt52, 2024-12-03:
same issue on my acrh17
```
[   74.550784] qca8k-ipq4019 c000000.switch lan1: Link is Up - 100Mbps/Full - flow control rx/tx
[   74.828480] ath10k_ahb a000000.wifi: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[   74.828554] ath10k_ahb a000000.wifi: msdu-desc: 2500  skid: 32
[   74.875852] ath10k_ahb a000000.wifi: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[   74.876796] ath10k_ahb a000000.wifi: wmi print 'free: 53252 iram: 13432 sram: 35752'
[   75.251926] ath10k_ahb a000000.wifi: rts threshold -1
[   75.252393] ath10k_ahb a000000.wifi: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
[   75.256034] ------------[ cut here ]------------
[   75.267022] WARNING: CPU: 1 PID: 2044 at /mnt/build/x-wrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/mac80211-regular/backports-6.11.2/net/mac80211/main.c:272 ieee80211_do_open+0x560/0x604 [mac80211]
[   75.271808] Modules linked in: qcserial pppoe ppp_async option l2tp_ppp cdc_mbim ath10k_pci(O) ath10k_core(O) ath(O) wireguard usb_wwan sierra_net sierra rndis_host qmi_wwan pptp pppox ppp_mppe ppp_generic pl2303 nft_fib_inet mac80211(O) libchacha20poly1305 ipt_REJECT huawei_cdc_ncm curve25519_neon cp210x cfg80211(O) cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_DSCP xt_DNETMAP(O) xt_DHCPMAC(O) xt_CLASSIFY xfrm_interface usbserial usbnet usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_fsm ts_bm tcp_scalable tcp_bbr slhc sch_cake poly1305_arm nft_tproxy nft_socket nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_queue nft_numgen nft_nat nft_meta_bridge nft_masq nft_log
[   75.272365]  nft_limit nft_hash nft_fwd_netdev nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_dup_netdev nft_ct nft_compat nft_chain_nat nfnetlink_queue nf_tproxy_ipv6 nf_tproxy_ipv4 nf_tables nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_syslog nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_sane nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_netbios_ns nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_bridge ts_kmp nf_conntrack_amanda nf_conncount macvlan libcurve25519_generic libcrc32c ipvlan iptable_nat iptable_mangle iptable_filter ipt_ECN ipheth ip_tables hwmon crc_ccitt compat_xtables(O) compat(O) chacha_neon cdc_wdm br_netfilter natflow(O) natcap(O) nf_nat nf_conntrack fuse ntfs3 sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact ledtrig_usbport cryptodev(O)
[   75.360489]  xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink jool_siit(O) jool(O) nf_defrag_ipv4 jool_common(O) ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 nfsv4 ksmbd nls_ucs2_utils cifs_arc4 asn1_decoder nfs msdos bonding tls ip6_gre ip_gre gre nat46(O) nf_defrag_ipv6 ip6_vti ip_vti sit l2tp_netlink l2tp_core udp_tunnel ip6_udp_tunnel ipcomp6 xfrm6_tunnel esp6 ah6 xfrm4_tunnel ipcomp esp4 ah4 ip6_tunnel oid_registry tunnel6 tunnel4 ip_tunnel udp_diag tcp_diag raw_diag inet_diag tun xfrm_user xfrm_ipcomp af_key xfrm_algo lockd sunrpc grace autofs4 dns_resolver nls_utf8 nls_iso8859_1 nls_cp437 crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha512_arm sha1_arm_neon sha1_arm md5 ghash_arm_ce
[   75.450207]  echainiv cmac authencesn authenc arc4 uas usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_qcom ohci_platform ohci_hcd fsl_mph_dr_of ehci_platform ehci_fsl sd_mod scsi_mod scsi_common ehci_hcd gpio_button_hotplug(O) vfat fat ext4 mbcache jbd2 exfat mii crc32c_generic
[   75.565529] CPU: 1 PID: 2044 Comm: hostapd Tainted: G        W  O       6.6.63 #0
[   75.587680] Hardware name: Generic DT based system
[   75.595312]  unwind_backtrace from show_stack+0x10/0x14
[   75.599993]  show_stack from dump_stack_lvl+0x40/0x4c
[   75.605131]  dump_stack_lvl from __warn+0x94/0xbc
[   75.610343]  __warn from warn_slowpath_fmt+0x158/0x15c
[   75.615024]  warn_slowpath_fmt from ieee80211_do_open+0x560/0x604 [mac80211]
[   75.620069]  ieee80211_do_open [mac80211] from ieee80211_open+0x58/0x7c [mac80211]
[   75.627272]  ieee80211_open [mac80211] from __dev_open+0xc4/0x170
[   75.634644]  __dev_open from __dev_change_flags+0x168/0x1dc
[   75.640813]  __dev_change_flags from dev_change_flags+0x18/0x54
[   75.646192]  dev_change_flags from devinet_ioctl+0x5ac/0x7dc
[   75.652101]  devinet_ioctl from inet_ioctl+0x170/0x274
[   75.658004]  inet_ioctl from sock_ioctl+0x23c/0x620
[   75.662943]  sock_ioctl from sys_ioctl+0x25c/0xc0c
[   75.667708]  sys_ioctl from ret_fast_syscall+0x0/0x54
[   75.672562] Exception stack(0xc4c67fa8 to 0xc4c67ff0)
[   75.677707] 7fa0:                   00000010 bef4a7e0 00000010 00008914 bef4a7e0 bef4a7d0
[   75.682753] 7fc0: 00000010 bef4a7e0 00000010 00000036 00000001 00000001 00000000 00000020
[   75.690905] 7fe0: 00611918 bef4a6a8 0052d50c b6efbdd8
[   75.699158] ---[ end trace 0000000000000000 ]---

```

brada4, 2024-12-03:
Does it change backtrace changing ath10k firmware from -ct to mñinline (without -ct) or vice versa?

brada4, 2024-12-05:
OK, so we have a workaround, c7v5 had some backtraces spit out already with v22, I just tried swapping drivers under assumption it cannot get worse and it worked. Question is how widespread is the problem and how many need to be switched to other driver.

meshing-m, 2024-12-08:
https://forum.openwrt.org/t/ipq4019-adding-support-for-tp-link-deco-m9-plus/216594/62?u=meshing
hitting the same.

brada4, 2024-12-08:
You can choose on a freshly ported platform.

closed 2025-03-31 by openwrt-bot