openwrt/openwrt #16898: [Intel I226-V] Periodic Link Down

MMitsuha, 2024-11-09:
### Describe the bug

On ESXi 8.0U3, pass-throughed Intel I226-V. Network adapter periodically go down and auto reconnect.

### OpenWrt version

r0-b5e1544

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

x86/64

### Device

VMware, Inc. VMware20,1

### Image kind

Self-built image

### Steps to reproduce

1. Configure pppoe
2. Wait few hours

### Actual behaviour

Adapter go down

### Expected behaviour

Adapter stay connected

### Additional info

System log:
```
[133455.817749] igc 0000:02:02.0 eth0: NIC Link is Down
[133455.818205] igc 0000:02:02.0 eth0: Register Dump
[133455.818532] igc 0000:02:02.0 eth0: Register Name   Value
[133455.818903] igc 0000:02:02.0 eth0: CTRL            081c0641
[133455.819258] igc 0000:02:02.0 eth0: STATUS          00780681
[133455.819609] igc 0000:02:02.0 eth0: CTRL_EXT        100000c0
[133455.819961] igc 0000:02:02.0 eth0: MDIC            18017949
[133455.820313] igc 0000:02:02.0 eth0: ICR             00000001
[133455.820670] igc 0000:02:02.0 eth0: RCTL            04408022
[133455.821051] igc 0000:02:02.0 eth0: RDLEN[0-3]      00001000 00001000 00001000 00001000
[133455.821489] igc 0000:02:02.0 eth0: RDH[0-3]        0000003c 00000000 00000000 00000008
[133455.821925] igc 0000:02:02.0 eth0: RDT[0-3]        0000003b 000000ff 000000ff 00000007
[133455.822394] igc 0000:02:02.0 eth0: RXDCTL[0-3]     02040808 02040808 02040808 02040808
[133455.822830] igc 0000:02:02.0 eth0: RDBAL[0-3]      06f8b000 06f8e000 06f91000 0704b000
[133455.823266] igc 0000:02:02.0 eth0: RDBAH[0-3]      00000000 00000000 00000000 00000000
[133455.823696] igc 0000:02:02.0 eth0: TCTL            a503f0fa
[133455.824061] igc 0000:02:02.0 eth0: TDBAL[0-3]      06f79000 06f7e000 06f83000 06f88000
[133455.824516] igc 0000:02:02.0 eth0: TDBAH[0-3]      00000000 00000000 00000000 00000000
[133455.824950] igc 0000:02:02.0 eth0: TDLEN[0-3]      00001000 00001000 00001000 00001000
[133455.825397] igc 0000:02:02.0 eth0: TDH[0-3]        00000076 000000e0 0000001b 00000023
[133455.825832] igc 0000:02:02.0 eth0: TDT[0-3]        0000007a 000000f9 0000001f 00000027
[133455.826278] igc 0000:02:02.0 eth0: TXDCTL[0-3]     02100108 02100108 02100108 02100108
[133455.826707] igc 0000:02:02.0 eth0: Reset adapter
[133460.017879] igc 0000:02:02.0 eth0: NIC Link is Up 2500 Mbps Full Duplex, Flow Control: RX
```

### Diffconfig

```text
[... 203 lines cut ...]
CONFIG_PACKAGE_tcpdump=y
CONFIG_PACKAGE_terminfo=y
CONFIG_PACKAGE_ttyd=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-lua=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uwsgi=y
CONFIG_PACKAGE_uwsgi-cgi-plugin=y
CONFIG_PACKAGE_uwsgi-luci-support=y
CONFIG_PACKAGE_uwsgi-syslog-plugin=y
CONFIG_PACKAGE_vim-fuller=y
CONFIG_PACKAGE_wget-ssl=y
CONFIG_PACKAGE_wireguard-tools=y
CONFIG_PACKAGE_xtables-legacy=y
CONFIG_PACKAGE_xtables-nft=y
CONFIG_PACKAGE_zlib=y
CONFIG_PARTED_READLINE=y
CONFIG_PCRE2_JIT_ENABLED=y
# CONFIG_TARGET_IMAGES_GZIP is not set
CONFIG_VMDK_IMAGES=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-11-09:
Is your device vmware compatible?
Is it virtual function or full device?
Is it isolated from resets on same pcie bus?

MMitsuha, 2024-11-10:
> Is your device vmware compatible?
No, I installed ESXi from an image. But I don't think it is VMware's issue.
> Is it virtual function or full device?
I configure the adapter as a pass-throughed device in ESXi.
> Is it isolated from resets on same pcie bus?
I don't know, but other adapters working properly.

brada4, 2024-11-10:
OK, you have ton of potential well documented problems on vmware side, wildly out of scope of OpenWRT. As a test run boot OpenWRT from USB stick and watch network card working undisrupted.

nasbdh9, 2024-11-10:
This is a common hardware problem on i225/i226 chip. 
You can try replacing the RJ45 cable to see if the situation improves.

MMitsuha, 2024-11-12:
It seems to be hardware issue, but I use vmxnet3 instead, no problem so far.

closed 2024-11-12 by MMitsuha