openwrt/openwrt #16903: Ignoring NSS change in dmesg

serj-47, 2024-11-10:
### Describe the bug

While browsing dmesg, I see messages of this type:
`Ignoring NSS change in VHT Operating Mode Notification from (client mac here) with invalid nss 2`
Apparently, this error is already known, and it is stated in the list of changes by 23.05.5 as fixed.

### OpenWrt version

r24106-10cc5fcd00

### OpenWrt release

23.05.5

### OpenWrt target/subtarget

ramips/mt7621

### Device

Xiaomi Mi Router 4A Gigabit Edition

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

milindpatel63, 2024-11-11:
I have this exact issue.
I get the same error in logs
```
Ignoring NSS change in VHT Operating Mode Notification from (client mac here) with invalid nss 2
```
I'm running Openwrt 23.05.5 on TP Link Archer A7v5. And this issue happens to only 1 of my client device (Oneplus 7 Pro).
When roaming from one AP to another, my phone would just fail to connect to wifi, and if i look at the AP logs, the see the above invalid nss 2 error.
This started happening AFTER the latest 23.05.5 update. Before that version everything was fine.

serj-47, 2024-11-11:
I noticed that the error always shows the same mac address, namely, this is my mi pad5 tablet. I have not met any other mac addresses, despite the fact that I have quite a lot of wifi clients, something about 10

brada4, 2024-11-11:
Where is the error and what reaction you expect to it? Some androids powersave disabling part of builtin antennas, now AP handles scaling back up correctly and prints warnings.

khanjui, 2024-11-14:
Same device, same warning. Occurs with iPhone SE 2020 and MacBook Air 2020 Intel.

brada4, 2024-11-15:
It is up to you to prove there is a bug.

milindpatel63, 2024-11-15:
> It is up to you to prove there is a bug.

well my phone fails to connect to the wifi (both 2.4ghz and 5ghz have same name with fast roaming enabled) after this `invalid nss 2` error. Is that considered a bug?

brada4, 2024-11-15:
"no internet connection" means xiaomi online beacon is down, far outside your router.

brada4, 2024-11-16:
If you get to connection your wifi with all wpa and nss and dhcp is already connecred, just that remaining infrastructure  is slow to warm up.

dashernln, 2024-11-16:
> "no internet connection" means xiaomi online beacon is down, far outside your router.
i think, that described problem is unrelated with nss message in kernel log.
today, caught a situation on redmi ax6000 with no nss messages and no connectivity from 5ghz wifi (no icmp or ssh connection to the ap ip address). 2.4 works fine at the time. 
maybe its related to some wifi settings (eg, FT, usteer, etc)?

ivysochyn, 2024-11-17:
The same error, with WLAN connection being randomly dropped, is observed for Pixel 8 Pro, Pixel 6a and Iphone 13 Pro (no access to internet, however still connected to WLAN).

OpenWRT version: `OpenWrt 23.05.5 r24106-10cc5fcd00`
Hardware: `Linksys EA8300`

brada4, 2024-11-17:
You are complaining about documented features
https://discussions.apple.com/docs/DOC-12271
https://support.google.com/fi/thread/2635155/disable-smart-network-switch-avoid-bad-wifi-connections?hl=en

ivysochyn, 2024-11-17:
Responding to @brada4 
> You are complaining about documented features https://discussions.apple.com/docs/DOC-12271 https://support.google.com/fi/thread/2635155/disable-smart-network-switch-avoid-bad-wifi-connections?hl=en

Thank you for pointing out the documented features related to 'sleep' mode. I understand that this might be a common assumption, and frankly speaking was mine. However, my concern is that the devices do not automatically restore internet access once they enter this state, even after some time. This issue persists beyond a brief 'loading' period, and only a manual reconnection to the WLAN resolves it. Interestingly, I haven't encountered this behavior with other routers that weren't running OpenWrt, and unfortunately, I don't have other routers available to test with OpenWrt.

Also, I must admit that some of the cases couldn't be categorized as 'devices are in a sleep mode', just because those devices were actively used at the moment (most of the cases I experienced internet access being lost during doom-scrolling an Instagram).

This suggests that the issue might not be related to the devices entering a low-power state, and it seems to be an unintended behavior.

brada4, 2024-11-17:
Sorry,  you all fail to link your mobile gadget dissatisfaction with diagnostic warning in logs.

ivysochyn, 2024-11-17:
I just experienced this issue again on my Pixel 8 Pro. I tried waiting longer (about 5 minutes while using the device), but the internet connection still did not appear.

Upon checking the system logs, I didn't see anything except the reconnection process itself (MAC is 86:75:2e:f2:70:10):
```
...
Sun Nov 17 02:20:57 2024 daemon.info hostapd: phy0-ap0: STA f4:c8:8a:3d:46:59 IEEE 802.11: associated (aid 4)
Sun Nov 17 02:20:57 2024 daemon.notice hostapd: phy0-ap0: AP-STA-CONNECTED f4:c8:8a:3d:46:59 auth_alg=open
Sun Nov 17 02:20:57 2024 daemon.info hostapd: phy0-ap0: STA f4:c8:8a:3d:46:59 WPA: pairwise key handshake completed (RSN)
Sun Nov 17 02:20:57 2024 daemon.notice hostapd: phy0-ap0: EAPOL-4WAY-HS-COMPLETED f4:c8:8a:3d:46:59
Sun Nov 17 02:25:30 2024 daemon.notice hostapd: phy0-ap0: AP-STA-DISCONNECTED f4:c8:8a:3d:46:59
Sun Nov 17 02:25:30 2024 kern.info kernel: [710363.596744] ath10k_pci 0000:01:00.0: mac flush vdev 0 drop 0 queues 0x1 ar->paused: 0x0  arvif->paused: 0x0
Sun Nov 17 03:25:54 2024 daemon.notice hostapd: phy0-ap0: AP-STA-DISCONNECTED 86:75:2e:f2:70:10
Sun Nov 17 03:25:54 2024 daemon.info hostapd: phy0-ap0: STA 86:75:2e:f2:70:10 IEEE 802.11: disassociated
Sun Nov 17 03:25:54 2024 kern.info kernel: [713987.723765] ath10k_pci 0000:01:00.0: mac flush vdev 0 drop 0 queues 0x1 ar->paused: 0x0  arvif->paused: 0x0
Sun Nov 17 03:25:55 2024 daemon.info hostapd: phy0-ap0: STA 86:75:2e:f2:70:10 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Sun Nov 17 03:25:58 2024 daemon.info hostapd: phy0-ap0: STA 86:75:2e:f2:70:10 IEEE 802.11: authenticated
Sun Nov 17 03:25:58 2024 daemon.notice hostapd: phy0-ap0: STA-OPMODE-MAX-BW-CHANGED 86:75:2e:f2:70:10 40
Sun Nov 17 03:25:58 2024 daemon.info hostapd: phy0-ap0: STA 86:75:2e:f2:70:10 IEEE 802.11: associated (aid 3)
Sun Nov 17 03:25:58 2024 daemon.notice hostapd: phy0-ap0: AP-STA-CONNECTED 86:75:2e:f2:70:10 auth_alg=open
Sun Nov 17 03:25:58 2024 daemon.info hostapd: phy0-ap0: STA 86:75:2e:f2:70:10 WPA: pairwise key handshake completed (RSN)
Sun Nov 17 03:25:58 2024 daemon.notice hostapd: phy0-ap0: EAPOL-4WAY-HS-COMPLETED 86:75:2e:f2:70:10
```

I've looked into `dmesg`, and didn't notice anything suspicious there as well (could miss).

Attaching full `system log` and `dmesg`, just in case.
[systemlog.txt](https://github.com/user-attachments/files/17788643/systemlog.txt)
[dmesg.txt](https://github.com/user-attachments/files/17788644/dmesg.txt)

brada4, 2024-11-17:
The NSS or "flush vdev" messages are too verbose around client powersaving and disconnecting, they do not directly cause any problems. The later problems are about reconnecting with potentially problematic extensions like sae-mixed, or 11R failing and leading to full reauthentication, then slow DNS and network path to "internet connection" check.