openwrt/openwrt #16905: OpenWRT 24.10: package/kernel/mac80211 failed to build (build variant: regular)

ZerBea, 2024-11-10:
### Describe the bug

Building firmware for target bcm2708 failed with "package/kernel/mac80211 failed to build (build variant: regular)"
```
/home/zerobeat/Source/raspberry/openwrt/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-bcm27xx_bcm2708/mac80211-regular/backports-6.11.2/net/wireless/nl80211.c:18038:9: error: too many arguments to function 'genlmsg_multicast_allns'
18038 |         genlmsg_multicast_allns(&nl80211_fam, msg, 0,
      |         ^~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/zerobeat/Source/raspberry/openwrt/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-bcm27xx_bcm2708/mac80211-regular/backports-6.11.2/backport-include/net/genetlink.h:3,
                 from /home/zerobeat/Source/raspberry/openwrt/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-bcm27xx_bcm2708/mac80211-regular/backports-6.11.2/net/wireless/nl80211.c:25:
./include/net/genetlink.h:484:5: note: declared here
  484 | int genlmsg_multicast_allns(const struct genl_family *family,
      |     ^~~~~~~~~~~~~~~~~~~~~~~
/home/zerobeat/Source/raspberry/openwrt/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-bcm27xx_bcm2708/mac80211-regular/backports-6.11.2/net/wireless/nl80211.c: In function 'nl80211_send_beacon_hint_event':
/home/zerobeat/Source/raspberry/openwrt/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-bcm27xx_bcm2708/mac80211-regular/backports-6.11.2/net/wireless/nl80211.c:18774:9: error: too many arguments to function 'genlmsg_multicast_allns'
18774 |         genlmsg_multicast_allns(&nl80211_fam, msg, 0,
      |         ^~~~~~~~~~~~~~~~~~~~~~~
./include/net/genetlink.h:484:5: note: declared here
  484 | int genlmsg_multicast_allns(const struct genl_family *family,
      |     ^~~~~~~~~~~~~~~~~~~~~~~
make[9]: *** [scripts/Makefile.build:243: /home/zerobeat/Source/raspberry/openwrt/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-bcm27xx_bcm2708/mac80211-regular/backports-6.11.2/net/wireless/nl80211.o] Error 1
make[8]: *** [scripts/Makefile.build:480: /home/zerobeat/Source/raspberry/openwrt/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-bcm27xx_bcm2708/mac80211-regular/backports-6.11.2/net/wireless] Error 2
make[7]: *** [Makefile:1921: /home/zerobeat/Source/raspberry/openwrt/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-bcm27xx_bcm2708/mac80211-regular/backports-6.11.2] Error 2
make[6]: *** [Makefile.build:13: modules] Error 2
make[5]: *** [Makefile.real:105: modules] Error 2
make[4]: *** [Makefile:121: modules] Error 2
make[3]: *** [Makefile:402: /home/zerobeat/Source/raspberry/openwrt/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-bcm27xx_bcm2708/mac80211-regular/backports-6.11.2/.built] Error 2
make[3]: Leaving directory '/home/zerobeat/Source/raspberry/openwrt/package/kernel/mac80211'
time: package/kernel/mac80211/regular/compile#2.09#0.60#2.65
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[2]: *** [package/Makefile:179: package/kernel/mac80211/compile] Error 1
make[2]: Leaving directory '/home/zerobeat/Source/raspberry/openwrt'
make[1]: *** [package/Makefile:173: /home/zerobeat/Source/raspberry/openwrt/staging_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/stamp/.package_compile] Error 2
```

### OpenWrt version

openwrt-24.10 (git branch -a)

### OpenWrt release

openwrt-24.10 (git branch -a)

### OpenWrt target/subtarget

bcm27xx_bcm2708

### Device

nl80211

### Image kind

Self-built image

### Steps to reproduce

compile latest githead (openwrt-24.10) for target bcm2708.

### Actual behaviour

build failed: 
```
/home/zerobeat/Source/raspberry/openwrt/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-bcm27xx_bcm2708/mac80211-regular/backports-6.11.2/net/wireless/nl80211.c:18038:9: error: too many arguments to function 'genlmsg_multicast_allns'
18038 |         genlmsg_multicast_allns(&nl80211_fam, msg, 0,
```

### Expected behaviour

successful build

[... 142 lines cut ...]
CONFIG_PACKAGE_rtl8192cu-firmware=y
CONFIG_PACKAGE_rtl8192de-firmware=y
CONFIG_PACKAGE_rtl8192du-firmware=y
CONFIG_PACKAGE_rtl8192eu-firmware=y
CONFIG_PACKAGE_rtl8192se-firmware=y
CONFIG_PACKAGE_terminfo=y
# CONFIG_PACKAGE_wpad-basic-mbedtls is not set
CONFIG_PACKAGE_zlib=y
# CONFIG_TARGET_EXT4_JOURNAL is not set
CONFIG_TARGET_KERNEL_PARTSIZE=128
CONFIG_TARGET_ROOTFS_PARTSIZE=2048
# CONFIG_TARGET_ROOTFS_SQUASHFS is not set
# CONFIG_USE_FS_ACL_ATTR is not set
CONFIG_ZSTD_OPTIMIZE_O3=y
CONFIG_PACKAGE_jansson=y
CONFIG_PACKAGE_kmod-lib-crc32c=y
CONFIG_PACKAGE_kmod-mmc=y
CONFIG_PACKAGE_libmnl=y
CONFIG_PACKAGE_libsmartcols=y
CONFIG_PACKAGE_ucode-mod-uloop=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ZerBea, 2024-11-10:
I switched to the master branch and mac80211 compiles fine:
```
 make[3] -C package/network/config/wifi-scripts compile
 make[3] -C package/kernel/mac80211 compile
 make[3] -C package/libs/ncurses compile
```

namiltd, 2024-11-10:
Fixed: https://github.com/openwrt/openwrt/commit/4b413687e68084d2c81472ae40eaa7d15e89c37b

ZerBea, 2024-11-13:
After a git pull the issue is back. Beside 24.10 branch, now the master is affected too.

namiltd, 2024-11-13:
I generated the ramips/mt7620 image from github 24.10 branch two hours ago and the problem didn't occur.

ZerBea, 2024-11-13:
I'll do a fresh git pull and retry to compile it.

ZerBea, 2024-11-13:
Ok, I see:
 ...nlmsg_multicast_allns-build-error-on-kernel-6.6.59.patch | 169 ++++++++++++++++++++++++++

ZerBea, 2024-11-13:
Ok, both branches compiling fine again:
 make[3] -C package/network/config/wifi-scripts compile
 make[3] -C package/kernel/mac80211 compile
 make[3] -C package/libs/ncurses compile

Again, thanks.

closed 2024-11-13 by ZerBea