openwrt/openwrt #16907: apk: ubus call rpc-sys packagelist '{ "all": true }'  not functional

dibdot, 2024-11-10:
### Describe the bug

Fresh apk-only snapshots build. The rpc-sys ubus object no longer works, e.g.:
```
root@blackhole:~# ubus -v call rpc-sys packagelist '{ "all": true }'
Command failed: Not found
```

```
rpc module is installed:
rpcd-mod-rpcsys-2024.09.17~9f4b86e7-r1 aarch64_cortex-a53 {feeds/base/system/rpcd} (ISC) [installed]

ubus object methods (ubus -v list):
'rpc-sys' @c6819a21
	"packagelist":{"all":"Boolean"}
	"password_set":{"user":"String","password":"String"}
	"upgrade_test":{}
	"upgrade_start":{"keep":"Boolean"}
	"upgrade_clean":{}
	"factory":{}
	"reboot":{}
```

### OpenWrt version

r28050-be952e98bc

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R3

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

efahl, 2024-11-10:
I asked about this on the devel list a couple weeks ago.  If I knew where the files were going to live, I'd do it myself but I haven't been able to get a working build with apk.

https://lists.openwrt.org/pipermail/openwrt-devel/2024-November/043378.html

john-tho, 2024-11-11:
> I asked about this on the devel list a couple weeks ago. If I knew where the files were going to live, I'd do it myself but I haven't been able to get a working build with apk.

Try:
https://lxr.openwrt.org/ident?i=rpc_sys_packagelist
https://lxr.openwrt.org/source/rpcd/sys.c#L184
Package from:
https://github.com/openwrt/openwrt/blob/main/package/system/rpcd/Makefile

efahl, 2024-11-11:
I know about the source, I need to know the installed package database location on the devices.  Is it in `/etc/apk/world` or somewhere else?

efahl, 2024-11-11:
And likewise for the whole installed packages database.  Trivially convert this to C, filter (or not) with the contents of `world` for the `all` parameter, and you're off to the races.  But only if you know where the files live in OpenWrt.

`awk -F: '/^P:/ {p=$2} /^V:/ {printf "%s %s\n", p, $2}' /lib/apk/db/installed`

john-tho, 2024-11-11:
> /lib/apk/db/installed

Looks right:

https://github.com/openwrt/openwrt/issues/16794#issue-2616013323

or

```
curl -L -o openwrt-x86-64-rootfs.tar.gz https://mirror-03.infra.openwrt.org/snapshots/targets/x86/64/openwrt-x86-64-rootfs.tar.gz

tar -tf openwrt-x86-64-rootfs.tar.gz | grep apk | head

./etc/apk/
./etc/apk/arch
./etc/apk/keys/
./etc/apk/keys/openwrt-snapshots.pem
./etc/apk/repositories
./etc/apk/world
./lib/apk/
./lib/apk/db/
./lib/apk/db/installed
./lib/apk/db/scripts.tar

tar -Oxf openwrt-x86-64-rootfs.tar.gz ./lib/apk/db/installed | head

C:Q16CbYpTzHBrFxsR0LfoNt8gPFyTY=
P:apk-mbedtls
V:3.0.0_pre20241108-r1
A:x86_64
S:142159
I:327680
T:apk package manager (mbedtls)
U:https://gitlab.alpinelinux.org/alpine/apk-tools.git
L:GPL-2.0-only
o:feeds/base/system/apk

# curl -OL https://mirror-03.infra.openwrt.org/snapshots/targets/armsr/armv8/openwrt-armsr-armv8-generic-initramfs-kernel.bin
# packages build not ready yet

curl -OL https://buildbot.aparcar.org/targets/armsr/armv8/openwrt-armsr-armv8-generic-initramfs-kernel.bin
qemu-system-aarch64 -machine virt -cpu cortex-a57 -nographic -kernel openwrt-armsr-armv8-generic-initramfs-kernel.bin -netdev user,id=testlan -net nic,netdev=testlan -netdev user,id=testwan -net nic,netdev=testwan

apk update

apk add strace

strace apk list --full

newfstatat(3, "lib/apk/db", {st_mode=S_IFDIR|0755, st_size=120, ...}, 0) = 0
openat(3, "etc/apk/arch", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 4
fstat(4, {st_mode=S_IFREG|0644, st_size=16, ...}) = 0
read(4, "aarch64_generic\n", 16)        = 16
close(4)                                = 0
openat(3, "etc/apk/protected_paths.d", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(3, "etc/apk/cache", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(3, "var/cache/apk", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 4
openat(3, "lib/apk/db", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 5
openat(3, "etc/apk/world", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 6

ls /var/cache/apk/
APKINDEX.18edc754.tar.gz  APKINDEX.7e2b4bb5.tar.gz  APKINDEX.fc17cb00.tar.gz
APKINDEX.368457d3.tar.gz  APKINDEX.8ce995ad.tar.gz
APKINDEX.61a771b6.tar.gz  APKINDEX.a7ceacd8.tar.gz

```

efahl, 2024-11-14:
More discussion starting here: https://github.com/openwrt/openwrt/commit/40b8fbaa9754c86480eefc3692c9116a51a64718#commitcomment-149041733

vincejv, 2024-11-15:
This breaks banIP when using w/ `uclient-fetch` which relies on that broken ubus call command, pretty sure the maintainer is aware since he raised the issue, but it might be worth noting for users looking up the issue...
```
Fri Nov 15 12:46:22 2024 user.info banIP-[55975]: start banIP processing (restart)
Fri Nov 15 12:46:22 2024 user.err banIP-[55975]: no download utility with SSL support
Fri Nov 15 12:46:24 2024 user.info banIP-[56276]: start banIP processing (reload)
Fri Nov 15 12:46:24 2024 user.err banIP-[56276]: no download utility with SSL support
Fri Nov 15 12:48:31 2024 user.info banIP-[61571]: start banIP processing (restart)
Fri Nov 15 12:48:31 2024 user.err banIP-[61571]: no download utility with SSL support
```

dibdot, 2024-11-15:
@vincejv Just disable the autodetection and set interfaces & tools manually, e.g.: 

[image]

closed 2024-12-02 by openwrt-bot

efahl, 2024-12-13:
The `Command failed: Not found` message usually indicates that either the package is not installed or that the installed one is not up-to-date.  Since you've verified those, seems like an `strace` of the ubus call would give us some clues.