openwrt/openwrt #16922: Network device is not present - No Internet after update

TurtlePL, 2024-11-11:
### Describe the bug

I don't have internet after updating.

### OpenWrt version

r28047-999ef8270f

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ramips/mt7621

### Device

Zyxel LTE5398-M904

### Image kind

Official downloaded image

### Actual behaviour

[image]

root@OpenWrt:~# cat /sys/kernel/debug/usb/devices
root@OpenWrt:~# (No answer)

### Additional info

`root@OpenWrt:~# uci show network.wan
network.wan=interface
network.wan.device='/dev/cdc-wdm0'
network.wan.proto='qmi'
network.wan.pdptype='ipv4'
network.wan.apn='internet'
network.wan.auth='pap'
network.wan.username='internet'
network.wan.password='internet'`

lsmod
`authenc                12288  2 crypto_hw_eip93,authencesn
authencesn             12288  0
cdc_wdm                16384  1 qmi_wwan
cfg80211              335872  5 mt7615_common,mt7603e,mt76_connac_lib,mt76,mac80211
cmac                   12288  0
compat                 12288  2 mac80211,cfg80211
crc_ccitt              12288  1 ppp_async
crc32c_generic         12288  1
crypto_hw_eip93        40960  0
des_generic            12288  0
drbg                   24576  0
geniv                  12288  1 seqiv
gpio_button_hotplug    12288  0
hmac                   12288  0
hwmon                  16384  1 mt7615_common
jitterentropy_rng      16384  0
leds_gpio              12288  0
ledtrig_usbport        12288  0
libcrc32c              12288  1 nf_tables
libdes                 24576  2 crypto_hw_eip93,des_generic
mac80211              675840  5 mt7615e,mt7615_common,mt7603e,mt76_connac_lib,mt76
md5                    12288  0
mii                    16384  1 usbnet
mt76                   65536  4 mt7615e,mt7615_common,mt7603e,mt76_connac_lib
mt76_connac_lib        53248  2 mt7615e,mt7615_common
mt7603e                45056  0
mt7615_common          73728  1 mt7615e
mt7615e                20480  0
nf_conntrack           94208  7 nft_redir,nft_nat,nft_masq,nft_flow_offload,nft_ct,nf_nat,nf_flow_table
nf_defrag_ipv4         12288  1 nf_conntrack
nf_defrag_ipv6         16384  1 nf_conntrack
nf_flow_table          32768  2 nf_flow_table_inet,nft_flow_offload
nf_flow_table_inet     12288  0
nf_log_syslog          20480  0
nf_nat                 36864  4 nft_redir,nft_nat,nft_masq,nft_chain_nat
nf_reject_ipv4         12288  2 nft_reject_ipv4,nft_reject_inet
nf_reject_ipv6         16384  2 nft_reject_ipv6,nft_reject_inet
nf_tables             208896156 nft_fib_inet,nf_flow_table_inet,nft_reject_ipv6,nft_reject_ipv4,nft_reject_inet,nft_reject,nft_redir,nft_quota,nft_numgen,nft_nat,nft_masq,nft_log,nft_limit,nft_hash,nft_flow_offload,nft_fib_ipv6,nft_fib_ipv4,nft_fib,nft_ct,nft_chain_nat
nfnetlink              16384  1 nf_tables
nft_chain_nat          12288  2
nft_ct                 20480  3
nft_fib                12288  3 nft_fib_inet,nft_fib_ipv6,nft_fib_ipv4
nft_fib_inet           12288  0
nft_fib_ipv4           12288  1 nft_fib_inet
nft_fib_ipv6           12288  1 nft_fib_inet
nft_flow_offload       12288  0
nft_hash               12288  0
nft_limit              12288  5
nft_log                12288  0
nft_masq               12288  1
nft_nat                12288  0
nft_numgen             12288  0
nft_quota              12288  0
nft_redir              12288  0
nft_reject             12288  3 nft_reject_ipv6,nft_reject_ipv4,nft_reject_inet
nft_reject_inet        12288  2
nft_reject_ipv4        12288  0
nft_reject_ipv6        12288  0
nls_base               12288  1 usbcore
option                 45056  0
ppp_async              16384  0
ppp_generic            36864  3 pppoe,ppp_async,pppox
pppoe                  20480  0
pppox                  12288  1 pppoe
qmi_wwan               24576  0
rng                    12288  3 jitterentropy_rng,drbg,geniv
seqiv                  12288  0
sha1_generic           12288  0
sha256_generic         12288  0
sha3_generic           16384  0
sha512_generic         16384  0
slhc                   16384  1 ppp_generic
usb_common             12288  3 xhci_plat_hcd,xhci_hcd,usbcore
usb_wwan               12288  1 option
usbcore               163840 11 option,usb_wwan,qmi_wwan,usbserial,usbnet,cdc_wdm,ledtrig_usbport,xhci_plat_hcd,xhci_pci,xhci_mtk_hcd,xhci_hcd
usbnet                 24576  1 qmi_wwan
usbserial              24576  2 option,usb_wwan
xhci_hcd              126976  3 xhci_plat_hcd,xhci_pci,xhci_mtk_hcd
xhci_mtk_hcd           16384  0
xhci_pci               12288  0
xhci_plat_hcd          12288  0`

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-11-11:
Use 3x backticks to format larger text blocks. OK, qmi module is installed, do you see modem in lsusb/lspci and in boot messages?

TurtlePL, 2024-11-11:
``````[    0.000000] Linux version 6.6.59 (builder@buildhost) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r28047-999ef8270f) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 SMP Sun Nov 10 03:27:19 2024
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is Zyxel LTE5398-M904
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] percpu: Embedded 12 pages/cpu s19104 r8192 d21856 u49152
[    0.000000] pcpu-alloc: s19104 r8192 d21856 u49152 alloc=12*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Kernel command line: console=ttyS0,57600 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Writing ErrCtl register=0000a040
[    0.000000] Readback ErrCtl register=0000a040
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64960
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 247044K/262144K available (8373K kernel code, 652K rwdata, 1776K rodata, 1284K init, 229K bss, 15100K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 256
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000005] sched_clock: 64 bits at 880MHz, resolution 1ns, wraps every 4398046511103ns
[    0.016094] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[    0.088334] pid_max: default: 32768 minimum: 301
[    0.106438] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.120861] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.145646] RCU Tasks Trace: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1.
[    0.161408] rcu: Hierarchical SRCU implementation.
[    0.170840] rcu: 	Max phase no-delay instances is 1000.
[    0.182156] smp: Bringing up secondary CPUs ...
[    0.192156] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.192190] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.192207] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.192265] CPU1 revision is: 0001992f (MIPS 1004Kc)
[    0.251548] Synchronize counters for CPU 1: done.
[    0.314270] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.314297] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.314312] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.314347] CPU2 revision is: 0001992f (MIPS 1004Kc)
[    0.373135] Synchronize counters for CPU 2: done.
[    0.433670] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.433696] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.433710] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.433749] CPU3 revision is: 0001992f (MIPS 1004Kc)
[    0.492738] Synchronize counters for CPU 3: done.
[    0.552359] smp: Brought up 1 node, 4 CPUs
[    0.568211] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.587762] futex hash table entries: 1024 (order: 3, 32768 bytes, linear)
[    0.605392] pinctrl core: initialized pinctrl subsystem
[    0.619224] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[    0.631842] thermal_sys: Registered thermal governor 'step_wise'
[    0.638643] FPU Affinity set after 11720 emulations
[    0.660883] reg-fixed-voltage regulator: Failed to allocate supply name
[    0.674035] reg-fixed-voltage: probe of regulator failed with error -12
[    0.687237] clocksource: Switched to clocksource GIC
[    0.707272] NET: Registered PF_INET protocol family
[    0.717142] IP idents hash table entries: 4096 (order: 3, 32768 bytes, linear)
[... 590 lines cut ...]
Sun Nov 10 19:18:15 2024 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Sun Nov 10 19:18:15 2024 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 4 names
Sun Nov 10 19:18:15 2024 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Sun Nov 10 19:18:15 2024 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.1.179 68:da:73:a7:96:e7
Sun Nov 10 19:18:15 2024 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.1.179 68:da:73:a7:96:e7 Air-Grzegorz
Sun Nov 10 19:18:16 2024 daemon.warn odhcpd[1688]: No default route present, overriding ra_lifetime to 0!
[... cut ...]
Sun Nov 10 19:27:37 2024 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.1.179 68:da:73:a7:96:e7
Sun Nov 10 19:27:38 2024 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.1.179 68:da:73:a7:96:e7
Sun Nov 10 19:27:38 2024 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.1.179 68:da:73:a7:96:e7 Air-Grzegorz
Sun Nov 10 19:27:38 2024 daemon.warn odhcpd[1688]: No default route present, overriding ra_lifetime to 0!
Sun Nov 10 19:27:43 2024 daemon.warn odhcpd[1688]: No default route present, overriding ra_lifetime to 0!```

brada4, 2024-11-11:
In principle modem is detected as present:If not obvious install those tools.
```
[   19.326949] usbserial: USB Serial support registered for GSM modem (1-port)
```

TurtlePL, 2024-11-11:
which tools..? usb-utils and pciutils?
Ok.

```root@OpenWrt:~#` lsusb
unable to initialize libusb: -99
root@OpenWrt:~# lspci
00:00.0 PCI bridge: Device 0e8d:0801 (rev 01)
00:01.0 PCI bridge: Device 0e8d:0801 (rev 01)
01:00.0 Network controller: MEDIATEK Corp. MT7603E 802.11bgn PCI Express Wireless Network Adapter
02:00.0 Unclassified device [0002]: MEDIATEK Corp. MT7615E 802.11ac PCI Express Wireless Network Adapter
root@OpenWrt:~#       ```

mkrle, 2024-11-11:
@neheb: Just noticed this is present in the 'dmesg' since e612900ae0fd6c4624fa661e004536216409107f 
```
[    0.660883] reg-fixed-voltage regulator: Failed to allocate supply name
[    0.674035] reg-fixed-voltage: probe of regulator failed with error -12
```
I don't quite get this since you did provide the name in the DTS.

TurtlePL, 2024-11-11:
> @neheb: Just noticed this is present in the 'dmesg' since [e612900](https://github.com/openwrt/openwrt/commit/e612900ae0fd6c4624fa661e004536216409107f)
> 
> ```
> [    0.660883] reg-fixed-voltage regulator: Failed to allocate supply name
> [    0.674035] reg-fixed-voltage: probe of regulator failed with error -12
> ```
> 
> I don't quite get this since you did provide the name in the DTS.

I don't know if it changes anything, but there is a typo in line 99.

mkrle, 2024-11-11:
> I don't know if it changes anything, but there is a typo in line 99. (reglator)

Ah LOL I think that's it! :)

@neheb would you agree?

neheb, 2024-11-11:
I would.

So this LTE modem is connected through USB instead of PCIe?

mkrle, 2024-11-11:
> So this LTE modem is connected through USB instead of PCIe?

Yes. I don't have the device at hand, but IIRC before the changes `usb_power` would control the external usb port and `lte_power` would control the internal USB port where the LTE modem is "connected". The logic as specified in the DTS was opposite in regard to GPIO output level between these two.  And the result was that you could power (cycle) these two independently. Very useful to (cold) restart the modem without rebooting the router.

neheb, 2024-11-11:
Sure. As implemented, the USB regulator does not support userspace control. Could add it fairly easily.

I could be wrong but regulators are probably there to deal with suspend, resume, poweroff, halt, etc...

neheb, 2024-11-11:
@TurtlePL fix was merged. Please retest.

mkrle, 2024-11-12:
The folks on the [forum](https://forum.openwrt.org/t/openwrt-support-for-zyxel-lte5398-m904/140157/224?u=mkrle) were kind enough to test my original assumption: `usb_power` controls the external port while `lte_power` controls the internal port with the LTE modem.

I'm not familiar with this topic, so I'm looking for opinion whether it's still correct to mark `usb_power` as the regulator for the entire XHCI controller with the above in mind.

TurtlePL, 2024-11-12:
Now it works as it should. 
Thanks