openwrt/openwrt #16924: [24.10] boot failure on mt7975

JKSTAFF, 2024-11-11:
### Describe the bug

Similar problem as #16273
crash log:
```
[   23.287279] mt798x-wmac 18000000.wifi: HW/SW Version: 0x8a108a10, Build Time: 20240823160739a
[   23.287279]
[   23.412779] mt798x-wmac 18000000.wifi: WM Firmware Version: ____000000, Build Time: 20240823160804
[   23.512406] mt798x-wmac 18000000.wifi: WA Firmware Version: DEV_000000, Build Time: 20240823160840
[   43.993447] mt798x-wmac 18000000.wifi: Retry message 00000002 (seq 11)
[   64.473442] mt798x-wmac 18000000.wifi: Message 00000002 (seq 11) timeout
[   64.480132] mt798x-wmac 18000000.wifi: Failed to start WA firmware
[   64.480152] Unable to handle kernel write to read-only memory at virtual address 0000000000000008
[   64.495146] Mem abort info:
[   64.497924]   ESR = 0x0000000096000045
[   64.501656]   EC = 0x25: DABT (current EL), IL = 32 bits
[   64.506951]   SET = 0, FnV = 0
[   64.509989]   EA = 0, S1PTW = 0
[   64.513114]   FSC = 0x05: level 1 translation fault
[   64.517976] Data abort info:
[   64.520841]   ISV = 0, ISS = 0x00000045, ISS2 = 0x00000000
[   64.526309]   CM = 0, WnR = 1, TnD = 0, TagAccess = 0
[   64.531341]   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0
[   64.536640] user pgtable: 4k pages, 39-bit VAs, pgdp=0000000043b82000
[   64.543059] [0000000000000008] pgd=0000000000000000, p4d=0000000000000000, pud=0000000000000000
[   64.551742] Internal error: Oops: 0000000096000045 [#1] SMP
[   64.557295] Modules linked in: mt7915e(O+) mt76_connac_lib(O) mt76(O) mac80211(O) libchacha20poly1305 ipt_REJECT huawei_cdc_ncm chacha_neon cfg80211(O) cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_quota xt_pkttype xt_physdev xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_cgroup xt_addrtype xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_DSCP xt_CLASSIFY usbserial usbnet tcp_bbr slhc sfp poly1305_neon nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_compat nft_chain_nat nf_tables nf_reject_ipv4 nf_log_syslog nf_flow_table nf_conncount mdio_i2c libcurve25519_generic libchacha iptable_nat iptable_mangle iptable_filter ipt_ECN ip_tables compat(O) cdc_wdm br_netfilter crypto_safexcel cls_bpf act_bpf sch_tbf sch_ingress sch_htb
[   64.557473]  sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact pwm_fan i2c_gpio i2c_algo_bit cryptodev(O) xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ksmbd nls_ucs2_utils cifs_arc4 asn1_decoder ip6_udp_tunnel udp_tunnel xsk_diag oid_registry veth nls_utf8 xxhash_generic sha512_arm64 sha1_ce sha1_generic seqiv md5 geniv des_generic libdes blake2b_generic authencesn authenc usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd nvme nvme_core gpio_button_hotplug(O) exfat btrfs xor xor_neon raid6_pq libcrc32c usbcore usb_common
[   64.647328]  aquantia mii
[   64.739545] CPU: 2 PID: 11 Comm: kworker/u8:0 Tainted: G           O       6.6.60 #0
[   64.747267] Hardware name: Bananapi BPI-R3 (DT)
[   64.751779] Workqueue: mt76 mt7915_mac_reset_work [mt7915e]
[   64.757352] pstate: a0400005 (NzCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[   64.764292] pc : __mt76_tx_complete_skb+0x5a8/0x738 [mt76]
[   64.769766] lr : mt76_txq_schedule_all+0x18/0x3c [mt76]
[   64.774978] sp : ffffffc081393c10
[   64.778276] x29: ffffffc081393c10 x28: ffffffc081393c80 x27: ffffff8007eb9fe0
[   64.785392] x26: ffffffffffff5af8 x25: ffffff8007eb1ff8 x24: ffffff8007eb5538
[   64.792508] x23: ffffff8007eb26d0 x22: ffffff8007eb1fe0 x21: ffffff8007eb2004
[   64.799623] x20: ffffff8007eb1fe0 x19: 0000000000000000 x18: 00000000dd591802
[   64.806739] x17: ffffffc081393c50 x16: ffffffc081393be0 x15: 000000d2a81d9100
[   64.813855] x14: 0000000000000000 x13: 0000000000000020 x12: 0101010101010101
[   64.820971] x11: 7f7f7f7f7f7f7f7f x10: fefefefefefefeff x9 : 7f7f7f7f7f7f7f7f
[   64.828086] x8 : 8080808080808080 x7 : ffffff800000aec0 x6 : 0000000000000000
[   64.835202] x5 : ffffffc081393c78 x4 : 0000000000000000 x3 : ffffff8007eb2008
[   64.842318] x2 : 0000000000000000 x1 : 0000000000000000 x0 : 0000000000000000
[   64.849433] Call trace:
[   64.851864]  __mt76_tx_complete_skb+0x5a8/0x738 [mt76]
[   64.856990]  mt76_txq_schedule_all+0x18/0x3c [mt76]
[   64.861855]  mt7915_mac_reset_work+0xc1c/0xe28 [mt7915e]
[   64.867157]  process_one_work+0x178/0x394
[   64.871154]  worker_thread+0x2e8/0x4d0
[   64.874887]  kthread+0xd8/0xdc
[   64.877927]  ret_from_fork+0x10/0x20
[   64.881491] Code: f94037e3 eb00007f 54000120 a94286c2 (f900045c)
[   64.887563] ---[ end trace 0000000000000000 ]---
[   64.895739] pstore: backend (ramoops) writing error (-28)
[   64.901122] Kernel panic - not syncing: Oops: Fatal exception in interrupt
[   64.907974] SMP: stopping secondary CPUs
[   64.911882] Kernel Offset: disabled
[   64.915353] CPU features: 0x0,00000000,00000000,1000400b
[   64.920646] Memory Limit: none
[   64.927162] Rebooting in 1 seconds..
```

### OpenWrt version

r28017-92eb8

### OpenWrt release

24.10

### OpenWrt target/subtarget

mediatek/filogic

### Device

bananapi bpi r3

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Jalakas, 2024-12-23:
Same here:

[   16.756324] mt798x-wmac 18000000.wifi: HW/SW Version: 0x8a108a10, Build Time: 20240823160739a
[   16.756324] 
[   16.882590] mt798x-wmac 18000000.wifi: WM Firmware Version: ____000000, Build Time: 20240823160804
[   16.980346] mt798x-wmac 18000000.wifi: WA Firmware Version: DEV_000000, Build Time: 20240823160840
[   37.592515] mt798x-wmac 18000000.wifi: Retry message 00000002 (seq 11)
[   58.072511] mt798x-wmac 18000000.wifi: Message 00000002 (seq 11) timeout
[   58.079200] mt798x-wmac 18000000.wifi: Failed to start WA firmware
[   58.079218] Unable to handle kernel write to read-only memory at virtual address 0000000000000008
[   58.094213] Mem abort info:
[   58.096992]   ESR = 0x0000000096000045
[   58.100723]   EC = 0x25: DABT (current EL), IL = 32 bits
[   58.106018]   SET = 0, FnV = 0
[   58.109056]   EA = 0, S1PTW = 0
[   58.112181]   FSC = 0x05: level 1 translation fault
[   58.117043] Data abort info:
[   58.119907]   ISV = 0, ISS = 0x00000045, ISS2 = 0x00000000
[   58.125375]   CM = 0, WnR = 1, TnD = 0, TagAccess = 0
[   58.130407]   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0
[   58.135702] user pgtable: 4k pages, 39-bit VAs, pgdp=000000004023d000
[   58.142122] [0000000000000008] pgd=0800000041c4b003, p4d=0800000041c4b003, pud=0800000041c4b003, pmd=0000000000000000
[   58.152711] Internal error: Oops: 0000000096000045 [#1] SMP
[   58.158265] Modules linked in: mt7915e(O+) mt76_connac_lib(O) mt76(O) mac80211(O) ipt_REJECT huawei_cdc_ncm cfg80211(O) cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_recent xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY usbserial usbnet slhc sfp nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_quota nft_numgen nft_log nft_limit nft_hash nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_compat nf_tables nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 mdio_i2c libcrc32c iptable_mangle iptable_filter ipt_ECN ip_tables compat(O) cdc_wdm cdc_acm crypto_safexcel pwm_fan i2c_gpio i2c_algo_bit xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink
[   58.158414]  ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 zram zsmalloc sha512_arm64 sha1_ce sha1_generic seqiv md5 geniv des_generic libdes authencesn authenc leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd gpio_button_hotplug(O) usbcore usb_common aquantia mii
[   58.274074] CPU: 3 PID: 11 Comm: kworker/u8:0 Tainted: G           O       6.6.67 #0
[   58.281795] Hardware name: Bananapi BPI-R3 (DT)
[   58.286308] Workqueue: mt76 mt7915_mac_reset_work [mt7915e]
[   58.291881] pstate: a0400005 (NzCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[   58.298821] pc : __mt76_tx_complete_skb+0x5bc/0xb4c [mt76]
[   58.304296] lr : mt76_txq_schedule_all+0x18/0x60 [mt76]
[   58.309507] sp : ffffffc080e13c20
[   58.312806] x29: ffffffc080e13c20 x28: ffffffc080e13c90 x27: ffffff8005899fe0
[   58.319922] x26: ffffffffffff5af8 x25: ffffff8005891ff8 x24: ffffff80058955f8
[   58.327038] x23: ffffff8005892790 x22: ffffff8005891fe0 x21: ffffff8005892008
[   58.334154] x20: ffffff8005891fe0 x19: ffffff8005891fe0 x18: 00000000fe53f70e
[   58.341269] x17: ffffffc082323c10 x16: 00000000a217a0ae x15: 0000003159ca8400
[   58.348386] x14: 0000000000000000 x13: 0000000000000020 x12: 0101010101010101
[   58.355501] x11: 7f7f7f7f7f7f7f7f x10: fefefefefefefeff x9 : 7f7f7f7f7f7f7f7f
[   58.362616] x8 : 8080808080808080 x7 : 0000000000000005 x6 : ffffffffffff24e0
[   58.369733] x5 : ffffffc080e13c88 x4 : 0000000000000000 x3 : ffffff8005892010
[   58.376847] x2 : 0000000000000000 x1 : 0000000000000000 x0 : 0000000000000000
[   58.383963] Call trace:
[   58.386395]  __mt76_tx_complete_skb+0x5bc/0xb4c [mt76]
[   58.391520]  mt76_txq_schedule_all+0x18/0x60 [mt76]
[   58.396386]  mt7915_mac_reset_work+0xc14/0xe18 [mt7915e]
[   58.401687]  process_one_work+0x174/0x2c8
[   58.405686]  worker_thread+0x2e4/0x4c8
[   58.409420]  kthread+0xd8/0xdc
[   58.412462]  ret_from_fork+0x10/0x20
[   58.416025] Code: f94037e3 eb00007f 54000120 a94306c2 (f900045c) 
[   58.422098] ---[ end trace 0000000000000000 ]---
[   58.429801] pstore: backend (ramoops) writing error (-28)
[   58.435181] Kernel panic - not syncing: Oops: Fatal exception in interrupt
[   58.442033] SMP: stopping secondary CPUs
[   58.445941] Kernel Offset: disabled
[   58.449413] CPU features: 0x0,00000000,00000000,1000400b
[   58.454706] Memory Limit: none
[   58.460714] Rebooting in 1 seconds..

Jalakas, 2024-12-23:
Same here:
- https://github.com/openwrt/openwrt/issues/10781
- https://github.com/openwrt/mt76/issues/702

closed 2024-12-23 by JKSTAFF