openwrt/openwrt #16929: apk: No buffer space available

vortexilation, 2024-11-12:
### Describe the bug

Getting the following when trying to build the image : 
```
(2351/2353) Installing zoneinfo-poles (2024b-r1)
(2352/2353) Installing zoneinfo-all (2024b-r1)
(2353/2353) Installing zstd (1.5.6-r1)
ERROR: System state may be inconsistent: failed to write database: No buffer space available
1 error; 2704 MiB in 2353 packages
make[2]: *** [package/Makefile:99: package/install] Error 1
```

The same configuration work fine when using opkg. Only happening after the APK switch.

### OpenWrt version

r28062-be3e7a84bd

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

x86/64

### Device

generic

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ynezz, 2024-11-12:
### Diffconfig

In case of a self-built image, please attach diffconfig.

```shell
./scripts/diffconfig.sh
```

> ERROR: System state may be inconsistent: failed to write database: No buffer space available

Possible callstack:

- apk_db_write_config()
   - apk_db_index_write_nr_cache()
     - apk_pkg_write_index_entry()
       - apk_db_fdb_write()
         - apk_blob_pushed() // returns NULL if buffer overflow
           - [-ENOBUFS](https://gitlab.alpinelinux.org/alpine/apk-tools/-/blob/master/src/database.c?ref_type=heads#L1065)

ynezz, 2024-11-12:
Seems to be already handled in https://gitlab.alpinelinux.org/alpine/apk-tools/-/issues/11038

vortexilation, 2024-11-13:
Yes, upstream committed a potential fix, still recompiling OpenWrt to check if it's really fixed it.

closed 2024-11-13 by openwrt-bot

peterwillcn, 2024-11-14:
https://github.com/openwrt/openwrt/actions/runs/11832629914/job/32969780999#step:34:711

commit a6c248e8 (2024-11-15): apk: fix long package description handling

Currently its not possible to generate apk package.adb package index if
the package has longer description field, which leads to following
failure:

 (2352/2353) Installing zoneinfo-all (2024b-r1)
 (2353/2353) Installing zstd (1.5.6-r1)
 ERROR: System state may be inconsistent: failed to write database: No buffer space available
 1 error; 2704 MiB in 2353 packages

The code to read/write installeddb does not really handle long
description well. Until the database is converted to apkv3 format,
truncate the apkv3 descriptions to allow existing code to  [... cut]
PR #16951 (2024-11-15): apk: fix long package description handling
### apk: fix long package description handling

Currently its not possible to generate apk package.adb package index if
the package has longer description field, which leads to following
failure:

```shell
 (2352/2353) Installing zoneinfo-all (2024b-r1)
 (2353/2353) Installing zstd (1.5.6-r1)
 ERROR: System state may be inconsistent: failed to write database: No buffer space available
 1 error; 2704 MiB in 2353 packages
```

The code to read/write installeddb does not really handle long
description well. Until the database is converted to apkv3 format [... cut]