openwrt/openwrt #16932: Official snapshot ImageBuilder stopped packing `/sbin/init` into images, causing kernel panic and bootloop

kuanyili, 2024-11-12:
### Describe the bug

As title.

### OpenWrt version

r28062-be3e7a84bd

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

x86/64

### Device

QEMU Standard PC (Q35 + ICH9, 2009)

### Image kind

Self-built image

### Steps to reproduce

1. download [ImageBuilder](https://downloads.openwrt.org/snapshots/targets/x86/64/openwrt-imagebuilder-x86-64.Linux-x86_64.tar.zst) generated by official CI
2. build image by running `make image PROFILE=generic`
3. boot with the output image

### Actual behaviour

Bootloop with error message

```
...skipped...
[    1.073224] Run /sbin/init as init process
[    1.074600] Run /etc/init as init process
[    1.075993] Run /bin/init as init process
[    1.076418] Run /bin/sh as init process
[    1.076823] Kernel panic - not syncing: No working init found.  Try passing init= option to kernel. See Linux Documentation/admin-guide/init.rst for guidance.
[    1.078045] Kernel Offset: disabled
[    1.078587] Rebooting in 1 seconds..
```

### Expected behaviour

Boot successfully

### Additional info

The images generated by official CI boots fine though.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ynezz, 2024-11-13:
Fixed via 4c65359af4

closed 2024-11-13 by ynezz

schm0, 2024-11-25:
Is this still broken?
I build an image with kernel name spaces disabled and it didn't create the sbin folder.