openwrt/openwrt #16935: [X86_64] Can't update opkg

congphat1998, 2024-11-13:
### Describe the bug

Collected errors:
 * opkg_download: Failed to download http://downloads.openwrt.org/snapshots/targets/x86/64/packages/Packages.gz, wget returned 8.
 * opkg_download: Failed to download http://downloads.openwrt.org/snapshots/packages/x86_64/base/Packages.gz, wget returned 8.
 * opkg_download: Failed to download http://downloads.openwrt.org/snapshots/packages/x86_64/luci/Packages.gz, wget returned 8.
 * opkg_download: Failed to download http://downloads.openwrt.org/snapshots/packages/x86_64/packages/Packages.gz, wget returned 8.
 * opkg_download: Failed to download http://downloads.openwrt.org/snapshots/packages/x86_64/routing/Packages.gz, wget returned 8.
 * opkg_download: Failed to download http://downloads.openwrt.org/snapshots/packages/x86_64/telephony/Packages.gz, wget returned 8.

### OpenWrt version

24.04

### OpenWrt release

(r28213+1-f422e381101)

### OpenWrt target/subtarget

24.04

### Device

x86 64 devices generic

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

congphat1998, 2024-11-13:
There is no url for that
[image]

openips, 2024-11-13:
same bug with bpi r4

castiel652, 2024-11-13:
opkg is not used anymore

ynezz, 2024-11-13:
Its not a bug, package manager [was switched](https://lists.openwrt.org/pipermail/openwrt-devel/2024-November/043378.html) from opkg to apk https://forum.openwrt.org/t/the-future-is-now-opkg-vs-apk/201164

closed 2024-11-13 by ynezz

openips, 2024-11-13:
luci still use opkg to  manage packages

Nifty-07, 2024-11-16:
I get the same error messages (wget return 8).  It appears as the repository is no longer there (404 Not Found) which makes it rather hard to install apk.  

I found a repository here: https://mirror-03.infra.openwrt.org/snapshots/packages/mipsel_24kc/

Unfortunately this returns a check signature error.

Appreciate any advice how I can download apk.

ssbbpp6162, 2024-12-03:
Collected errors:
 * opkg_download: Failed to download https://downloads.openwrt.org/snapshots/targets/x86/64/packages/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/snapshots/packages/x86_64/base/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/snapshots/packages/x86_64/diskman/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/snapshots/packages/x86_64/kenzo/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/snapshots/packages/x86_64/luci/Packages.gz, wget returned 8.

.......

adamwhats, 2024-12-03:
@Nifty-07  I've bumped into the same issue. Looks like the most recent x86 docker image was created on the 5/11/24, which was just before the switch from opkg to APK.
```
adam@z100:~$ docker image inspect openwrt/rootfs:latest | grep created
                "org.opencontainers.image.created": "2024-11-05T17:52:06.069Z",
``` 
I've been able to build a new rootfs docker image using:
```
docker pull openwrt/imagebuilder
mkdir -p ./bin/targets
docker run --rm -v ./bin/:/builder/bin -it openwrt/imagebuilder sh -c "./setup.sh && make image PROFILE=generic PACKAGES=tmate"
mkdir openwrt-rootfs
tar -xvzf ./bin/targets/x86/64/openwrt-x86-64-generic-rootfs.tar.gz -C openwrt-rootfs
tar -C openwrt-rootfs -c . | docker import - openwrt/rootfs:latest
```