openwrt/openwrt #16936: gcc: enable linker-build-id when configuring?

Ra2-IFV, 2024-11-13:
### Describe the bug

When running `configure` for `elfutils`, it will give this warning:
`configure: WARNING: compiler doesn't generate build-id by default`
This is because gcc isn't configured with `--enable-linker-build-id`, as the maintainer of `elfutils` pointed out in this [bug report](https://sourceware.org/bugzilla/show_bug.cgi?id=32311). I don't see this option in gcc configure options in this [makefile](https://github.com/openwrt/openwrt/blob/main/toolchain/gcc/common.mk).

> You might want to discuss with your distro to enable them, I believe every other distro does. And various elfutils tools kind of depend on build-ids.

### OpenWrt version

cdfe413d263409e33eda936c34c4338eed66b5a5

### OpenWrt release

cdfe413d263409e33eda936c34c4338eed66b5a5

### OpenWrt target/subtarget

bcm27xx/bcm2711

### Device

Raspberry Pi Compute Module 4

### Image kind

Self-built image

### Steps to reproduce

```bash
make package/elfutils/{clean,prepare,compile} -j1 V=s > elfutils.log 2>&1
```

### Diffconfig

```text
CONFIG_PACKAGE_apk-mbedtls=y
# CONFIG_PACKAGE_opkg is not set
CONFIG_PACKAGE_zlib=y
CONFIG_USE_APK=y
CONFIG_USE_GC_SECTIONS=y
CONFIG_USE_LTO=y
CONFIG_USE_MOLD=y
CONFIG_ZLIB_OPTIMIZE_SPEED=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-11-14:
It is extra super repetitive metainfo line.

brada4, 2024-11-18:
Tbis distro does tbose pesky reproducible builds, where your build-id option may not apply ever.

ynezz, 2024-11-18:
> Tbis distro does tbose pesky reproducible builds, where your build-id option may not apply ever.

Its not only distro doing reproducible builds and I believe, that it can be set in reproducible way.
Additionally build-id is probably going to be needed for https://github.com/openwrt/openwrt/pull/15116

Anyways, this is bug tracker, this is not bug. For feature requests use other communications channels https://openwrt.org/contact

closed 2024-11-18 by ynezz