openwrt/openwrt #16946: apk: a few transitional issues

stangri, 2024-11-13:
### Describe the bug

Sorry @aparcar if there's a master issue for all transitional problems I couldn't find it when I searched open issues for `apk`.

I'm facing a few issues:
1. When using snapshots SDK (dunno if it matters, but I cloned the openwrt repo yesterday and built the SDK from sources and using that SDK) to build my packages, my luci apps fail to build with the following error:
```
/home/ubuntu/development/Builder/SDK_x86-64_snapshot/staging_dir/host/bin/fakeroot /home/ubuntu/development/Builder/SDK_x86-64_snapshot/staging_dir/host/bin/apk mkpkg --info "name:luci-app-adblock-fast" --info "version:1.1.2-12" --info "description:Provides Web UI for adblock-fast service." --info "arch:all" --info "license:AGPL-3.0-or-later" --info "origin:feeds/base/luci/applications/luci-app-adblock-fast" --info "url:https://github.com/stangri/luci-app-adblock-fast/" --info "maintainer:OpenWrt LuCI community" --info "provides:"  --script "post-install:/home/ubuntu/development/Builder/SDK_x86-64_snapshot/build_dir/target-x86_64_musl/luci-app-adblock-fast/apk-all/luci-app-adblock-fast/post-install" --script "pre-deinstall:/home/ubuntu/development/Builder/SDK_x86-64_snapshot/build_dir/target-x86_64_musl/luci-app-adblock-fast/apk-all/luci-app-adblock-fast/pre-deinstall" --info "depends:libc luci-base adblock-fast jsonfilter" --files "/home/ubuntu/development/Builder/SDK_x86-64_snapshot/build_dir/target-x86_64_musl/luci-app-adblock-fast/ipkg-all/luci-app-adblock-fast" --output "/home/ubuntu/development/Builder/SDK_x86-64_snapshot/bin/packages/x86_64/base/luci-app-adblock-fast-1.1.2-12.apk" --sign "/home/ubuntu/development/Builder/SDK_x86-64_snapshot/private-key.pem"
ERROR: info field 'version' has invalid value: package version is invalid
ERROR: failed to create package: /home/ubuntu/development/Builder/SDK_x86-64_snapshot/bin/packages/x86_64/base/
```
They have exactly the same PKG_VERSION/PKG_RELEASE as principal packages which build just fine. I remember @hnyman bringing the issue up a few days ago, so while something was fixed that the luci apps with explicit PKG_VERSION/PKG_RELEASE set do build from sources ((evidence)[https://downloads.openwrt.org/snapshots/packages/x86_64/luci/luci-app-adblock-fast-1.1.2-r10.apk]), the SDK created from sources fails to build the luci apps with explicit PKG_VERSION/PKG_RELEASE set.

2. The apk binaries produced by SDK do not contain arch in the filename like the ipk binaries did. For example, a filename for the binary produced by the APK-enabled SDK is: `https-dns-proxy-2023.12.26-r1.apk`, while ipk binary is called `https-dns-proxy_2023.12.26-1_x86_64.ipk`. Is it by design or is this an oversight which needs to be addressed?

3. For ipk binaries there's a `./scripts/ipkg-make-index.sh` which I use to create the manifest file and set up a 3rd-party repository with the binaries I produce, can you point me in the direction of the script/code I could use to achieve the same for apk?

Thanks!

### OpenWrt version

snapshots

### OpenWrt release

snapshots

### OpenWrt target/subtarget

x86_64

### Device

Dell Edge 620

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

aparcar, 2024-11-13:
1. Curious issue, I haven't witnessed this yet. I'm guessing it's a problem how SDK combine RELEASE/VERSION, it's obvious that the `r` is missing right? Do you mind having a look?
2. Yes historically the architecture is missing from the APK packages. Also the packages use dashes instead of underscores, not sure why they decided on that. Somewhere is already an issue to have a more flexible naming design, however what's special is that APK doesn't store filenames in the index but concatenates metadata to figure out the filename. Such custom schema could be done on a per index basis, but then again is it really worth it? Do you face any troubles at this point not having the architecture in the name?
3. I'd suggest to use `make package/index V=s`, if you have to do it outside of build root or SDK, use `apk mkndx`. There is no script needed, APK itself does the magic. After als it's a binary file and not just a bunch of strings written, so a simple shell script doesn't cut it anymore

ptpt52, 2024-11-14:
Also report some apk install issue:

```
IPKG_INSTROOT=/mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/target-dir-7872e783 /mnt/Data/Sources/openwrt/x-wrt/staging_dir/host/bin/fakeroot /mnt/Data/Sources/openwrt/x-wrt/staging_dir/host/bin/apk --root /mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/target-dir-7872e783 --keys-dir /mnt/Data/Sources/openwrt/x-wrt --no-logfile --preserve-env --no-scripts --repositories-file /dev/zero --repository file:///mnt/Data/Sources/openwrt/x-wrt/staging_dir/packages/mediatek/packages.adb add kmod-mt7915e apk-openssl dawn gre iptables-mod-dhcpmac iptables-mod-dnetmap jool-tools-netfilter kmod-bonding kmod-crypto-acompress kmod-crypto-cbc kmod-crypto-deflate kmod-crypto-echainiv kmod-inet-diag kmod-ip6-vti kmod-ipsec kmod-ipsec4 kmod-ipsec6 kmod-ipt-compat-xtables kmod-ipt-dhcpmac kmod-ipt-dnetmap kmod-ip-vti kmod-jool-netfilter kmod-l2tp kmod-lib-zlib-deflate kmod-lib-zlib-inflate kmod-mt7981-firmware kmod-nat46 kmod-nf-conntrack-netlink kmod-pppol2tp kmod-xfrm-interface libcap-ng libgcrypt libgpg-error liblz4-1 liblzo2 libnl200 libnl-cli200 libnl-core200 libnl-genl200 libnl-nf200 libnl-route200 libopenssl3 libopenssl-conf libopenssl-legacy libpcre2 libqrencode libxml2 luci-app-dawn luci-app-openvpn luci-app-usteer luci-i18n-dashboard-zh-cn luci-i18n-openvpn-zh-cn luci-lib-uqr luci-mod-dashboard luci-nginx luci-proto-bonding luci-proto-gre luci-proto-relay luci-proto-vti luci-proto-wireguard luci-proto-xfrm mt7981-wo-firmware natflow-hostacl nginx-mod-luci nginx-mod-ubus nginx-ssl nginx-ssl-util openvpn-openssl ppp-mod-pppol2tp proto-bonding qrencode relayd tc-tiny urllogger uwsgi uwsgi-cgi-plugin uwsgi-luci-support uwsgi-syslog-plugin vti wireguard-tools wpad-openssl xfrm xl2tpd
ERROR: failed to read repositories: /dev/zero: No buffer space available
ERROR: unable to select packages:
  6in4-29:
    error: uninstallable
    arch: all
    satisfies: world[6in4]
  6rd-13:
    error: uninstallable
    arch: all
    satisfies: world[6rd]
  6to4-13:
    error: uninstallable
    arch: all
    satisfies: world[6to4]
  ddns-scripts-services-2.8.2-r52:
    error: uninstallable
    arch: all
    satisfies: world[ddns-scripts-services] ddns-scripts-2.8.2-r52[ddns-scripts-services]
  ddns-scripts-2.8.2-r52:
    error: uninstallable
    arch: all
    satisfies: world[ddns-scripts] ddns-scripts_aliyun-2.8.2-r52[ddns-scripts] luci-app-ddns-24.315.53681~2da07ab[ddns-scripts]
  ddns-scripts_aliyun-2.8.2-r52:
    error: uninstallable
    arch: all
    satisfies: world[ddns-scripts_aliyun]
  ds-lite-9:
    error: uninstallable
    arch: all
    satisfies: world[ds-lite]
  wifi-scripts-1.0-r1:
    error: uninstallable
    arch: all
    satisfies: world[wifi-scripts] kmod-cfg80211-6.6.60.6.11.2-r1[wifi-scripts]
  wireless-regdb-2024.10.07-r1:
    error: uninstallable
    arch: all
    satisfies: world[wireless-regdb] kmod-cfg80211-6.6.60.6.11.2-r1[wireless-regdb]
  lua-ipops-0.0.1-r1:
    error: uninstallable
    arch: all
    satisfies: world[lua-ipops] natflow-auth-20241020[lua-ipops]
  luci-app-autoreboot-1.0.1-r1:
    error: uninstallable
    arch: all
    satisfies: world[luci-app-autoreboot]
  luci-lib-base-24.079.46064~3e3005c:
    error: uninstallable
    arch: all
    satisfies: world[luci-lib-base] luci-lua-runtime-24.315.53681~7b8a56f[luci-lib-base]
  luci-app-ddns-24.315.53681~2da07ab:
    error: uninstallable
    arch: all
    satisfies: world[luci-app-ddns] luci-i18n-ddns-zh-cn-24.311.29558~b49fdd1[luci-app-ddns]
  luci-app-fakemesh-1.0.2-r1:
    error: uninstallable
    arch: all
    satisfies: world[luci-app-fakemesh]
  luci-app-firewall-24.297.73189~1300761:
    error: uninstallable
    arch: all
    satisfies: world[luci-app-firewall] luci-i18n-firewall-zh-cn-24.315.31321~c16643f[luci-app-firewall] luci-light-23.024.33244~34dee82[luci-app-firewall]
               luci-nginx-24.298.47280~0535c07[luci-app-firewall]
  luci-app-ledtrig-switch-24.315.53681~2c75bed:
    error: uninstallable
    arch: all
[... 93 lines cut ...]
    satisfies: world[luci-theme-bootstrap] luci-light-23.024.33244~34dee82[luci-theme-bootstrap] luci-nginx-24.298.47280~0535c07[luci-theme-bootstrap]
  luci-light-23.024.33244~34dee82:
    error: uninstallable
    arch: all
    satisfies: world[luci-light] luci-ssl-23.035.26083~7550ad6[luci-light]
  luci-ssl-23.035.26083~7550ad6:
    error: uninstallable
    arch: all
    satisfies: world[luci-ssl]
  luci-theme-openwrt-2020-24.315.53681~8d48db9:
    error: uninstallable
    arch: all
    satisfies: world[luci-theme-openwrt-2020]
/mnt/Data/Sources/openwrt/x-wrt/include/image.mk:373: recipe for target 'target-dir-7872e783' failed
make[5]: *** [target-dir-7872e783] Error 53
make[5]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt/target/linux/mediatek/image'
Makefile:17: recipe for target 'install' failed
make[4]: *** [install] Error 2
make[4]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt/target/linux/mediatek'
Makefile:11: recipe for target 'install' failed
make[3]: *** [install] Error 2
make[3]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt/target/linux'
time: target/linux/install#38.99#5.81#36.07
    ERROR: target/linux failed to build.
```

aparcar, 2024-11-14:
I think it has to be noarch and not "all"

hnyman, 2024-11-14:
```
ERROR: unable to select packages:
  6in4-29:
    error: uninstallable
    arch: all
```
> I think it has to be noarch and not "all"

alternative to "noarch"(? ) might be to add "all" to the allowed architectures in `/etc/apk/arc`h file.
See that mentioned in forum:
https://forum.openwrt.org/t/the-future-is-now-opkg-vs-apk/201164/78

> I checked the /etc/apk/arch file, and it was set to x86_64, which is not the architecture of the my BPI R4. I changed the contents to
> 
> aarch64_cortex-a53
> all
> 
> and now it works as expected. The all line is essential, since without it, some packages, like wireless-regdb will not be satisfied

GlebChili, 2024-11-14:
Here is the issue for the forum discussion mentioned by @hnyman: https://github.com/openwrt/openwrt/issues/16953

vincejv, 2024-11-16:
may i suggest that, just like with iptables to nftables transition/migration we maintain a list of packages that needs to be updated, just like this issue: https://github.com/openwrt/packages/issues/16818 since they somehow depend on the old opkg functionality?

For example:
- https://github.com/openwrt/packages/issues/25378

ynezz, 2024-11-16:
Please, don't mix multiple issues in one, always create a separate issue for every problem you face. Otherwise its very hard to keep track of anything that way, you can't use `Fixes: XXXX` tag to close such multi issues etc. Not to mention, that nobody wants to go through wall of texts just to get to the root of the problem.

That being said, please split this up into separate issues, ideally prefixed with `apk:` if you think, that its related to the opkg->apk transition, thanks!

stangri, 2024-11-22:
As per suggestion from @ynezz I'm closing this in favour of individual issues.