openwrt/openwrt #16950: apk: “apk mkpkg” with double quotes in the “--info” argument

nxhack, 2024-11-14:
### Describe the bug

The error occurs if there are double quotes in the arguments when generating the apk package.

Example: my custom package description has double quotes and I had a problem when building this package.

https://github.com/nxhack/openwrt-node-packages/blob/master/node-homebridge-config-ui-x/Makefile#L34-L61

```
/mnt/node/openwrt/staging_dir/host/bin/fakeroot /mnt/node/openwrt/staging_dir/host/bin/apk mkpkg --info "name:node-homebridge-config-ui-x" --info "version:4.62.0-r1" --info "description:Homebridge Config UI X is a web based management tool for Homebridge that allows you to manage all aspects of your Homebridge setup. Run as root: hb-service install --user homebridge -U /usr/share/homebridge hb-service restart *** The hb-service command is not full compatible with the openwrt environment. *** *** You can edit config.json using the Homebridge Config UI X. *** See: https://github.com/oznu/homebridge-config-ui-x/wiki/Manual-Configuration https://github.com/oznu/homebridge-config-ui-x/wiki/Manual-Configuration#logs-from-custom-command https://github.com/oznu/homebridge-config-ui-x/wiki/Manual-Configuration#sudo-mode "platforms": [ { "platform": "config", "name": "Config", "port": 8581, "restart": "sudo /etc/init.d/homebridge restart", "sudo": true, "log": { "method": "custom", "command": "sudo logread -f -l 100 -e homebridge" } } ]" --info "arch:aarch64_generic" --info "license:MIT" --info "origin:feeds/node/node-homebridge-config-ui-x" --info "url:https://www.npmjs.org/package/homebridge-config-ui-x" --info "maintainer:Hirokazu MORIKAWA <morikw2@gmail.com>" --info "provides:"  --script "post-install:/mnt/node/openwrt/build_dir/target-aarch64_generic_musl/node-homebridge-config-ui-x-4.62.0/apk-aarch64_generic/node-homebridge-config-ui-x/post-install" --script "pre-deinstall:/mnt/node/openwrt/build_dir/target-aarch64_generic_musl/node-homebridge-config-ui-x-4.62.0/apk-aarch64_generic/node-homebridge-config-ui-x/pre-deinstall" --info "depends:libc node131 node-homebridge sudo node-homebridge-node-pty-prebuilt-multiarch node-bufferutil node-utf-8-validate" --files "/mnt/node/openwrt/build_dir/target-aarch64_generic_musl/node-homebridge-config-ui-x-4.62.0/ipkg-aarch64_generic/node-homebridge-config-ui-x" --output "/mnt/node/openwrt/bin/packages/aarch64_generic/node/node-homebridge-config-ui-x-4.62.0-r1.apk" --sign "/mnt/node/openwrt/private-key.pem"
/mnt/node/openwrt/staging_dir/host/bin/apk: invalid option -- 'l'
apk-tools 3.0.0_pre20241113, compiled for x86_64.

ERROR: This apk-tools has been built without help
```

opkg has no problem.
```
(cd /mnt/node/openwrt/build_dir/target-x86_64_musl/node-homebridge-config-ui-x-4.62.0/ipkg-x86_64/node-homebridge-config-ui-x/CONTROL; ( echo "$CONTROL"; printf "Description: "; echo "$DESCRIPTION" | sed -e 's,^[[:space:]]*, ,g'; ) > control; chmod 644 control; ( echo "#!/bin/sh"; echo "[ \"\${IPKG_NO_SCRIPT}\" = \"1\" ] && exit 0"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_postinst \$0 \$@"; ) > postinst; ( echo "#!/bin/sh"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_prerm \$0 \$@"; ) > prerm; chmod 0755 postinst prerm; echo "$V_Package_node_homebridge_config_ui_x_postrm" > postrm; chmod 0755 postrm; )
```

### OpenWrt version

r28076-27c2c140b1

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

GL.iNet GL-MT6000

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

geldot, 2024-11-14:
Proposed an approach to fix apk description field escaping here: https://github.com/openwrt/openwrt/pull/16908#issuecomment-2466843625

Feel free to tidy up, test and submit it.

nxhack, 2024-11-15:
I applied this patch based on @geldot information.
```diff
diff --git a/include/package-pack.mk b/include/package-pack.mk
index 9d06c8b647..5b1dc22bc6 100644
--- a/include/package-pack.mk
+++ b/include/package-pack.mk
@@ -45,6 +45,7 @@ dep_if=$(if $(findstring !,$(1)),$(call dep_neg,$(1)),$(call dep_pos,$(1)))
 dep_val=$(word 2,$(call dep_split,$(1)))
 strip_deps=$(strip $(subst +,,$(filter-out @%,$(1))))
 filter_deps=$(foreach dep,$(call strip_deps,$(1)),$(if $(findstring :,$(dep)),$(call dep_if,$(dep)),$(dep)))
+sanitize_field=$(subst ",\",$(subst `,\`,$(strip $(1))))
 
 define AddDependency
   $$(if $(1),$$(if $(2),$$(foreach pkg,$(1),$$(PACK_$$(pkg))): $$(foreach pkg,$(2),$$(PACK_$$(pkg)))))
@@ -339,7 +340,7 @@ else
        $(FAKEROOT) $(STAGING_DIR_HOST)/bin/apk mkpkg \
          --info "name:$(1)$$(ABIV_$(1))" \
          --info "version:$(VERSION)" \
-         --info "description:$$(strip $$(Package/$(1)/description))" \
+         --info "description:$(call sanitize_field,$(Package/$(1)/description))" \
          --info "arch:$(PKGARCH)" \
          --info "license:$(LICENSE)" \
          --info "origin:$(SOURCE)" \
```

I had assumed that the escape characters would still be there in the apk info because they were simply escaped, but they had disappeared.

```
root@OpenWrt:~# apk info -d node-homebridge-config-ui-x
node-homebridge-config-ui-x-4.62.0-r1 description:
Homebridge Config UI X is a web based management tool for Homebridge that allows you to manage all aspects of your Homebridge setup. Run as root: hb-service install --user homebridge -U /usr/share/homebridge hb-service restart TEST STRINGS: ` -A ` AAA *** The hb-service command is not full compatible with the openwrt environment. *** *** You can edit config.json using the Homebridge Config UI X. *** See: https://github.com/oznu/homebridge-config-ui-x/wiki/Manual-Configuration https://github.com/oznu/homebridge-config-ui-x/wiki/Manual-Configuration#logs-from-custom-command https://github.com/oznu/homebridge-config-ui-x/wiki/Manual-Configuration#sudo-mode "platforms": [ { "platform": "config", "name": "Config", "port": 8581, "restart": "sudo /etc/init.d/homebridge restart", "sudo": true, "log": { "method": "custom", "command": "sudo logread -f -l 100 -e homebridge" } } ]

```

Ansuel, 2024-11-17:
Pushed a fix for this 43455f1075855d9ed2760f70b2d8acbe8e8bb16c
@geldot used a different version more similar to version.mk with a dedicated function.

closed 2024-11-17 by Ansuel