openwrt/openwrt #16953: apk: Wrong CPU architecture

GlebChili, 2024-11-14:
### Describe the bug

# TLDR
The `/etc/apk/arch` file may contain the wrong architecture, like `x86_64` on Mediatek Filogic platform, making apk unusable out of the box.

# A longer version
I have built a last SNAPSHOT version (see below) using Image Builder for my Banana Pi BPI R4 with some additional packages included. Unexpectedly, I was not able to install any additional packages on running system due to the following errors (list is not full):
```
root@OpenWrt-bpi-r4:/# apk add golang
ERROR: unable to select packages:
  libgcc1-13.3.0-r4:
    error: uninstallable
    arch: aarch64_cortex-a53
    satisfies: world[libgcc] libc-1.2.5-r4[libgcc] libpthread-1.2.5-r4[libgcc1]
  libc-1.2.5-r4:
    error: uninstallable
    arch: aarch64_cortex-a53
    satisfies: world[libc] apk-mbedtls-3.0.0_pre20241113-r1[libc]
               base-files-1633~b902284290[libc] busybox-1.36.1-r2[libc]
               ca-bundle-20240203-r1[libc] cgi-io-2022.08.10~901b0f04-r21[libc]
               dnsmasq-2.90-r3[libc] dropbear-2024.85-r1[libc]
               e2fsprogs-1.47.0-r2[libc] eip197-mini-firmware-20241017-r1[libc]
               f2fsck-1.16.0-r3[libc] firewall4-2024.05.21~4c01d1eb-r1[libc]
               fitblk-2[libc] fstools-2024.07.14~408c2cc4-r1[libc]
               fwtool-2019.11.12~8f7fe925-r1[libc]
               getrandom-2024.04.26~85f10530-r1[libc] golang-1.23.2-r1[libc]
               golang-src-1.23.2-r1[libc]
               hostapd-common-2024.09.15~5ace39b0-r1[libc] iw-6.9-r1[libc]
               iwinfo-2024.10.20~b94f066e-r1[libc] jansson4-2.14-r3[libc]
               jshn-2024.03.29~eb9bcb64-r1[libc]
               jsonfilter-2024.01.23~594cfa86-r1[libc]
               kernel-6.6.60~0561031e21f60f05c8f540bffb723036-r1[libc]
               libblkid1-2.40.2-r1[libc]
               libblobmsg-json20240329-2024.03.29~eb9bcb64-r1[libc]
               libcomerr0-1.47.0-r2[libc] libe2p2-1.47.0-r2[libc]
               libext2fs2-1.47.0-r2[libc] libf2fs6-1.16.0-r3[libc]
               libiwinfo-data-2024.10.20~b94f066e-r1[libc]
               libiwinfo20230701-2024.10.20~b94f066e-r1[libc]
               libjson-c5-0.18-r1[libc]
               libjson-script20240329-2024.03.29~eb9bcb64-r1[libc]
               liblucihttp-ucode-2023.03.15~9b5b683f-r1[libc]
               liblucihttp0-2023.03.15~9b5b683f-r1[libc]
               libmbedtls21-3.6.2-r1[libc] libmnl0-1.0.5-r1[libc]
               libnftnl11-1.2.8-r1[libc]
               libnl-tiny1-2023.12.05~965c4bf4-r1[libc] libss2-1.47.0-r2[libc]
               libubox20240329-2024.03.29~eb9bcb64-r1[libc]
               libubus20241020-2024.10.20~252a9b0c-r1[libc]
               libuci20130104-2023.08.10~5781664d-r1[libc]
               libuclient20201210-2024.10.22~88ae8f20-r1[libc]
               libucode20230711-2024.07.22~b610860d-r3[libc]
               libudebug-2023.12.06~6d3f51f9[libc]
               libustream-mbedtls20201210-2024.07.28~99bd3d2b-r1[libc]
               libuuid1-2.40.2-r1[libc] logd-2024.04.26~85f10530-r1[libc]
               luci-24.315.52043~d650688[libc]
               luci-app-firewall-24.315.52043~d650688[libc]
               luci-app-package-manager-24.315.52043~d650688[libc]
               luci-base-24.315.52043~d650688[libc]
               luci-light-24.315.52043~d650688[libc]
               luci-mod-admin-full-24.315.52043~d650688[libc]
               luci-mod-network-24.315.52043~d650688[libc]
               luci-mod-status-24.315.52043~d650688[libc]
               luci-mod-system-24.315.52043~d650688[libc]
               luci-proto-ipv6-24.315.52043~d650688[libc]
               luci-proto-ppp-24.315.52043~d650688[libc]
               luci-theme-bootstrap-24.315.52043~d650688[libc]
               mkf2fs-1.16.0-r3[libc] mt7988-wo-firmware-20241017-r1[libc]
               mtd-26[libc] netifd-2024.11.08~34eb11eb-r1[libc]
               nftables-json-1.1.1-r1[libc]
               odhcp6c-2023.05.12~bcd28363-r20[libc]
               odhcpd-ipv6only-2024.05.08~a2988231-r1[libc]
               openwrt-keyring-2024.11.01~fbae29d7-r2[libc] ppp-2.5.1-r1[libc]
               ppp-mod-pppoe-2.5.1-r1[libc] procd-2024.11.06~109fa41b-r1[libc]
               procd-seccomp-2024.11.06~109fa41b-r1[libc]
               procd-ujail-2024.11.06~109fa41b-r1[libc]
               rpcd-2024.09.17~9f4b86e7-r1[libc]
               rpcd-mod-file-2024.09.17~9f4b86e7-r1[libc]
               rpcd-mod-iwinfo-2024.09.17~9f4b86e7-r1[libc]
               rpcd-mod-luci-20240305-r1[libc] rpcd-mod-rrdns-20170710[libc]
               rpcd-mod-ucode-2024.09.17~9f4b86e7-r1[libc]
               ubi-utils-2.2.1-r1[libc] uboot-envtools-2024.07-r1[libc]
               ubox-2024.04.26~85f10530-r1[libc]
               ubus-2024.10.20~252a9b0c-r1[libc]
               ubusd-2024.10.20~252a9b0c-r1[libc]
               uci-2023.08.10~5781664d-r1[libc]
               uclient-fetch-2024.10.22~88ae8f20-r1[libc]
               ucode-2024.07.22~b610860d-r3[libc]
               ucode-mod-fs-2024.07.22~b610860d-r3[libc] ucode-mod-html-1[libc]
               ucode-mod-math-2024.07.22~b610860d-r3[libc]
               ucode-mod-nl80211-2024.07.22~b610860d-r3[libc]
               ucode-mod-rtnl-2024.07.22~b610860d-r3[libc]
               ucode-mod-ubus-2024.07.22~b610860d-r3[libc]
               ucode-mod-uci-2024.07.22~b610860d-r3[libc]
               ucode-mod-uloop-2024.07.22~b610860d-r3[libc]
               uhttpd-2023.06.25~34a8a74d-r4[libc]
               uhttpd-mod-ubus-2023.06.25~34a8a74d-r4[libc]
               urandom-seed-3[libc] urngd-2023.11.01~44365eb1-r1[libc]
               usign-2020.05.23~f1f65026-r1[libc] wifi-scripts-1.0-r1[libc]
               wireless-regdb-2024.10.07-r1[libc]
               wpad-basic-mbedtls-2024.09.15~5ace39b0-r1[libc]
               zlib-1.3.1-r1[libc]
  libmbedtls21-3.6.2-r1:
    error: uninstallable
    arch: aarch64_cortex-a53
    satisfies: apk-mbedtls-3.0.0_pre20241113-r1[libmbedtls21]
               libustream-mbedtls20201210-2024.07.28~99bd3d2b-r1[libmbedtls21]
               wpad-basic-mbedtls-2024.09.15~5ace39b0-r1[libmbedtls21]
  zlib-1.3.1-r1:
    error: uninstallable
    arch: aarch64_cortex-a53
    satisfies: apk-mbedtls-3.0.0_pre20241113-r1[zlib]
  apk-mbedtls-3.0.0_pre20241113-r1:
    error: uninstallable
    arch: aarch64_cortex-a53
    satisfies: world[apk-mbedtls]
  busybox-1.36.1-r2:
    error: uninstallable
    arch: aarch64_cortex-a53
    satisfies: base-files-1633~b902284290[busybox]
...
```
After the investigation I've figured out, that something is wrong with `/etc/apk/arch` file: it was set to the `x86_64` instead of BPI R4's native `aarch64_cortex-a53`. After manually setting the contents of the file to
```
aarch64_cortex-a53
all
```
(the values are `newline` separated) everything was fixed. 

The `all` line is essential, since without it, some packages, like wireless-regdb will not be satisfied:
```
wireless-regdb-2024.10.07-r1:
    error: uninstallable
    arch: all
    satisfies: kmod-cfg80211-6.6.60.6.11.2-r1[wireless-regdb]
```

The issue was also discussed on the forum: https://forum.openwrt.org/t/the-future-is-now-opkg-vs-apk/201164/77

The somewhat related issue: #16946 (see comment https://github.com/openwrt/openwrt/issues/16946#issuecomment-2475089849)

### OpenWrt version

r28072-b902284290

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R4

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

KanjiMonster, 2024-11-14:
This seems to be a generic IB issue, at least bmips_bcm6368 is also affected.

EnzoAlbornoz, 2024-11-14:
Images from the firmware selector are affected, too. I was wondering why I could not install `nano` on a new install 😅

efahl, 2024-11-14:
> Images from the firmware selector are affected, too.

Yes, all of FS, owut, auc and LuCI ASU app use imagebuilders, so the bug percolates up to all of them.

hnyman, 2024-11-17:
In addition to imagebuilder, the issue will come up with normal building soon when we upgrade apk for a newer version which has arch validation. (we already had that for one day, but it was reveretd)

We have lots of script-only packages (most LuCI packages, database packages like 'ca-bundle' , script packages like 6in4 & 6rd4) that currently have `PKGARCH:=all` 

Apk sets "noarch" as the always compatible arch, while in OpenWrt it is "all".
Looking at the apk source code, I think that the easiest way for us is to patch apk to also handle "all" like "noarch".

Based on https://gitlab.alpinelinux.org/alpine/apk-tools/-/commit/c1a3e69f24e235fc78f4dddb1f85fdf281464446 , the required change is likely just 3 lines: one new variable, setting that variable to be an "all" arch item, and adding an 'or' test for 'all' to the noarch check line. 

I will test that later.

KanjiMonster, 2024-11-18:
> In addition to imagebuilder, the issue will come up with normal building soon when we upgrade apk for a newer version which has arch validation. (we already had that for one day, but it was [...]

We could also let `include/package-pack.mk` translate `all` to `noarch` when generating apk packages, then we don't need to patch apk.

robimarko, 2024-11-19:
Hm, call me crazy but I cannot force it to fail installing something with `PKGARCH:=all`, I have updated APK to `e3464d096708b270138653ef93db59134bb24813` but it still happily installed `kitty-terminfo` (Which has `PKGARCH:=all`)

ynezz, 2024-11-19:
> Hm, call me crazy but I cannot force it to fail installing something with `PKGARCH:=all`, I have updated APK to `e3464d096708b270138653ef93db59134bb24813` but it still happily installed `kitty-terminfo` (Which has `PKGARCH:=all`)

IMO to reproduce the issue, you need to build the image using ImageBuilder, `apk` probably adds the `x86_64` (from the host) into `/etc/apk/arch` when its building/committing the package index in one of those steps? https://github.com/openwrt/openwrt/blob/8ec35a2a1533ccff8a4b0e6f4f0a0954d1bee6a6/target/imagebuilder/files/Makefile#L203
https://github.com/openwrt/openwrt/blob/8ec35a2a1533ccff8a4b0e6f4f0a0954d1bee6a6/target/imagebuilder/files/Makefile#L203

robimarko, 2024-11-19:
You were right, the `apk add --initdb` is the issue but we can override the arch during its creation and that fixes this issue.
I am just adding the commit to fix it.

commit 1c308bad (2024-11-19): imagebuilder: fix APK architecture setting

Currently, when "apk add --initdb" is called it will populate the
/etc/apk/arch file with the architecture of the host instead of the target
package architecture and this will then make it impossible for new packages
to be installed on the device as the target architecture will not match
after APK is updated to include [1].

So, lets use override the architecture by passing the target package one
by using the "--arch" argument to "apk add".

[1] https://gitlab.alpinelinux.org/alpine/apk-tools/-/commit/c1a3e69f24e235fc78f4dddb1f85fdf281464446

Fixes:  [... cut]
PR #17015 (2024-11-19): APK: architecture related fixes
First commit fixes the architecture setting when ImageBuilder is used to generate the image as currently, if IB is used it will be set to the architecture of the host used to generate the image and not the target device.

Second commit translates the "all" architecture into "noarch" which APK supports to indicate a package without any compilable sources.

And third one updates APK so we can drop the backport as issues that previously were exposed by APK update are now fixed by this PR.

Fixes: #16953
closed 2024-11-19 by openwrt-bot