openwrt/openwrt #16954: apk: Unable to build x86 image after switching to APK `ca-bundle-20240203-r1` uninstallable

vincejv, 2024-11-14:
### Describe the bug

Log
```
2024-11-14T15:04:24.0022050Z mkdir -p /home/runner/work/openwrt/openwrt/bin/targets/x86/64 /home/runner/work/openwrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/tmp
2024-11-14T15:04:24.0050551Z rm -rf /home/runner/work/openwrt/openwrt/build_dir/target-x86_64_musl/json_info_files
2024-11-14T15:04:24.0076450Z tar -cp --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name --mtime="@1731581768" -C /home/runner/work/openwrt/openwrt/build_dir/target-x86_64_musl/root-x86/ . | gzip -9n > /home/runner/work/openwrt/openwrt/bin/targets/x86/64/openwrt-snapshot-x86-64-rootfs.tar.gz
2024-11-14T15:04:32.9662703Z /home/runner/work/openwrt/openwrt/staging_dir/host/bin/make_ext4fs -L rootfs -l 2147483648 -b 4096 -m 0  -T 1731581768 /home/runner/work/openwrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/root.ext4 /home/runner/work/openwrt/openwrt/build_dir/target-x86_64_musl/root-x86/
2024-11-14T15:04:33.0635032Z Creating filesystem with parameters:
2024-11-14T15:04:33.0635718Z     Size: 2147483648
2024-11-14T15:04:33.0636130Z     Block size: 4096
2024-11-14T15:04:33.0636456Z     Blocks per group: 32768
2024-11-14T15:04:33.0636851Z     Inodes per group: 8192
2024-11-14T15:04:33.0637371Z     Inode size: 256
2024-11-14T15:04:33.0637858Z     Journal blocks: 8192
2024-11-14T15:04:33.0638370Z     Label: rootfs
2024-11-14T15:04:33.0638828Z     Blocks: 524288
2024-11-14T15:04:33.0639262Z     Block groups: 16
2024-11-14T15:04:33.0639994Z     Reserved blocks: 0
2024-11-14T15:04:33.0640491Z     Reserved block group size: 127
2024-11-14T15:04:33.0641278Z Created filesystem with 1618/131072 inodes and 24654/524288 blocks
2024-11-14T15:04:33.0676258Z rm -f /home/runner/work/openwrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/generic-kernel.bin
2024-11-14T15:04:33.0703606Z cp /home/runner/work/openwrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/bzImage /home/runner/work/openwrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/generic-kernel.bin
2024-11-14T15:04:33.0878614Z cp /home/runner/work/openwrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/generic-kernel.bin /home/runner/work/openwrt/openwrt/bin/targets/x86/64/openwrt-snapshot-x86-64-generic-kernel.bin
2024-11-14T15:04:33.1016950Z rm -rf /home/runner/work/openwrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/target-dir-461aeb09
2024-11-14T15:04:33.1422223Z cp -fpR /home/runner/work/openwrt/openwrt/build_dir/target-x86_64_musl/root.orig-x86 /home/runner/work/openwrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/target-dir-461aeb09
2024-11-14T15:04:33.2468086Z mv /home/runner/work/openwrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/target-dir-461aeb09/etc/apk/repositories /home/runner/work/openwrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/target-dir-461aeb09.repositories
2024-11-14T15:04:33.2505511Z IPKG_INSTROOT=/home/runner/work/openwrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/target-dir-461aeb09 /home/runner/work/openwrt/openwrt/staging_dir/host/bin/fakeroot /home/runner/work/openwrt/openwrt/staging_dir/host/bin/apk --root /home/runner/work/openwrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/target-dir-461aeb09 --keys-dir /home/runner/work/openwrt/openwrt --no-logfile --preserve-env --no-scripts --repositories-file /dev/zero --repository file:///home/runner/work/openwrt/openwrt/staging_dir/packages/x86/packages.adb add kmod-amazon-ena kmod-amd-xgbe kmod-bnx2 kmod-dwmac-intel kmod-e1000e kmod-e1000 kmod-forcedeth kmod-fs-vfat kmod-igb kmod-igc kmod-ixgbe kmod-r8169 kmod-tg3
2024-11-14T15:04:33.2702130Z ERROR: failed to read repositories: /dev/zero: No buffer space available
2024-11-14T15:04:33.2709214Z ERROR: unable to select packages:
2024-11-14T15:04:33.2710005Z   ca-bundle-20240203-r1:
2024-11-14T15:04:33.2710462Z     error: uninstallable
2024-11-14T15:04:33.2710871Z     arch: all
2024-11-14T15:04:33.2711297Z     satisfies: world[ca-bundle]
2024-11-14T15:04:33.2725069Z make[5]: *** [/home/runner/work/openwrt/openwrt/include/image.mk:377: target-dir-461aeb09] Error 1
2024-11-14T15:04:33.2727451Z make[5]: Leaving directory '/home/runner/work/openwrt/openwrt/target/linux/x86/image'
2024-11-14T15:04:33.2728484Z make[4]: *** [Makefile:24: install] Error 2
2024-11-14T15:04:33.2731361Z make[3]: *** [Makefile:11: install] Error 2
2024-11-14T15:04:33.2732453Z make[4]: Leaving directory '/home/runner/work/openwrt/openwrt/target/linux/x86'
2024-11-14T15:04:33.2733538Z make[3]: Leaving directory '/home/runner/work/openwrt/openwrt/target/linux'
2024-11-14T15:04:33.2735180Z time: target/linux/install#21.76#8.01#30.17
2024-11-14T15:04:33.2742833Z     ERROR: target/linux failed to build.
2024-11-14T15:04:33.2746531Z make[2]: *** [target/Makefile:32: target/linux/install] Error 1
2024-11-14T15:04:33.2748668Z make[2]: Leaving directory '/home/runner/work/openwrt/openwrt'
2024-11-14T15:04:33.2756969Z make[1]: *** [target/Makefile:26: /home/runner/work/openwrt/openwrt/staging_dir/target-x86_64_musl/stamp/.target_install] Error 2
2024-11-14T15:04:33.2758030Z make[1]: Leaving directory '/home/runner/work/openwrt/openwrt'
2024-11-14T15:04:33.2762805Z make: *** [/home/runner/work/openwrt/openwrt/include/toplevel.mk:248: world] Error 2
2024-11-14T15:04:33.2773701Z ##[error]Process completed with exit code 2.
2024-11-14T15:04:33.2868222Z Post job cleanup.
2024-11-14T15:04:33.3624039Z [command]/usr/bin/git version
2024-11-14T15:04:33.3659350Z git version 2.47.0
```

### OpenWrt version

r28072-b902284290

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

x86/64

### Device

Dell Inc. OptiPlex 790

### Image kind
[... 743 lines cut ...]
CONFIG_VERSION_HOME_URL=""
CONFIG_VERSION_HWREV=""
CONFIG_VERSION_MANUFACTURER=""
CONFIG_VERSION_MANUFACTURER_URL=""
CONFIG_VERSION_NUMBER=""
CONFIG_VERSION_PRODUCT=""
CONFIG_VERSION_REPO="https://openwrt.vincejv.com/releases/testing"
CONFIG_VERSION_SUPPORT_URL=""
# CONFIG_WPA_WOLFSSL is not set
# CONFIG_COLLECT_KERNEL_DEBUG is not set
# CONFIG_IB is not set
# CONFIG_KERNEL_DEBUG_INFO is not set
# CONFIG_MAKE_TOOLCHAIN is not set
# CONFIG_PACKAGE_bridger is not set
# CONFIG_PACKAGE_libtraceevent is not set
# CONFIG_PACKAGE_perf is not set
# CONFIG_REPRODUCIBLE_DEBUG_INFO is not set
# CONFIG_SDK is not set
# CONFIG_SDK_LLVM_BPF is not set
# CONFIG_VERSION_CODE_FILENAMES is not set
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2024-11-14:
Already fixed: https://github.com/openwrt/openwrt/commit/dab5ca818ef46b32a9c120b5136fd5a1a34a88a5

vincejv, 2024-11-14:
Thanks might have missed this commit

closed 2024-11-14 by vincejv