openwrt/openwrt #16957: Not working usb port on SNR-CPE-ME2-SFP

bropines, 2024-11-14:
### Describe the bug

Apparently, the flash drive is not powered. If I connect it via a USB HUB with external power, then it seems to be detected for a couple of seconds. I tried to turn on the GPIO power, but there was no result. And libgpio complained that something was not working. There was an issue somewhere that it was broken.

### OpenWrt version

r24106-10cc5fcd00

### OpenWrt release

23.05.5

### OpenWrt target/subtarget

ramips/mt7621

### Device

SNR-CPE-ME2-SFP

### Image kind

Official downloaded image

### Steps to reproduce

1. Plug in the USB flash drive.
2. The flash drive does not even start to glow (LED indicator)

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2024-11-14:
The commit https://github.com/openwrt/openwrt/commit/6ff598306fa2964fab5def3ca2fc572a5d801c64 should also be applied to the 23.05 branch.

bropines, 2024-11-15:
> Please check if the problem also occurs for snapshot build,

I tried it a month ago on a snapshot assembly. The problem remains. I can try again now if it's very important

namiltd, 2024-11-15:
There were many changes in the code during the month.

bropines, 2024-11-15:
I put a snapshot, as a result, everything that could have flown was lost, LUCI was deleted and completely disappeared from the OPKG system. Also, the flash drive never received power. I'm rolling back because I spent 2 hours trying to install opkg

namiltd, 2024-11-15:
In the snapshot, the opkg package manager has been replaced with apk.

bropines, 2024-11-15:
I returned to the snapshot. Here's the log

```bash
gpiochip0: GPIOs 512-543, parent: platform/1e000600.gpio, 1e000600.gpio-bank0:
 gpio-521 (                    |enable_sfp          ) out hi ACTIVE LOW
 gpio-525 (                    |green:usb           ) out hi ACTIVE LOW
 gpio-526 (                    |reset               ) in  hi IRQ ACTIVE LOW
 gpio-527 (                    |green:status        ) out lo ACTIVE LOW
 gpio-528 (                    |green:status_1      ) out hi ACTIVE LOW
 gpio-529 (                    |regulator-usb       ) out hi
 gpio-531 (                    |reset               ) out hi ACTIVE LOW

gpiochip1: GPIOs 544-575, parent: platform/1e000600.gpio, 1e000600.gpio-bank1:

gpiochip2: GPIOs 576-607, parent: platform/1e000600.gpio, 1e000600.gpio-bank2:
root@OpenWrt:~#
```

bropines, 2024-11-15:
And also... There is documentation for this `APK`

namiltd, 2024-11-15:
Try:
```
echo "0" > /sys/class/gpio/gpio529/value
cat /sys/kernel/debug/gpio
```
and test USB

bropines, 2024-11-15:
```bash
root@OpenWrt:~# echo "0" > /sys/class/gpio/gpio529/value
s/kernel/debug/gpio-ash: can't create /sys/class/gpio/gpio529/value: nonexistent directory
```

I already received this

namiltd, 2024-11-15:
echo "0" > /sys/class/gpio/gpiochip529/value

bropines, 2024-11-15:
```bash
root@OpenWrt:~# echo "0" > /sys/class/gpio/gpiochip529/value
-ash: can't create /sys/class/gpio/gpiochip529/value: nonexistent directory
root@OpenWrt:~#
```

namiltd, 2024-11-15:
cat  /sys/class/gpio/

bropines, 2024-11-15:
```bash
root@OpenWrt:~# ls /sys/class/gpio/
export       gpiochip512  gpiochip544  gpiochip576  unexport
```

namiltd, 2024-11-15:
lsmod | grep gpio

bropines, 2024-11-15:
```bash
root@OpenWrt:/sys/class/gpio# lsmod | grep gpio
gpio_button_hotplug    12288  0
leds_gpio              12288  0
root@OpenWrt:/sys/class/gpio#
```

namiltd, 2024-11-15:
lsmod

namiltd, 2024-11-15:
The problem is that you can control the GPIO port manually when the port is not in use.
So you should edit mt7621_snr_snr-cpe-me2-sfp.dts and disable USB power control or reverse polarity. I can compile a version with changed polarity for you to test.

bropines, 2024-11-15:
> . I can compile a version with changed polarity for you to test.

I'd be glad.

namiltd, 2024-11-15:
https://filetransfer.io/data-package/aEM5UaTC#link

bropines, 2024-11-15:
> Try this: [filetransfer.io/data-package/aEM5UaTC#link](https://filetransfer.io/data-package/aEM5UaTC#link)
> 
> There is also a possibility that the power supply is there but it is too weak for the connected device.

I'm a little dumb, but what can I do about it?

namiltd, 2024-11-15:
Exactly what you did: external power supply.

bropines, 2024-11-15:
> Exactly what you did: external power supply.

The funny thing is that on the factory firmware there was enough power for the hard drive connected to the router. The method I used to connect external power, alas, is not yet available (I broke the hub)

bropines, 2024-11-15:
> If the same disk worked on the factory firmware, it is clearly a problem with the OpenWrt firmware.

Well, that's what I came here for)

namiltd, 2024-11-15:
So first test the firmware I've attached. If it doesn't work, you'll need to find the right GPIO port that will turn on the USB power.

bropines, 2024-11-15:
> So first test the firmware I've attached. If it doesn't work, you'll need to find the right GPIO port that will turn on the USB power.

Here. Should I just install the image compiled there? 
[image]

namiltd, 2024-11-15:
Yes, exactly this file.

bropines, 2024-11-15:
I'm install sysupgrade. Is there anything additional that needs to be installed?

neheb, 2024-11-15:
@namiltd GPIO is not controllable in latest snapshot

@bropines you need to try 24.10. No release is available so you need to compile.

namiltd, 2024-11-15:
https://mirror-03.infra.openwrt.org/releases/24.10-SNAPSHOT/targets/ramips/mt7621/

bropines, 2024-11-15:
> [mirror-03.infra.openwrt.org/releases/24.10-SNAPSHOT/targets/ramips/mt7621](https://mirror-03.infra.openwrt.org/releases/24.10-SNAPSHOT/targets/ramips/mt7621/)

Installed

bropines, 2024-11-15:
What next, where to point, turned off the flash drive and plugged it in, no effect

namiltd, 2024-11-15:
What about my image? https://filetransfer.io/data-package/aEM5UaTC#link

bropines, 2024-11-15:
> What about my image? [filetransfer.io/data-package/aEM5UaTC#link](https://filetransfer.io/data-package/aEM5UaTC#link)

Likewise. I connected the flash drive, it didn’t automatically supply power, and I didn’t really figure out how to manually.

neheb, 2024-11-15:
OK bad news. It seems my conversion to regulator had nothing to do with it. This is a regulator since inception looks like.

I find it strange that there's a 3.3v regulator with no gpio. Probably to get rid of dmesg warning.

The guy that added support for this device claims it works: https://github.com/openwrt/openwrt/pull/12036#discussion_r1118038446

ping @alealexpro100

namiltd, 2024-11-16:
Try this: https://github.com/openwrt/openwrt/pull/12036#discussion_r1118038446

bropines, 2024-11-16:
> Try this: [#12036 (comment)](https://github.com/openwrt/openwrt/pull/12036#discussion_r1118038446)

rebuild again?

bropines, 2024-11-16:
So, well, it didn’t work out to dispense food using the method described there....
```bash
root@OpenWrt:~# echo '1-1' | tee /sys/bus/usb/drivers/usb/bind
1-1
tee: /sys/bus/usb/drivers/usb/bind: I/O error
```
```bash
root@OpenWrt:~# ls /sys/bus/usb/devices/
1-0:1.0  2-0:1.0  usb1     usb2
```

bropines, 2024-11-18:
> Just checked it with latest snapshot. USB drive connects and works without any additional actions. Did USB work on original firmware?

Yep. 
I plan to go back to stock and double-check again. because I think I'm already a schizoid and made up information out of nowhere

neheb, 2024-11-18:
If it works with a cold boot and fails when restarting the device, sounds like reset pins are missing.

bropines, 2024-11-18:
> If it works with a cold boot and fails when restarting the device, sounds like reset pins are missing.

I returned to the stock, configured everything, even updated the firmware to las, zero effect. I disassembled the cover, there is zero effect on the USB indicator when connected. I'll go to the service tomorrow and find out what the problem is.

csharper2005, 2024-12-08:
@neheb do you know how to control USB power using regulator from user space? I can't do it. https://github.com/openwrt/openwrt/blob/9c55866fada91c45679f75d03e6d13e1e0cd4202/target/linux/ramips/dts/mt7621_snr_snr-cpe-me1.dts#L65

neheb, 2024-12-08:
Disable the USB port with uhubctl

csharper2005, 2024-12-08:
> Disable the USB port with uhubctl

It works. Thanks.
```
[50338.416876] usb 1-1: USB disconnect, device number 4
[50376.506292] usb 1-1: new full-speed USB device number 5 using xhci-mtk
[50443.193026] usb 1-1: USB disconnect, device number 5
[50447.215242] usb 1-1: new full-speed USB device number 6 using xhci-mtk
```