openwrt/openwrt #16966: build: inconsistent image names - x86, armsr, loongarch

efahl, 2024-11-15:
### Describe the bug

Background: https://github.com/efahl/owut/issues/21

In attempting to determine the appropriate image name for a device in `owut`, I'm checking existence of `/sys/firmware/efi`, which if it exists indicates an EFI boot device.  For both x86 and loongarch targets, this is consistently satisfied by an image containing either just `combined` on non-EFI devices, or  `combined-efi` on EFI devices, but on the `armsr` target which is always EFI, the `-efi` part is missing in the image names.

Here are the respective Makefile lines that are inconsistent:
```
target/linux/armsr/image/Makefile:        IMAGE/combined.img := grub-config efi | combined efi | grub-install efi | append-metadata
target/linux/loongarch64/image/Makefile:  IMAGE/combined-efi.img := grub-config efi | combined | append-metadata
target/linux/x86/image/Makefile:          IMAGE/combined-efi.img := grub-config efi | combined efi | grub-install efi | append-metadata
```

@mcbridematt  @ynezz   If I submit a PR adding `-efi` to the armsr image names, would that stand a chance of being accepted?

### OpenWrt version

r27695-dae6a871de

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

armsr/armv8

### Device

generic

### Image kind

Official downloaded image

### Steps to reproduce

Run `owut` on an armsr device, crashes out saying "combined" isn't "combined-efi".

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

mcbridematt, 2024-11-15:
I actually submitted a change to `combined-efi` a while ago at the request of the attended sysupgrade maintainers, however, they decided to change the handling of x86 instead (IIRC loongarch64 was not in the tree at the time)

https://github.com/openwrt/openwrt/pull/12963

efahl, 2024-11-15:
My github-searchfoo is lacking, thanks for pointing it out!  It's exactly what is needed, now to get it merged.

efahl, 2024-11-16:
And now that I'm digging through all the image names, I also find that `ipq806x/generic` sticks a superfluous `nand-` prefix on the `UbiFit` image types, while `qualcommax/ipq807x` leaves that out of the `UbiFit` images...