openwrt/openwrt #16968: strongswan: `strongswan-mod-kernel-netlink` (no such package) on SNAPSHOT r28085-6720c4ccba (after apk transition)

vincejv, 2024-11-16:
Maintainer: @pprindeville @Thermi 
Environment: x86_64 SNAPSHOT r28085-6720c4ccba

Description: Checked https://downloads.openwrt.org/snapshots/faillogs/x86_64/ and nothing there that says build is failing for this package (or did i miss something?????)

```
$ apk add --simulate strongswan-full
ERROR: unable to select packages:
  strongswan-mod-kernel-netlink (no such package):
    required by: strongswan-full-5.9.14-r4[strongswan-mod-kernel-netlink]
```

robimarko, 2024-11-23:
This should be fixed by: https://github.com/openwrt/openwrt/commit/2784f78c6c22c0731ef551e4fcd42295f08a07a5

robimarko, 2024-11-23:
Yeah, apk search now sees it, so closing as completed.

closed 2024-11-23 by robimarko