openwrt/openwrt #16969: On  Openwrt SNAPSHOT (r28085-6720c4ccba) opkg is missing

5gfreak, 2024-11-16:
### Describe the bug

On  Openwrt SNAPSHOT (r28085-6720c4ccba) opkg is missing
no packages can be installed, and opkg update ends with opkg not found

### OpenWrt version

 r28085-6720c4ccba

### OpenWrt release

Snapshot

### OpenWrt target/subtarget

 qualcommax/ipq807x

### Device

Zbtlink ZBT-Z800AX

### Image kind

Official downloaded image

### Steps to reproduce

opkg is missing from the system

### Actual behaviour

system reports opkg not found

### Expected behaviour

in order to install packages manually opkg is required

### Additional info

no additional info

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2024-11-16:
Its not a bug, package manager [was switched](https://lists.openwrt.org/pipermail/openwrt-devel/2024-November/043378.html) from opkg to apk https://forum.openwrt.org/t/the-future-is-now-opkg-vs-apk/201164

5gfreak, 2024-11-16:
root@OpenWrt:~# apk update
-ash: apk: not found

it works neither
is there any simple installation instructions?

5gfreak, 2024-11-16:
and it is a valid release from https://firmware-selector.openwrt.org/?version=SNAPSHOT&target=qualcommax%2Fipq807x&id=zbtlink_zbt-z800ax

5gfreak, 2024-11-16:
I don't find any apk file on this image, or any package with this name
the only folder is on root etc apk but how can this be started?

namiltd, 2024-11-16:
Did you download the images or compile them yourself? If you compiled them yourself, include the configuratin,

5gfreak, 2024-11-16:
I have downloaded it from https://firmware-selector.openwrt.org/?version=SNAPSHOT&target=qualcommax%2Fipq807x&id=zbtlink_zbt-z800ax

robimarko, 2024-11-16:
This looks like another ASU issue since using the same image builder version locally includes APK just fine in the sysupgrade image:
```
[robimarko@fedora squashfs-root]$ find . -name "apk"
./etc/apk
./lib/apk
./run/apk
./usr/bin/apk
```

vincejv, 2024-11-16:
is this really an ASU/Firmware selector issue? Seems like `make defconfig` is not installing `apk-ssl` or `apk-mbedtls` by default hence the missing apk binaries and library..

make defconfig generates the following for me, idk why, maybe as a work around, you can add either apk-mbedtls or openssl packages in asu, but not sure/how if that would work.
```
#
# Base system
#
CONFIG_PACKAGE_apk-mbedtls=m
CONFIG_PACKAGE_apk-openssl=m
CONFIG_PACKAGE_base-files=y
CONFIG_PACKAGE_block-mount=m
CONFIG_PACKAGE_blockd=m
```

robimarko, 2024-11-16:
No, the root cause is OpenWrt itself which recently stopped including some packages including procd and apk/opkg in the `DEFAULT_PACKAGES` variable thus the `default_packages` list in `profiles.json` lacks APK in ASU.

For more details:
https://github.com/openwrt/asu/issues/1084

PR #16986 (2024-11-17): Fix firmware images generated by ASU
### base-files,imagebuilder: unify handling of apk and opkg

Remove the remaining special handling of `apk` and `opkg` in a same way
as the rest of the packages was handled in the commit 4c65359af49b
("build: fix including busybox, procd and apk/opkg in imagebuilder") and
commit 8a38814b246a ("target,base-files: unify handling of
procd-ujail").

Fixes: #16969
Fixes: openwrt/asu/issues/1084
Fixes: 44598c233dd9 ("build: remove broken dependency of metadata on toplevel .config variables")

### target,base-files: unify handling of procd-ujail

Remove the remain [... cut]
commit 90f0be85 (2024-11-17): linux: fix missing default packages in profiles.json

Robert reported, that in firmware images generated by ASU, there is
`apk` package manager missing after the commit 44598c233dd9 ("build:
remove broken dependency of metadata on toplevel .config variables").

That is happening, because apk got removed from `default_packages` list in
`profiles.json`, which is being generated by `json_overview_image_info` Make
target, which uses `scripts/json_overview_image_info.py` helper script,
which gets the information from `DEFAULT_PACKAGES` Make variable.

So lets fix it by providing `DEFAULT_PACKAGES`  [... cut]
commit 19db0920 (2025-01-03): linux: fix missing default packages in profiles.json

Robert reported, that in firmware images generated by ASU, there is
`apk` package manager missing after the commit 44598c233dd9 ("build:
remove broken dependency of metadata on toplevel .config variables").

That is happening, because apk got removed from `default_packages` list in
`profiles.json`, which is being generated by `json_overview_image_info` Make
target, which uses `scripts/json_overview_image_info.py` helper script,
which gets the information from `DEFAULT_PACKAGES` Make variable.

So lets fix it by providing `DEFAULT_PACKAGES`  [... cut]
closed 2024-11-17 by openwrt-bot