openwrt/openwrt #16972: [23.05] mediatek: Xiaomi AX3000t: nmbm doesn't work with Winbond SPI NAND

csharper2005, 2024-11-16:
### Describe the bug

nmbm doesn't work with Winbond SPI NAND.
```
[    0.751645] spi spi0.0: setup: ignoring unsupported mode bits a00
[    0.758136] spi-nand spi0.0: Winbond SPI NAND was found.
[    0.763449] spi-nand spi0.0: 128 MiB, block size: 128 KiB, page size: 2048, OOB size: 96
[    0.772150] Signature found at block 1023 [0x07fe0000]
[    0.777310] ni->signature.spare_size: 64 [0x00800000]
[    0.782348] bmtd.mtd->oobsize: 96 [0x00c00000]
[    0.786783] NMBM configuration mismatch
[    0.790876] 10 fixed-partitions partitions found on MTD device spi0.0
[    0.797361] Creating 10 MTD partitions on "spi0.0":
```

### OpenWrt version

r24133-33b45c0a0e

### OpenWrt release

23.05-SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

Xiaomi Mi Router AX3000T

### Image kind

Self-built image

### Steps to reproduce

Install OpenWrt `23.05-SNAPSHOT` or `SNAPSHOT` Xiaomi Mi Router AX3000T on Winbond SPI NAND.

### Actual behaviour

nmbm doesn't work with Winbond SPI NAND.

### Expected behaviour

nmbm works (after applying the workaround):
```
[    0.756896] spi-nand spi0.0: Winbond SPI NAND was found.
[    0.762209] spi-nand spi0.0: 128 MiB, block size: 128 KiB, page size: 2048, OOB size: 96
[    0.770915] Signature found at block 1023 [0x07fe0000]
[    0.776069] ni->signature.spare_size: 64 [0x00800000]
[    0.781108] bmtd.mtd->oobsize: 96 [0x00c00000]
[    0.785545] NMBM management region starts at block 960 [0x07800000]
[    0.793557] First info table with writecount 1 found in block 960
[    0.804841] Second info table with writecount 1 found in block 963
[    0.811018] NMBM has been successfully attached
[    0.815888] 10 fixed-partitions partitions found on MTD device spi0.0
[    0.822331] Creating 10 MTD partitions on "spi0.0":
```

### Additional info

Workarond
-------------
https://github.com/openwrt/openwrt/blob/main/target/linux/generic/files/drivers/mtd/nand/mtk_bmt_nmbm.c#L2203
Change:
`	    ni->signature.spare_size != bmtd.mtd->oobsize) {`
to:
`	    ni->signature.spare_size > bmtd.mtd->oobsize) {`

Xiaomi stock 1.0.84 bootlog:
```
[    0.077171] spi-nand spi0.0: Winbond SPI NAND was found.
[    0.077185] spi-nand spi0.0: 128 MiB, block size: 128 KiB, page size: 2048, OOB size: 64
....
[    0.106918] nmbm nmbm_spim_nand: Signature found at block 1023 [0x07fe0000]
[    0.107614] nmbm nmbm_spim_nand: First info table with writecount 0 found in block 960
[    0.109644] nmbm nmbm_spim_nand: Second info table with writecount 0 found in block 963
[    0.109657] nmbm nmbm_spim_nand: NMBM has been successfully attached 
[    0.109847] 12 fixed-partitions partitions found on MTD device nmbm_spim_nand
[    0.109854] Creating 12 MTD partitions on "nmbm_spim_nand":
```

The start of the discussion:
https://github.com/openwrt/openwrt/pull/16669#issuecomment-2420220484

### Diffconfig

```text
Lost. Tested device is with unsupported Airoha AN8855 switch. That's the only reason using self-built image.
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

csharper2005, 2024-11-16:
> Invalid Version reported.
> Invalid Release reported.
> Invalid Target/Subtarget.

😱 No, all is valid 😉

nstepanets, 2024-12-07:
> No, all is valid

It looks like the bot can't recognize 23.05-SNAPSHOT.

I think you should use the version that is officially available for download. For example a snapshot from the main branch.

csharper2005, 2024-12-07:
> I think you should use the version that is officially available for download. For example a snapshot from the main branch.

Tested device is with unsupported Airoha AN8855 switch. So I can't use official build.

csharper2005, 2025-01-08:
24.10.0-rc5 is affected too.

csharper2005, 2025-01-08:
@nbd168, do you have any ideas on how we should handle situations when u-boot use only a part of available nand oobsize (64 of 96)? 

My idea. Add new dts property:
`mediatek,bmt-spare-size = 64`
If `mediatek,bmt-spare-size` exists then we use this value instead of `bmtd.mtd->oobsize`.

What do you think?

cc @robimarko

csharper2005, 2025-01-09:
Since there are no additional ideas, I opened a PR with the current version of the fix.

closed 2025-02-07 by openwrt-bot