openwrt/openwrt #16976: apk: libudev dependency for usbutils not automatically installed

totkeks, 2024-11-16:
### Describe the bug

When using the imagebuilder to create the image for my device, it fails on installing the `usbutils` package due to this error:

```
Installing packages...
ERROR: unable to select packages:
  libudev (virtual):
    note: please select one of the 'provided by'
          packages explicitly
    provided by: libudev-zero
    required by: usbutils-017-r1[libudev]
make[2]: *** [Makefile:229: package_install] Error 2
make[1]: *** [Makefile:160: _call_image] Error 2
make: *** [Makefile:332: image] Error 2
```

When I add the `libudev-zero` dependency manually to the `PACKAGES` list for the `make` command, the imagebuilder finishes.

It seems counterintuitive, that `libudev-zero` is not automatically selected and installed as dependency, when it is the only provider available. As it has been with opkg before.

After discussing the issue on IRC it seems that this behavior is not intended, hence the bug report.

### OpenWrt version

snapshot

### OpenWrt release

r28091

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R4

### Image kind

Official downloaded image

### Steps to reproduce

Download and extract [imagebuilder](https://downloads.openwrt.org/snapshots/targets/mediatek/filogic/openwrt-imagebuilder-mediatek-filogic.Linux-x86_64.tar.zst)

Run build with `make image PROFILE=bananapi_bpi-r4 PACKAGES="usbutils"`

### Actual behaviour

Build fails due to error.

### Expected behaviour

Build completes successfully.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

totkeks, 2024-11-16:
> Invalid Release reported. `r28091` Is this from a clean repository?

Thanks bot, I edited the information.

vincejv, 2024-11-16:
@totkeks try running this instead, seems like the libudev dependency is marked as a "meta package"... this is not isolated to `usbutils` only, `luci-app-vnstat2` too
```
apk add libudev-zero usbutils
```

```
apk add -s luci-app-vnstat2
ERROR: unable to select packages:
  libjpeg (virtual):
    note: please select one of the 'provided by' packages explicitly
    provided by: libjpeg-turbo
    required by: libgd-2.3.3-r1[libjpeg]
```
in old opkg manager, the first alternative (of that meta package) is automatically selected, but with apk, you have to explicitly choose which alternative package to install, even if there is only one alternative, is this more of an upstream issue? or just an apk configuration issue?

jszinger, 2024-11-17:
Another affected package is luci-app-sqm.

I expect, in general, for the package manager to use 'provides' to resolve 'requires' automatically without intervention.  Other package managers (opkg, dnf, apt, macports) do it this way.

Is this the expected behavior from apk?

robimarko, 2024-11-18:
It seems that it does not like the fact that there is no `provider-priority` in the generated package meta-data, only those that have `DEFAULT_VARIANT` have it populated.

This is kind of stupid in cases where only one package provides a certain thing and OPKG used to resolve it just fine

closed 2024-11-19 by openwrt-bot

totkeks, 2024-11-22:
Can confirm, this is working now with the latest snapshot (r28144-2784f78c6c)

commit dea83977 (2024-11-19): include/package-pack: add default 'provider_priority' for APK packages

Currently, trying to use APK to install a package that has a dependency on
a virtual package that needs to be resolved via 'provides' it will fail if
package does not have 'DEFAULT_VARIANT:=1' like:
apk add usbutils
ERROR: unable to select packages:
  libudev (virtual):
    note: please select one of the 'provided by' packages explicitly
    provided by: libudev-zero
    required by: usbutils-017-r1[libudev]

Issue is that we dont set 'provider_priority' that APK uses to break ties
when choosing a virtual package to satisf [... cut]
PR #17008 (2024-11-19): include/package-pack: add default 'provider_priority' for APK packages
Currently, trying to use APK to install a package that has a dependency on a virtual package that needs to be resolved via 'provides' it will fail if package does not have 'DEFAULT_VARIANT:=1' like:
```
apk add usbutils
ERROR: unable to select packages:
  libudev (virtual):
    note: please select one of the 'provided by' packages explicitly
    provided by: libudev-zero
    required by: usbutils-017-r1[libudev]
```

Issue is that we dont set 'provider_priority' that APK uses to break ties when choosing a virtual package to [... cut]