openwrt/openwrt #16983: AppleTalk DDP multicast broken - traffic not forwarded to socket

APCCV, 2024-11-17:
### Describe the bug

Netatalk v4 restores AppleTalk functionality. `atalkd` operates without issues on 23.05.

On Snapshot or 24.10, DDP (AppleTalk layer 3) traffic sent to OpenWrt can be seen on `br-lan` with `tcpdump`. However, it is not received by the `atalkd`.  It would seem that the mutlicast is never sent to the socket opened by the daemon.

Exact same code and setup works OK in 23.05.

Issue tested both on ipq608x/generic (C2600) and ath79/generic (DIR 835).

### OpenWrt version

r28105-afffcd09e5

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ipq806x/generic

### Device

TP-Link Archer C2600

### Image kind

Self-built image

### Steps to reproduce

1. Refer to PR https://github.com/openwrt/openwrt/pull/16979 in order to build the `appletalk` module.
2. Build image.
3. Install image / kernel module / netatalk v4 & tcdump package on two devices. `modprobe appletalk`. Connect both devices to the same Ethernet segment. IP setup is not relevant to AppleTalk operation.
4. `ssh` to each device and execute `tcpdump -e -i br-lan | grep Appletalk`
5. `ssh` to a separate session and execute `atalkd -d` on each.

### Actual behaviour

Traffic from *both* devices appears on `tcpdump`. Like this.
```
23:14:20.760699 <MAC-DEV1> (oui Unknown) > 09:00:07:ff:ff:ff (oui Unknown), 802.3, length 36: LLC, dsap SNAP (0xaa) Individual, ssap SNAP (0xaa) Command, ctrl 0x03: oui Ethernet (0x000000), ethertype Appletalk ARP (0x80f3), length 28: aarp probe 65280.124 tell 65280.124
23:14:21.799737 <MAC-DEV2> (oui Unknown) > 09:00:07:ff:ff:ff (oui Unknown), 802.3, length 43: LLC, dsap SNAP (0xaa) Individual, ssap SNAP (0xaa) Command, ctrl 0x03: oui Appletalk (0x080007), pid Appletalk (0x809b), length 35: 6400.148.rtmp > 0.rtmp:  at-rtmp 22
```
Each device's `atalkd` however only displays own packages. As in:
```
root@TestWrt-65:~# atalkd -d
lo -phase 2 -net 0 -addr 0.0 -caddr 0.0
br-lan -phase 2 -net 65280-65534 -addr 0.0 -caddr 65280.0
packet from 65280.124 on br-lan (12) 6 (9)
06 05 00 00 00 00 00 00                         	........        
(end)
packet from 65280.124 on br-lan (12) 6 (9)
06 05 00 00 00 00 00 00                         	........        
(end)
packet from 65280.124 on br-lan (12) 6 (9)
06 05 00 00 00 00 00 00                         	........        
(end)
```
In this example `65280.124` is an autoconfigured address for the device.
After three probes, both `atalkd` daemons stop posting packages to `stdout` because they cannot hear each other and go dormant. No packages but from `65280.124` (in this example) are received by this `atalkd`.

`tcpdump` will continue to show traffic, but it does not seem to reach the daemons.

### Expected behaviour

AppleTalk is chatty. There will be activity at least every 5 seconds as a beacon is sent across.
Once the two devices negotiate who the router is, `atalkd -d` in each device will continue to display traffic from both devices.
As:
```
....
packet from 6400.147 on br0 (4fa) 1 (16)
01 19 00 08 93 19 00 80 19 00 82 03 E8 80 03 E8 	................
82                                              	.               
(end)
packet from 1000.148 on br0 (4fa) 1 (16)
01 03 E8 08 94 03 E8 80 03 E8 82 19 00 80 19 00 	................
82                                              	.               
(end)
....
```
In that example 6400.147 is one device, 1000.148 is the other. This volley of packets continues indefinitely.
`tcpdump` shows traffic accordingly, including the occasional burst of Appletalk ARP packages that are not displayed by `atalkd -d`. 

### Additional info

Exact same code and setup works on devices running 23.05.
Same behavior with ipx806x and ath79. I'd assume this affects all platforms.
DSA related? Non-IP multicast related? I know close to nothing on how the kernel or DSA could impact this.

A working third device added to the Ethernet segment (OpenWrt 23.05 or a Linux workstation) will see the traffic coming out of the Snapshot/24.10 devices, both at Ethernet level with `tcpdump` and daemon level on `stdout` of `atalkd -d`. In all cases the daemons can send traffic out, and `tcpdump` shows it properly formed. `tcpdump` shows it arriving at all devices at Ethernet level and showing up on the interface (bridge or other, on Linux this was on a wireless adapter). Somehow the multicast is not being forwarded to the daemon open socket on 24/10 or Snapshot.

### Diffconfig

```text
CONFIG_TARGET_ipq806x=y
CONFIG_TARGET_ipq806x_generic=y
CONFIG_TARGET_ipq806x_generic_DEVICE_tplink_c2600=y
CONFIG_ALL_KMODS=y
CONFIG_ALL_NONSHARED=y
CONFIG_DEVEL=y
CONFIG_AUTOREMOVE=y
CONFIG_BPF_TOOLCHAIN_BUILD_LLVM=y
# CONFIG_BPF_TOOLCHAIN_NONE is not set
CONFIG_BUILDBOT=y
CONFIG_COLLECT_KERNEL_DEBUG=y
CONFIG_HAS_BPF_TOOLCHAIN=y
CONFIG_IB=y
CONFIG_JSON_CYCLONEDX_SBOM=y
CONFIG_KERNEL_BUILD_DOMAIN="buildhost"
CONFIG_KERNEL_BUILD_USER="builder"
CONFIG_MAKE_TOOLCHAIN=y
CONFIG_PACKAGE_ath10k-board-qca9887=m
CONFIG_PACKAGE_ath10k-board-qca988x=m
CONFIG_PACKAGE_ath10k-board-qca9984=m
CONFIG_PACKAGE_ath10k-board-qca99x0=m
CONFIG_PACKAGE_ath10k-firmware-qca9887-ct=m
CONFIG_PACKAGE_ath10k-firmware-qca988x-ct=m
CONFIG_PACKAGE_ath10k-firmware-qca9984-ct=m
CONFIG_PACKAGE_ath10k-firmware-qca99x0-ct=m
CONFIG_PACKAGE_cgi-io=m
CONFIG_PACKAGE_e2fsprogs=m
CONFIG_PACKAGE_kmod-lib-zlib-deflate=m
CONFIG_PACKAGE_kmod-lib-zlib-inflate=m
CONFIG_PACKAGE_kmod-pstore=m
CONFIG_PACKAGE_kmod-ramoops=m
CONFIG_PACKAGE_kmod-reed-solomon=m
CONFIG_PACKAGE_libblkid=m
CONFIG_PACKAGE_libbpf=m
CONFIG_PACKAGE_libcomerr=m
CONFIG_PACKAGE_libdb47=m
CONFIG_PACKAGE_libe2p=m
CONFIG_PACKAGE_libelf=m
CONFIG_PACKAGE_libevent2=m
CONFIG_PACKAGE_libext2fs=m
CONFIG_PACKAGE_libgcrypt=m
CONFIG_PACKAGE_libgpg-error=m
CONFIG_PACKAGE_liblucihttp=m
CONFIG_PACKAGE_liblucihttp-ucode=m
CONFIG_PACKAGE_libpcap=m
CONFIG_PACKAGE_libsmartcols=m
CONFIG_PACKAGE_libss=m
CONFIG_PACKAGE_libuuid=m
CONFIG_PACKAGE_libxdp=m
CONFIG_PACKAGE_losetup=m
CONFIG_PACKAGE_luci-app-firewall=m
CONFIG_PACKAGE_luci-base=m
CONFIG_PACKAGE_luci-light=m
CONFIG_PACKAGE_luci-mod-admin-full=m
CONFIG_PACKAGE_luci-mod-network=m
CONFIG_PACKAGE_luci-mod-status=m
CONFIG_PACKAGE_luci-mod-system=m
CONFIG_PACKAGE_luci-proto-ipv6=m
CONFIG_PACKAGE_luci-proto-ppp=m
CONFIG_PACKAGE_luci-ssl=m
CONFIG_PACKAGE_luci-theme-bootstrap=m
CONFIG_PACKAGE_netatalk=m
CONFIG_PACKAGE_px5g-mbedtls=m
CONFIG_PACKAGE_qosify=m
CONFIG_PACKAGE_rpcd=m
CONFIG_PACKAGE_rpcd-mod-file=m
CONFIG_PACKAGE_rpcd-mod-iwinfo=m
CONFIG_PACKAGE_rpcd-mod-luci=m
CONFIG_PACKAGE_rpcd-mod-rrdns=m
CONFIG_PACKAGE_rpcd-mod-ucode=m
CONFIG_PACKAGE_tc-tiny=m
CONFIG_PACKAGE_ucode-mod-html=m
CONFIG_PACKAGE_ucode-mod-math=m
CONFIG_PACKAGE_uhttpd=m
CONFIG_PACKAGE_uhttpd-mod-ubus=m
CONFIG_PACKAGE_xdp-filter=m
CONFIG_PACKAGE_xdp-loader=m
CONFIG_PACKAGE_xdpdump=m
CONFIG_REPRODUCIBLE_DEBUG_INFO=y
CONFIG_SDK=y
CONFIG_SDK_LLVM_BPF=y
CONFIG_USE_LLVM_BUILD=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

APCCV, 2024-11-17:
For the ath79 device:
Version: r28105-afffcd09e5
```
CONFIG_TARGET_ath79=y
CONFIG_TARGET_ath79_generic=y
CONFIG_TARGET_ath79_generic_DEVICE_dlink_dir-835-a1=y
CONFIG_ALL_KMODS=y
CONFIG_ALL_NONSHARED=y
CONFIG_DEVEL=y
CONFIG_AUTOREMOVE=y
CONFIG_BPF_TOOLCHAIN_BUILD_LLVM=y
# CONFIG_BPF_TOOLCHAIN_NONE is not set
CONFIG_BUILDBOT=y
CONFIG_COLLECT_KERNEL_DEBUG=y
CONFIG_HAS_BPF_TOOLCHAIN=y
CONFIG_IB=y
CONFIG_JSON_CYCLONEDX_SBOM=y
CONFIG_KERNEL_BUILD_DOMAIN="buildhost"
CONFIG_KERNEL_BUILD_USER="builder"
CONFIG_MAKE_TOOLCHAIN=y
CONFIG_OPENSSL_ENGINE=y
CONFIG_OPENSSL_PREFER_CHACHA_OVER_GCM=y
CONFIG_OPENSSL_WITH_ASM=y
CONFIG_OPENSSL_WITH_CHACHA_POLY1305=y
CONFIG_OPENSSL_WITH_CMS=y
CONFIG_OPENSSL_WITH_DEPRECATED=y
CONFIG_OPENSSL_WITH_ERROR_MESSAGES=y
CONFIG_OPENSSL_WITH_IDEA=y
CONFIG_OPENSSL_WITH_MDC2=y
CONFIG_OPENSSL_WITH_PSK=y
CONFIG_OPENSSL_WITH_SEED=y
CONFIG_OPENSSL_WITH_SRP=y
CONFIG_OPENSSL_WITH_TLS13=y
CONFIG_OPENSSL_WITH_WHIRLPOOL=y
CONFIG_PACKAGE_adb=m
CONFIG_PACKAGE_adb-enablemodem=m
CONFIG_PACKAGE_ath10k-board-qca9887=m
CONFIG_PACKAGE_ath10k-board-qca9888=m
CONFIG_PACKAGE_ath10k-board-qca988x=m
CONFIG_PACKAGE_ath10k-board-qca9984=m
CONFIG_PACKAGE_ath10k-board-qca99x0=m
CONFIG_PACKAGE_ath10k-firmware-qca9887-ct=m
CONFIG_PACKAGE_ath10k-firmware-qca9887-ct-full-htt=m
CONFIG_PACKAGE_ath10k-firmware-qca9888-ct=m
CONFIG_PACKAGE_ath10k-firmware-qca988x-ct=m
CONFIG_PACKAGE_ath10k-firmware-qca9984-ct=m
CONFIG_PACKAGE_ath10k-firmware-qca99x0-ct=m
CONFIG_PACKAGE_e2fsprogs=m
CONFIG_PACKAGE_fconfig=m
CONFIG_PACKAGE_fritz-tffs=m
CONFIG_PACKAGE_kmod-nls-base=m
CONFIG_PACKAGE_kmod-owl-loader=m
CONFIG_PACKAGE_kmod-phy-ath79-usb=m
CONFIG_PACKAGE_kmod-rtc-ds1307=m
CONFIG_PACKAGE_kmod-rtc-ds1374=m
CONFIG_PACKAGE_kmod-rtc-pcf8563=m
CONFIG_PACKAGE_kmod-usb-core=m
CONFIG_PACKAGE_kmod-usb-ehci=m
CONFIG_PACKAGE_kmod-usb2=m
CONFIG_PACKAGE_libblkid=m
CONFIG_PACKAGE_libbpf=m
CONFIG_PACKAGE_libcomerr=m
CONFIG_PACKAGE_libdb47=m
CONFIG_PACKAGE_libe2p=m
CONFIG_PACKAGE_libelf=m
CONFIG_PACKAGE_libevent2=m
CONFIG_PACKAGE_libext2fs=m
CONFIG_PACKAGE_libgcrypt=m
CONFIG_PACKAGE_libgpg-error=m
CONFIG_PACKAGE_libopenssl=m
CONFIG_PACKAGE_libpcap=m
CONFIG_PACKAGE_libss=m
CONFIG_PACKAGE_libuuid=m
CONFIG_PACKAGE_libxdp=m
CONFIG_PACKAGE_netatalk=m
CONFIG_PACKAGE_qosify=m
CONFIG_PACKAGE_rssileds=m
CONFIG_PACKAGE_tc-tiny=m
CONFIG_PACKAGE_tcpdump=m
CONFIG_PACKAGE_tcpdump-mini=m
CONFIG_PACKAGE_uencrypt-mbedtls=m
CONFIG_PACKAGE_xdp-filter=m
CONFIG_PACKAGE_xdp-loader=m
CONFIG_PACKAGE_xdpdump=m
CONFIG_REPRODUCIBLE_DEBUG_INFO=y
CONFIG_SDK=y
CONFIG_SDK_LLVM_BPF=y
CONFIG_USE_LLVM_BUILD=y
```

APCCV, 2024-11-17:
Issue tested with two ipq806x devices or one ipq806x and one ath79 device.
I don't think this is platform specific.

Neustradamus, 2024-11-21:
@APCCV: Good luck to solve it...

APCCV, 2024-11-23:
Built ipq608x off commit `3665440` for kernel 6.1. Issue is present. Whatever caused this started with a commit older than this.
```
root@TestWrt-63:~# uname -a
Linux TestWrt-63 6.1.86 #0 SMP Fri Apr 26 10:24:46 2024 armv7l GNU/Linux
```
Next: build at commit when device was migrated to DSA.

Neustradamus, 2024-11-24:
@APCCV: 2 new versions after Netatalk 4.0.4 -> 4.0.5 and 4.0.6 :)

Links:
- https://github.com/Netatalk/netatalk/releases/tag/netatalk-4-0-4
- https://github.com/Netatalk/netatalk/releases/tag/netatalk-4-0-5
- https://github.com/Netatalk/netatalk/releases/tag/netatalk-4-0-6

APCCV, 2024-11-25:
Issue started happening between commits 548c437f7ddfabc54ce8857a847be77c6235c2d0 (Jul 21, 2023) and 337e36e0ef988d850105cfa5f7588e36d90aef2b (Sep 30, 2023, conversion to DSA).
Non-DSA build at commit 548c437f7ddfabc54ce8857a847be77c6235c2d0 works.

```
root@OpenWrt:~# ./atalkd -d
lo -phase 2 -net 0 -addr 0.0 -caddr 0.0
br-lan -phase 2 -net 65280-65534 -addr 0.0 -caddr 65280.0
packet from 65280.30 on br-lan (12) 6 (9)
06 05 00 00 00 00 00 00                         	........        
(end)
packet from 6400.148 on br-lan (12) 6 (9)
06 06 A0 19 00 19 00 00 06 09 00 07 00 00 3D 0A 	..............=.
4F 70 65 6E 57 72 74 2D 36 34                   	OpenWrt-64      
(end)
packet from 6400.148 on br-lan (b2) 1 (12)
01 19 00 08 94 19 00 80 19 00 82 19 00 80 19 00 	................
82                                              	.               
(end)
packet from 6400.148 on br-lan (b2) 6 (9)
06 02 01 19 00 0A 4F 70 65 6E 57 72 74 2D 36 34 	......OpenWrt-64
(end)
packet from 6400.148 on br-lan (b2) 1 (12)
01 19 00 08 94 19 00 80 19 00 82 19 00 80 19 00 	................
82                                              	.               
(end)
packet from 6400.148 on br-lan (b2) 1 (12)
01 19 00 08 94 19 00 80 19 00 82 19 00 80 19 00 	................
82                                              	.               
(end)
^C
```
Help on pointers how DSA broke this would be appreciated.

APCCV, 2024-11-25:
> @APCCV: 2 new versions after Netatalk 4.0.4 -> 4.0.5 and 4.0.6 :)

No impact on this problem.
Please stop spamming every ticket I opened with this.

APCCV, 2024-11-27:
Closing as reporting the issue using standard images.

closed 2024-11-27 by APCCV

Neustradamus, 2024-11-27:
Replaced by:
- https://github.com/openwrt/openwrt/issues/16983