openwrt/openwrt #16987: apk: package uci-defaults script may not get run in runtime installation

hnyman, 2024-11-17:
### Describe the bug

A package's uci-defaults script may not get properly run when package is installed with apk into a live router.

Since commit 9b9c78e in 2016 it has been the practice that the possible uci-defaults script of a package is run when the package is installed at run-time. The https://github.com/openwrt/openwrt/blob/main/package/base-files/files/lib/functions.sh function `default_postinst` should get evaluated and it should run the possible uci-defaults files.

With apk this seems broken.

This was first noticed when installing a LuCI theme. The new theme does not appear into the theme selection menu as the proper config items do not get places into /etc/config/luci.  Discussion in https://github.com/openwrt/luci/issues/7395

While testing that, I noticed that the uci-defaults file was properly placed into /etc/uci-defaults by apk, but it was not run.  At the next reboot the file then got properly run and the theme came available.

I tried adding debug statements into `default_postinst`, but they were never reached. Almost looks like the whole function is not reached

The apk package info file seems to contain the uci-defaults filename, so it should be found if the whole  `default_postinst` gets run:
```
root@router4:~# cat /lib/apk/packages/luci-theme-openwrt-2020.list
/usr/share/ucode/luci/template/themes/openwrt2020/footer.ut
/usr/share/ucode/luci/template/themes/openwrt2020/header.ut
/etc/uci-defaults/30_luci-theme-openwrt-2020
/lib/apk/packages/luci-theme-openwrt-2020.list
/www/luci-static/resources/menu-openwrt2020.js
...
```
 

### OpenWrt version

r28105-afffcd09e

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

 mediatek/mt7622

### Device

Linksys E8450 (UBI)

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2024-11-20:
Ok, this is reproducible at least.

Now we gotta figure out what is calling uci-defaults with OPKG

robimarko, 2024-11-20:
Ok, its all being done via `default_postinst` which is part of `functions.sh` in base-files.

Weird thing is that there is nothing APK specific regarding uci-defaults handling

robimarko, 2024-11-20:
The same was true for `default_prerm` which was also never called as `prerm-pkg` happened before it and it would also exit with 0 before ever reaching `default_prerm` thus not removing the config options that dont exist anymore.

hnyman, 2024-11-20:
> so I traced this down to default_postinst never actually being called at all.

Yep, that was also the outcome from  my initial debug when I opened this bug issue:

> > tried adding debug statements into default_postinst, but they were never reached. Almost looks like the whole function is not reached

Great that you figured out  a solution in #17023
I will test it once I get home (tomorrow) if testting is still needed then.

commit 5b6f8353 (2024-11-21): include/package-pack: actually call default_postinist/prerm when using APK

Currently, when you install a package that ships uci-defaults scripts they
are not getting executed during package installation.

For example, if you install a new LuCI theme it ships a uci-defaults script
to add it to the UCI LuCI config so the theme is actually selectable but
that does not happen until the device is rebooted so that uci-defaults
script is actually executed.

It turns out that the recipe for post-install is the issue since it will
include contents of postinst-pkg before the call to default_postinist.
 [... cut]
PR #17023 (2024-11-21): include/package-pack: actually call default_postinist/prerm with APK
Currently, when you install a package that ships uci-defaults scripts they are not getting executed during package installation.

For example, if you install a new LuCI theme it ships a uci-defaults script to add it to the UCI LuCI config so the theme is actually selectable but that does not happen until the device is rebooted so that uci-defaults script is actually executed.

It turns out that the recipe for post-install is the issue since it will include contents of postinst-pkg before the call to default_postinist. This is [... cut]
closed 2024-11-21 by openwrt-bot