openwrt/openwrt #16991: apk: restore links to busybox after "rich" package deletion

bol-van, 2024-11-17:
### Describe the bug

This is as simple as
apk add gzip
apk del gzip
no gzip at all. link is pointing to deleted binary.

Same thing with many other "rich" busybox analogs

### OpenWrt version

snapshot

### OpenWrt release

snapshot

### OpenWrt target/subtarget

x86+64

### Device

PC

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

efahl, 2025-06-11:
The root cause has two parts:

1) `apk` deinstall doesn't restore alternatives on deinstall, which should be fixed with https://github.com/openwrt/openwrt/pull/19093

2) The busybox Makefile does not define alternatives for its apps, which happened in https://github.com/openwrt/openwrt/commit/62be427067ee3883b68bcfb08dfc0c43dce22fa3.  So, unless we patch `apk` to behave like `opkg` in this regard (seems hacky to me), the fix will require some changes to the Makefile.

efahl, 2025-06-11:
For reference, here's where `opkg` was changed to special case busybox alternatives: https://github.com/openwrt/opkg-lede/commit/21b7bd73ae88ba2ce9a5bfeedf67bc204d10fd08

closed 2025-06-23 by openwrt-bot